jrowberg/i2cdevlib #500: mpu6050_dmp6  example does not work for ESP32

dsyleixa, 2019-12-16:


ZHomeSlice, 2019-12-17:
Arduino UNO must store the large BLOB of Firmware in ProgMemory as it can't load it into regular memory. 
The DMP portion is developed around this requirement and would have to be rewritten for other processors.
you may be able to remove the prog_ from all the files and get it to work.

Z

dsyleixa, 2019-12-17:
but then it won't work anymore on UNOs, and Nanos, right?

ZHomeSlice, 2019-12-17:
dsyleixa,
> Probably it would be better to rework the entire code and enable it for all boards via #if defined BOARDMACRO ?

Or would it be simpler to create a library for the ESP32 that can use more of the features of the MPU6050 and the MPU9250 which with the addition of the Library available (Not the BLOB firmware for the MPU but a separate Library) that will integrate the compass into the mix and provide true 9 Degrees of freedom?

If you're interested in the details I can point you to the files. It's a future project of mine but it currently is over my head (The library is not readable "libmplmpu.lib" for the 16-bit RISC CPU and comes with a .h file to provide the definitions) and as far as I can tell it won't fit into the UNO could be wrong. There also is a version for the ARM (libmplmpu.a). I have not figured out how to use these with the Arduino IDE. This would provide true 9 degrees of freedom using a magnetometer and would be well worth the time to develop a new and or improved library.

Z

dsyleixa, 2019-12-17:
I'm actually a simple Arduino user, just capable fo some ISO C code but almost nothing about C++ unfortunately, so tbh I'm afraid I'd need sort of full code ready to work with both my ESP32 and my Nano, my SAMD51,  and my ESP8266  (always never using IRQ pins though)

josefadamcik, 2020-04-16:
For anyone who found this while having the same problem: 

The compilation error messages are actually saying that the types are being defined again while they have been already defined. They even say where they have been defined in the first place. The whole discussion above is based on misunderstanding of those compilation errors.

There is no need to remove prog_ from all files at all or do some big changes. i2cdevlib can work on ESP32 Arduino core without issues.

The definitions are actually part of `esp32/pgmspace.h`  which is already included. The piece of code causing the error is actually a block of definitions meant for teensy which checks via `#ifndef __PGMSPACE_H_` but esp32 uses  `#define PGMSPACE_INCLUDE` to protect from redefinitions (those checks are around line 50 in all 3 header files with DMU support code - MPU6050_9Axis_MotionApps41.h etc).

**Workaround** is just to comment out those definitions. 
Better fix would be to check for `PGMSPACE_INCLUDE` too or maybe add a branch for ESP32. 

I could invest a bit more time and create a proper PR, but not sure if it makes sense. This project seems rather unmaintained (or maybe overwhelmed) and there are many opened PR's.

dsyleixa, 2020-04-16:
@ josefadamcik:
there are apearenty not only issues about ESP32 but also about SAMD51. So IMO a fully cross-platform-compliant version would be highly appreaciated!

ZHomeSlice, 2020-04-17:
The goal for this mpu6050 Version is to compact the MPU6050 firmware into a package that will fit into an UNO (atmega328). When we start looking at the much more powerful chips like the ESP32 additional features could be added. Such as the addition that will allow for 9 degrees of freedom rather than just the 6 degrees of freedom that the eDMP firmware allows.

IMO we would need a new DMP Firmware file that would achieve this.

dsyleixa, 2020-04-17:
there actually would be no need for additonal features currenty, just making the lib cross-platform-compliant by it's current features  would be highly appreciated, including a cross-platform-no-interrupt-version as requested in https://github.com/jrowberg/i2cdevlib/issues/441 , which all are supposed to work with all  boards, especially the modern ones (ESP8266, ESP32, SAM, SAMD21, SAMD51) out of the box.

josefadamcik, 2020-04-18:
> The goal for this mpu6050 Version is to compact the MPU6050 firmware into a package that will fit into an UNO (atmega328). When we start looking at the much more powerful chips like the ESP32 [...]

That of course makes sense.

But this example(?) coul be still used with ESP32 just by modifying few #ifdefs. Right now it's just not compiling and people might think that it is because it cannot work and would require a lot of work to make it work. But it is working like a charm for me. Even with interrupts (which are for some reason not really used in current examples).

dsyleixa, 2020-04-18:
> But this example(?) coul be still used with ESP32 just by modifying few #ifdefs. 

when I need to compile for different target boards then I always would have to change the entire lib to adjust it to the current arbitrary requirements and vice versa for the next different target board (Mega2560, Due, SAMD21. ESP32, SAMD51, ESP32).
So currently always having to patch the mpu6050_dmp lib and then repatch and then patch anew, and repath either was aagain would actually be much to cumbersome, tbh.

dsyleixa, 2020-04-18:
> But this example(?) coul be still used with ESP32 just by modifying few #ifdefs. 

when I need to compile for different target boards then I always would have to change the entire lib to adjust it to the current arbitrary requirements and vice versa for the next different target board (Mega2560, SAM Due, SAMD21 Zero, ESP32, SAMD51, ESP8266, SAM Due, SAMD51,...).
So currently always having to patch the mpu6050_dmp lib and then re-patch and after that patch anew, and re-patch either was aagain would actually be much to cumbersome, tbh.
So like for all (most of all) Arduino libs they actually are supposed to attach automatically to the target board automatically, and work out of the box (e.g., like Wire, Serial, Adafruit_GFX, Adafruit_ILI9341, elapsedMillis, Keypad, PCF8574, MCP3421, SdFat).
This cross-platform-compatibility is finally the fundamental Arduino philosophy.

ZHomeSlice, 2020-04-18:
While I've assisted in making the mpu6050 more efficient for the atmega328p boards I've worked little with other boards in General and I don't think I would be the one to make these changes.
The beauty about this type of environment is that your contributions and efforts can be shared with everybody. using your experience with other boards share with us these changes You are a requesting. so that we can understand what you are doing to get this library to work with other clips.
Z

dsyleixa, 2020-04-18:
I actually don't use it for different boads any more, because now that I have a working ESP32/noIRQ version it does not compile anymore for other boards, it was too cumbersome to always make all those lib patches and then bck again, so I dropped using the lib for all of them. 
Just if a new cross-platform compatible lib was provided then it would be of interest again.

ZHomeSlice, 2020-04-18:
@dsyleixa
> I actually don't use it for different boads any more, because now that I have a working ESP32/noIRQ version it does not compile anymore for other boards, it was too cumbersome to always make all those lib patches and then back again, eventually all was messed and muddled up and so I dropped using the lib for all of them.
Just if a new cross-platform compatible lib was provided then it would be of interest again.

Is it your creation or is it available so I can compare it
Does it use the eDMP  (Embeded Digital Motion Processing) firmware?

z

dsyleixa, 2020-04-18:
it's the one I have from https://github.com/jrowberg/i2cdevlib/issues/441

ZHomeSlice, 2020-04-18:
That's great 
So you're saying just by commenting out
```
/*
        typedef void prog_void;
        typedef char prog_char;
        typedef unsigned char prog_uchar;
        typedef int8_t prog_int8_t;
        typedef uint8_t prog_uint8_t;
        typedef int16_t prog_int16_t;
        typedef uint16_t prog_uint16_t;
        typedef int32_t prog_int32_t;
        typedef uint32_t prog_uint32_t;
*/
```
you got it to work properly on all but samd51?

Z

dsyleixa, 2020-04-18:
no, just for the ESP32, plus adding the 
`#define BUFFER_LENGTH 32`
thing.
most of my other boards I tried refuse to work because of similar things like for SAMD51, and so I resigned (frustrated)

ZHomeSlice, 2020-04-18:
I'll restate this.
```
#if defined(__ESP32__, __ESP8266__)
// We need to add this 
#define BUFFER_LENGTH 32

// We do not tneed these typedrfs
/*
        typedef void prog_void;
        typedef char prog_char;
        typedef unsigned char prog_uchar;
        typedef int8_t prog_int8_t;
        typedef uint8_t prog_uint8_t;
        typedef int16_t prog_int16_t;
        typedef uint16_t prog_uint16_t;
        typedef int32_t prog_int32_t;
        typedef uint32_t prog_uint32_t;
*/
#else 
        typedef void prog_void;
        typedef char prog_char;
        typedef unsigned char prog_uchar;
        typedef int8_t prog_int8_t;
        typedef uint8_t prog_uint8_t;
        typedef int16_t prog_int16_t;
        typedef uint16_t prog_uint16_t;
        typedef int32_t prog_int32_t;
        typedef uint32_t prog_uint32_t;
#endif
```
This should be the proper compiler directive if the #if defined(__ESP32__, __ESP8266__) is defigned properly.
Z

dsyleixa, 2020-04-18:
tbh, I am quite sure that it won't work for SAMD51 yet ever though, see 
https://github.com/jrowberg/i2cdevlib/issues/441#issuecomment-586882312

ZHomeSlice, 2020-04-19:
how does this look

```
#ifdef __AVR__
    #include <avr/pgmspace.h>

#elif defined(ESP8266)
    #define BUFFER_LENGTH 32
    #ifndef __PGMSPACE_H_
        #define __PGMSPACE_H_ 1
        #include <inttypes.h>

        #define PROGMEM
        #define PGM_P  const char *
        #define PSTR(str) (str)
        #define F(x) x

        #define strcpy_P(dest, src) strcpy((dest), (src))
        #define strcat_P(dest, src) strcat((dest), (src))
        #define strcmp_P(a, b) strcmp((a), (b))
        
        #define pgm_read_byte(addr) (*(const unsigned char *)(addr))
        #define pgm_read_word(addr) (*(const unsigned short *)(addr))
        #define pgm_read_dword(addr) (*(const unsigned long *)(addr))
        #define pgm_read_float(addr) (*(const float *)(addr))
        
        #define pgm_read_byte_near(addr) pgm_read_byte(addr)
        #define pgm_read_word_near(addr) pgm_read_word(addr)
        #define pgm_read_dword_near(addr) pgm_read_dword(addr)
        #define pgm_read_float_near(addr) pgm_read_float(addr)
        #define pgm_read_byte_far(addr) pgm_read_byte(addr)
        #define pgm_read_word_far(addr) pgm_read_word(addr)
        #define pgm_read_dword_far(addr) pgm_read_dword(addr)
        #define pgm_read_float_far(addr) pgm_read_float(addr)
    #endif

#else
    // Teensy 3.0 library conditional PROGMEM code from Paul Stoffregen
    #ifndef __PGMSPACE_H_
        #define __PGMSPACE_H_ 1
        #include <inttypes.h>

        #define PROGMEM
        #define PGM_P  const char *
        #define PSTR(str) (str)
        #define F(x) x

        typedef void prog_void;
 /*       
        typedef char prog_char;
        typedef unsigned char prog_uchar;
        typedef int8_t prog_int8_t;
        typedef uint8_t prog_uint8_t;
        typedef int16_t prog_int16_t;
        typedef uint16_t prog_uint16_t;
        typedef int32_t prog_int32_t;
        typedef uint32_t prog_uint32_t;
*/

        #define strcpy_P(dest, src) strcpy((dest), (src))
        #define strcat_P(dest, src) strcat((dest), (src))
        #define strcmp_P(a, b) strcmp((a), (b))
        
        #define pgm_read_byte(addr) (*(const unsigned char *)(addr))
        #define pgm_read_word(addr) (*(const unsigned short *)(addr))
        #define pgm_read_dword(addr) (*(const unsigned long *)(addr))
        #define pgm_read_float(addr) (*(const float *)(addr))
        
        #define pgm_read_byte_near(addr) pgm_read_byte(addr)
        #define pgm_read_word_near(addr) pgm_read_word(addr)
        #define pgm_read_dword_near(addr) pgm_read_dword(addr)
        #define pgm_read_float_near(addr) pgm_read_float(addr)
        #define pgm_read_byte_far(addr) pgm_read_byte(addr)
        #define pgm_read_word_far(addr) pgm_read_word(addr)
        #define pgm_read_dword_far(addr) pgm_read_dword(addr)
        #define pgm_read_float_far(addr) pgm_read_float(addr)
    #endif
#endif
```
Z

dsyleixa, 2020-04-19:
I don't see the #if defined ESP32 , but as stated, ESP32 current is NOT the most urgent problem to me because my patched version works - so no need to patch again and by that risking to mess the working version up eventually. 
Instead it's NOW more urgent about SAMD51, see https://github.com/jrowberg/i2cdevlib/issues/441#issuecomment-586882312

josefadamcik, 2020-04-19:
Hi @ZHomeSlice , thank you very much for trying to help and figure that out. As I said in my comment far above, I was always ready to help by creating a PR. Since I think the discussion here is a bit unfocused (we should be talking only about ESP32 in this issue) I have created the PR already (#530). The solution there is proven to work on hardware.

Few remarks: I have quickly tried to compile for ESP8266 and there were no compilation problems for that with the original library. So I think the checks and changes for ESP8266 are not necessary at all. 

Also, the BUFFER_LENGHT definition needs to be in MPU6050.h for things to compile properly.

dsyleixa, 2020-04-19:
just to clarify,
the ESP32 issue was the first one I had detected, but then it turned out that there are also issues about SAMD51, and I'm not sure if even more cores might be possibly affected.
So after all, only a cross-platform-compatible version would actually really be useful.
AFAIU, most issues are related to
- buffer size
- prog_int data types
- dtostrf (depricated, partially not supported)
- sprintf("%f", ...) (motly not available on AVR)

josefadamcik, 2020-04-19:
@dsyleixa I do understand that already form your previous comments. I sadly cannot help you with those issues. 

What I can do is to help with the original issue. Yes, that does not fix your actual problems with other boards. But it might help other people who would run into the original issue in the same way as I did. 

Normally I would create a new issue and later a PR to fix the problem. But since the exactly same issue was already here (yes, this one!), I just commented on it and offered my help.  I see no reason why fixing this issue for ESP32 should involve any other boards or issues you have found. If there are problems with other boards they should have their own tickets ( and you are already refering to #441 so there are some tickets out there).

dsyleixa, 2020-04-19:
yes, I see, thank you! I highly appreciate your work and your efforts, I was just thinking about some kinds of "extended perspective", particularly with respect to (e.g.) not further using the depricated dtostrf function.

ZHomeSlice, 2020-04-19:
`#define printfloatx(Name,Variable,Spaces,Precision,EndTxt) { Serial.print(F(Name)); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(dtostrf((float)Variable,Spaces,Precision ,S));}Serial.print(F(EndTxt)); }//Name,Variable,Spaces,Precision,EndTxt`
change to:
`#define printfloatx(Name,Variable,Spaces,Precision,EndTxt) Serial.print(Name); Serial.print(" "); Serial.print(Variable);Serial.print(EndTxt);`

dtostrf doesn't work on other processors bummer

Z

dsyleixa, 2020-04-21:
@ZHomeSlice  
yes, perhaps this might work! Can you issue a PR for that?

ZHomeSlice, 2020-04-21:
My version displays floating point numbers with padded spaces so the decimal point stays in the same position making the numbers easy to read. The change your wanting does not provide any formatting. 
Z

orhanyor, 2020-05-29:
> IMO that would be a minor drawback: better it works sufficiently than (on many boards) eventually not at all.
> Nonetheless, perhaps a proprietary ftoa() also might be a good choice instead.
> 
> `http://www.ars-informatica.ca/eclectic/ftoa-convert-a-floating-point-number-to-a-character-array-on-the-arduino/ `
> 
> ```
> void ftoa(float f, char *str, uint8_t precision) {
>   uint8_t i, j, divisor = 1;
>   int8_t log_f;
>   int32_t int_digits = (int)f;             //store the integer digits
>   float decimals;
>   char s1[12];
> 
>   memset(str, 0, sizeof(s));  
>   memset(s1, 0, 10);
> 
>   if (f < 0) {                             //if a negative number 
>     str[0] = '-';                          //start the char array with '-'
>     f = abs(f);                            //store its positive absolute value
>   }
>   log_f = ceil(log10(f));                  //get number of digits before the decimal
>   if (log_f > 0) {                         //log value > 0 indicates a number > 1
>     if (log_f == precision) {              //if number of digits = significant figures
>       f += 0.5;                            //add 0.5 to round up decimals >= 0.5
>       itoa(f, s1, 10);                     //itoa converts the number to a char array
>       strcat(str, s1);                     //add to the number string
>     }
>     else if ((log_f - precision) > 0) {    //if more integer digits than significant digits
>       i = log_f - precision;               //count digits to discard
>       divisor = 10;
>       for (j = 0; j < i; j++) divisor *= 10;    //divisor isolates our desired integer digits 
>       f /= divisor;                             //divide
>       f += 0.5;                            //round when converting to int
>       int_digits = (int)f;
>       int_digits *= divisor;               //and multiply back to the adjusted value
>       itoa(int_digits, s1, 10);
>       strcat(str, s1);
>     }
>     else {                                 //if more precision specified than integer digits,
>       itoa(int_digits, s1, 10);            //convert
>       strcat(str, s1);                     //and append
>     }
>   }
> 
>   else {                                   //decimal fractions between 0 and 1: leading 0
>     s1[0] = '0';
>     strcat(str, s1);
>   }
> 
>   if (log_f < precision) {                 //if precision exceeds number of integer digits,
>     decimals = f - (int)f;                 //get decimal value as float
>     strcat(str, ".");                      //append decimal point to char array
> 
>     i = precision - log_f;                 //number of decimals to read
>     for (j = 0; j < i; j++) {              //for each,
>       decimals *= 10;                      //multiply decimals by 10
>       if (j == (i-1)) decimals += 0.5;     //and if it's the last, add 0.5 to round it
>       itoa((int)decimals, s1, 10);         //convert as integer to character array
>       strcat(str, s1);                     //append to string
>       decimals -= (int)decimals;           //and remove, moving to the next
>     }
>   }
> }
> ```

im currently waiting for my mpu6050 to arrive and seeing its not compatible with samd51 boards was a let down. thanks to this thread i managed to get rid of the compile errors and for now it compiles for samd51 but i cant test it with the module because its still in the post :)
have you got it working with samd51?

orhanyor, 2020-05-29:
> me? unfortunately not, I'm still waiting for a fixed version which can be downloaded here.

can you compile and upload the code or you get errors? because strangely i does compile for me using feather m4.. so it might work, if only i had the module to test it.

dsyleixa, 2020-05-29:
are you talking about the version WITH or WITHOUT interrupt pin?

dsyleixa, 2020-05-29:
perhaps we should open a new issue about 
1 lib for dmp_6 without IRQ pin
1 example for dmp_6 without IRQ pin
to be compiled for ESP32 and ESP8266 and SAMD51 and SAM3X8E (Due) and Mega2560

dsyleixa, 2020-05-29:
this code:
```
/*
 * Name:       Arduino MPU6050 Polling Test.ino
    Created:  10/5/2019 1:31:08 PM
    Author:     FRANKWIN10\Frank
 
  This is a complete, working MPU6050 (GY-521) example using polling vs interrupts.
  It is based on Jeff Rowberg's MPU6050_DMP6_using_DMP_V6.12 example.  After confirming
  that the example worked properly using interrupts, I modified it to remove the need
  for the interrupt line.  
*/
 
#include "I2Cdev.h"
#include "MPU6050_6Axis_MotionApps_V6_12.h"  //changed to this version 10/04/19
#include "elapsedMillis.h"
#if defined (SAM)
   #include "avr/dtostrf.h"
#elseif defined(ESP32) || defined(ESP8266)  // <<<<<<<<<<<<< ????
   #include <stdlib_noniso.h>
#endif

// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69
MPU6050 mpu;
//MPU6050 mpu(0x69); // <-- use for AD0 high

#define _BV(bit) (1 << (bit))  
#define LED_PIN  LED_BUILTIN // (Arduino is 13, Teensy is 11, Teensy++ is 6)

 
bool blinkState = false;
 
// MPU control/status vars
bool dmpReady = false;  // set true if DMP init was successful
uint8_t mpuIntStatus;   // holds actual interrupt status byte from MPU
uint8_t devStatus;      // return status after each device operation (0 = success, !0 = error)
uint16_t packetSize;    // expected DMP packet size (default is 42 bytes)
uint16_t fifoCount;     // count of all bytes currently in FIFO
uint8_t fifoBuffer[64]; // FIFO storage buffer
 
// orientation/motion vars
Quaternion q;           // [w, x, y, z]         quaternion container
VectorInt16 aa;         // [x, y, z]            accel sensor measurements
VectorInt16 aaReal;     // [x, y, z]            gravity-free accel sensor measurements
VectorInt16 aaWorld;    // [x, y, z]            world-frame accel sensor measurements
VectorFloat gravity;    // [x, y, z]            gravity vector
float euler[3];         // [psi, theta, phi]    Euler angle container
float ypr[3];           // [yaw, pitch, roll]   yaw/pitch/roll container and gravity vector
 
//extra stuff
int IMU_CHECK_INTERVAL_MSEC = 100;
elapsedMillis sinceLastIMUCheck;
float global_yawval = 0.0; //contains most recent yaw value from IMU
int global_fifo_count = 0; //made global so can monitor from outside GetIMUHeadingDeg() fcn
 
// ================================================================
// ===                      INITIAL SETUP                       ===
// ================================================================
 
void setup()
{
  // I2C bus init done in SBWIRE.h
 
  Serial.begin(115200);
  while (!Serial); // wait for Leonardo enumeration, others continue immediately
 
  // initialize device
[... 103 lines cut ...]
 
    // otherwise, check for DMP data ready interrupt (this should happen frequently)
  }
  else if (mpuIntStatus & _BV(MPU6050_INTERRUPT_DMP_INT_BIT))
  {
    // read all available packets from FIFO
    while (fifoCount >= packetSize) // Lets catch up to NOW, in case someone is using the dreaded delay()!
    {
      mpu.getFIFOBytes(fifoBuffer, packetSize);
      // track FIFO count here in case there is > 1 packet available
      // (this lets us immediately read more without waiting for an interrupt)
      fifoCount -= packetSize;
    }
    global_fifo_count = mpu.getFIFOCount(); //should be zero here
 
    // display Euler angles in degrees
    mpu.dmpGetQuaternion(&q, fifoBuffer);
    mpu.dmpGetGravity(&gravity, &q);
    mpu.dmpGetYawPitchRoll(ypr, &q, &gravity);
  }
 
  float yawval = ypr[0] * 180 / M_PI;
  return  yawval;
}
```

dsyleixa, 2020-05-29:
for ESP32:

Arduino: 1.8.9 (Windows 7), Board: "Adafruit ESP32 Feather, 80MHz, 921600, None, Default"

D:\arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none,PartitionScheme=default -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_641572 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -verbose D:\Akten\Programmierung\ArduinoProgs\DriverLibs\MPU6050-dmp6-lib_JRowberg\MPU6050_DMP6_noIRQ\MPU6050_DMP6_noIRQ.ino
D:\arduino\arduino-builder -compile -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none,PartitionScheme=default -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_641572 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -verbose D:\Akten\Programmierung\ArduinoProgs\DriverLibs\MPU6050-dmp6-lib_JRowberg\MPU6050_DMP6_noIRQ\MPU6050_DMP6_noIRQ.ino
Using board 'featheresp32' from platform in folder: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4
Using core 'esp32' from platform in folder: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4
Detecting libraries used...
"D:\\arduino\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/heap" "-ID:\\arduino\\portable\\packages\\es [... cut ...]
Bibliothek I2Cdev im Ordner: D:\arduino\portable\sketchbook\libraries\I2Cdev (legacy) wird verwendet
Bibliothek Wire in Version 1.0.1 im Ordner: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\libraries\Wire  wird verwendet
Bibliothek MPU6050 im Ordner: D:\arduino\portable\sketchbook\libraries\MPU6050 (legacy) wird verwendet
Bibliothek elapsedMillis in Version 1.0.5 im Ordner: D:\arduino\portable\sketchbook\libraries\elapsedMillis  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Adafruit ESP32 Feather.

dsyleixa, 2020-05-29:
for Arduino Due SAMD3X8E:

```
Arduino: 1.8.9 (Windows 7), Board: "Arduino Due (Programming Port)"

D:\arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=arduino:sam:arduino_due_x_dbg -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_641572 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.bossac.path=D:\arduino\portable\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=D:\arduino\portable\packages\arduino\tools\bossac\1.6.1-arduino -verbose D:\Akten\Programmierung\ArduinoProgs\DriverLibs\MPU6050-dmp6-lib_JRowberg\MPU6050_DMP6_noIRQ\MPU6050_DMP6_noIRQ.ino
D:\arduino\arduino-builder -compile -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=arduino:sam:arduino_due_x_dbg -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_641572 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.bossac.path=D:\arduino\portable\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=D:\arduino\portable\packages\arduino\tools\bossac\1.6.1-arduino -verbose D:\Akten\Programmierung\ArduinoProgs\DriverLibs\MPU6050-dmp6-lib_JRowberg\MPU6050_DMP6_noIRQ\MPU6050_DMP6_noIRQ.ino
Using board 'arduino_due_x_dbg' from platform in folder: D:\arduino\portable\packages\arduino\hardware\sam\1.6.12
Using core 'arduino' from platform in folder: D:\arduino\portable\packages\arduino\hardware\sam\1.6.12
Build-Optionen wurden verändert, alles wird neu kompiliert
Detecting libraries used...
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10809 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "C:\\Users\\hw\\AppData\\Local\\Temp\\arduino_build_641572\\sketch\\MPU6050_DMP6_noIRQ.ino.cpp" -o nul
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10809 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "-ID:\\arduino\\portable\\sketchbook\\libraries\\I2Cdev" "C:\\Users\\hw\\AppData\\Local\\Temp\\arduino_build_641572\\sketch\\MPU6050_DMP6_noIRQ.ino.cpp" -o nul
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10809 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\syste [... cut ...]
Bibliothek I2Cdev im Ordner: D:\arduino\portable\sketchbook\libraries\I2Cdev (legacy) wird verwendet
Bibliothek Wire in Version 1.0 im Ordner: D:\arduino\portable\packages\arduino\hardware\sam\1.6.12\libraries\Wire  wird verwendet
Bibliothek MPU6050 im Ordner: D:\arduino\portable\sketchbook\libraries\MPU6050 (legacy) wird verwendet
Bibliothek elapsedMillis in Version 1.0.5 im Ordner: D:\arduino\portable\sketchbook\libraries\elapsedMillis  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Arduino Due (Programming Port).
```

dsyleixa, 2020-05-29:
So tbh, I really don't have any idea any more which working code actually to take.

orhanyor, 2020-06-01:
> So tbh, I really don't have any idea any more which working code actually to take.

im talking about with interrupt version. i had the same errors the ones you linked for the M4 but then i changed couple of things which were suggested here. one of them was changing buffer length and another thing which i cant remember right now but i dont get  any compilation errors right now. i can zip the file if you want to test it but it is for "with IRQ pin". 
i also should receive my my module in these days so i should be able to test it myself.

also have you tried this version: https://github.com/ElectronicCats/mpu6050

this seems like a modified version and compiles for samd51 without any modification

dsyleixa, 2020-06-01:
https://github.com/ElectronicCats/mpu6050
also seems to require IRQs and I needed to have  a  lib version which compiles without any changes to target either Board type

CS-Account, 2021-05-02:
I had a similar issue with my Seeeduino Xiao with a SAMD21.

In MPU6050.h, replace the relevant section with this:

`#ifdef __AVR__
#include <avr/pgmspace.h>
#elif defined(ESP32)
    #include <pgmspace.h>
    #ifndef BUFFER_LENGTH
        #define BUFFER_LENGTH 32
    #endif
#elif defined(_SAM_INCLUDED_)
    #ifndef BUFFER_LENGTH
        #define BUFFER_LENGTH 32
    #endif
#else
//#define PROGMEM /* empty */
//#define pgm_read_byte(x) (*(x))
//#define pgm_read_word(x) (*(x))
//#define pgm_read_float(x) (*(x))
//#define PSTR(STR) STR
#endif
`

This is the code I added which allowed it to build. Its partially from the PR mentioned above. I took out the #include <pgmspace.h> blind, because it was not available hoping it would work and it seems that it does. I have not tested further but it seems to work.

I also don't know if _SAM_INCLUDED_ is the correct Macro to be searching for, but it seems to be related.

I apologize for not being more knowledgeable on what's going on but I wanted to at least leave something in case someone else came along having the same problem.

dsyleixa, 2021-05-02:
the problem is that things work different for AVR,  SAMD21, SAMD51, SAM3X8E, ESP8266, ESP32, because most boards use different libs, e.g., some are sprintf()-capable for floats, some are not.
The 1st step must be to erase all non-standard functions from these dmp6050 libs,  like e.g., itoa(), atoi() or printfloatx()
then 2nd, always use just stdint.h data types (and don't redefine them!!),
and for all things which can't be standardized then valid macros for board types have to be declared, but as few as possible.

One also must consider, that dwarf-boards like UNO or Nano are no longer state of the art, as more and more powerful boards become more and more common and popular (SAMD21, SAMD51, ESP8266, ESP32, Cortex H7) and so libs like this dmp6050 should expectedly more target the new ones and in doubt drop, abandon and depricate the old puny and measly AVR-boards.

PR #530 (2021-06-25): Allow MPU6050 suport (including DMP) for Arduino to compile on ESP32 with Arduino core 
This is a tiny change which allows MPU6050 support library to compile on ESP32 with Arduino core. It also allows to compile DMP support and run the MotionApps examples on ESP32. I have verified this on hardware, everything works very well. This practical fixes #500

Changes are only small and related to 2 compilation problems:

Things related to progmem being redecalerd in MPU6050_..Axis_MotionApps_... header files. The code in those 3 files assumes that when`__AVR__` is not defined, it should be defining t [... cut]
closed 2021-06-25 by jrowberg