jrowberg/i2cdevlib #506: Setting up librarie. Too newbie please help

SMG90, 2020-01-18:
Hi, im new to using External libraries and im having problems adding it to the arduino  IDE. This directory for the MPU6050 is not available for the .zip download and i dont know  how to add it.

Someone please help :(

ZHomeSlice, 2020-01-18:
[https://github.com/jrowberg/i2cdevlib](https://github.com/jrowberg/i2cdevlib)
download the works here

SMG90, 2020-01-18:
> https://github.com/jrowberg/i2cdevlib
> download the works here

i tried that, but arduino said is not a valid library. I also tried renaming the zip file from i2cdevlib-master.zip to just i2cdevlib.zip and still wont recognize it.

ZHomeSlice, 2020-01-19:
Find the Arduino folder and find the MPU6050 folder only copy the MPU6050 folder into the Library folder located where your stretches are. you may wish to post this on the arduino.cc forum as you will have a quicker response to questions such as this.