jrowberg/i2cdevlib #512: Upgrade MPU-6050 library to Invensense legacy code

EmanuelFeru, 2020-02-11:
Hi @jrowberg ,

Great work on this library! I just want to come with a suggestion to upgrade the MPU-6050 library to the Invensense legacy code. I already tested it on GD32 and STM32 hoverboard sideboard, see:
* [hoverboard-sideboard-hack-GD](https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD)
* [hoverboard-sideboard-hack-STM](https://github.com/EmanuelFeru/hoverboard-sideboard-hack-STM)
which works great!

The advantage of using Legacy Invensense code is that they make available all the following nice features such as:
* Accelerometer and Gyroscope raw data output
* Quaternion output
* Temperature output
* Pedometer
* Interrupts on gesture events such as Tap on all axes or Orientation change
* Low-power modes
* Self-test and calibration

intensite, 2020-06-21:
Hello @EmanuelFeru,

I am indeed really interested in this, as I am suffering from limitation of the jrowberg/i2cdevlib implementation.  

Namely, since my board design has the IMU mounted vertically (with the Y axis pointing down instead of Z-Axis), the lack of proper initial orientation using the dmp_set_orientation() function makes the powerful DMP processing almost useless.

If my code is currently using MPU6050_6Axis_MotionApps_V6_12, is a total rewrite necessary to use the Invensense legacy code?

EmanuelFeru, 2020-06-21:
Hi @intensite,

In my implementation (links above), I refactored the original library provided by Invense. I also tryied using the library in this repo, but at least for me it was not able to write successfully the DMP firmware to the MPU6050. I noticed, (at least the writing function) is much shorter in the Invense library than in this repo:
-> [Invense Write DMP](https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/e9d74bea2949a5514f87384fd8ccf6a18188dec0/Src/mpu6050.c#L2705-L2736) vs [this repo](https://github.com/jrowberg/i2cdevlib/blob/3cc80d885ac9fcbee0d802c0f17e4cb4de56e4d5/Arduino/MPU6050/MPU6050.cpp#L3076-L3151). The one from Invense worked out-of-the-box on my  GD32 and STM32 microcontrollers.

Now, coming back to your questions, to use the Invense library, I think it shouldn't be too difficult to just copy my files. Roughly, I think you just need these files from my repo:
- `mpu6050.c`
- `mpu6050_dmp.c`

with the headers:
- `mpu6050.h`
- `mpu6050_dmp.h`
- `mpu6050_dmpKey.h`
- `mpu6050_dmpmap.h`
- `defines.h`

You might need to redirect [these functions](https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/e9d74bea2949a5514f87384fd8ccf6a18188dec0/Inc/defines.h#L38-L41) from `defines.h` to your i2c read/write function.

EmanuelFeru, 2020-06-22:
I forgot to say, the sensor orientation you can easily set it [here](https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD/blob/1e7bf7cd908325b07a5c98a541b92f42c9776c2c/Src/mpu6050.c#L51-L53).

intensite, 2020-06-22:
Thank you, I will start experimenting and report my progress (maybe on your repo)

ZHomeSlice, 2020-06-23:
This is an excellent example of the full Invensense code. 
 Jeffs reasoning for his version is how compact it can be and it is easier to understand as the embedded DMP firmware is already configured and ready to load. 
It has just the following features enabled for a small packet size across the i2c bus.
- Accelerometer and Gyroscope raw data output
- Quaternion output
- Automatic Gyro Calibration with V6.12
With Jeffs version, the atmega328p has more memory to perform other tasks than the core Invensense version. 
But with an ARM processor memory shouldn't be an issue, you can get Full 9 degrees of freedom once you add the MPL library they provide with the eMPL library!!!
This integrates the compass through the arm processor calibrating the gyro to magnetic north!
look for the MPL folder with the libmplmpu.lib library and associated .h files
Z

intensite, 2020-06-23:
@EmanuelFeru, 

I am afraid your port requires a little tinkering from my part to use on an Arduino / ESP32 platform.  
I am a bit lost at the moment.  
Would you happen to have a simple test example that compiles on an Arduino (or ESP32 using the Arduino platform).

EmanuelFeru, 2020-06-24:
Unfortunately, I don't have Arduino example because I only worked with STM32 and GD32 microcontroller for this MPU6050 project.
However, where are you stuck? Are there parts of the code not compatible with Arduino code style?

closed 2021-07-21 by EmanuelFeru