jrowberg/i2cdevlib #513: MPU6050_DMP6_using_DMP_V6.12 runtime calibration.

beprecon, 2020-02-12:
I saw your a comment (Run-time calibration routine is enabled which calibrates after no motion state is detected) in MPU6050_6Axis_MotionApps_V6_12.h.

Dose it means that  this code has MPL algorithm Gyro Calibration (Fast No Motion)  ?
(Gyro Calibration (Fast No Motion) : Run-time calibration routine. Once a no motion state is detected the gyro calibration will trigger. Calibration will complete within .5 seconds of no
motion state detection. in invensense manual)

I couldn't find the code of the part in this project.

ZHomeSlice, 2020-02-12:
When I created the V6.12 version, I configured the firmware the hard way and tested it. Then I added a function I created that captured the configured firmware before execution. this simplified the process of startup and eliminated about 100 lines of code to configure this feature on the fly. The intent was that the firmware is a black box that is extremely difficult to configure so why make you do it.
And memory is valuable so why to waste it. 
By the way, it's there you can check the firmware memory locations that get configured if you would like. 
Z

ZHomeSlice, 2020-02-12:
oh ya here's the code to capture the firmware:

```
bool view_DMP_firmware_Instance(uint16_t  length) {
	uint16_t  ii;
	uint16_t  this_read;
	uint16_t bankNum = 0;
	#define LOAD_CHUNK  (16)
	uint8_t cur[LOAD_CHUNK];
	Serial.print(F("const unsigned char dmp_memory[DMP_CODE_SIZE] PROGMEM = {\n"));
		for (ii = 0; ii < length; ii += this_read) {
			this_read = min(LOAD_CHUNK, length - ii);
			writeWords(devAddr, 0x6D, 1,  ii);
			readBytes(devAddr, 0x6F,  this_read, cur);
			if ((ii % (16 * 16)) == 0) {
				Serial.print(F("/* bank # "));
				Serial.print(bankNum++);
				Serial.println(F(" */"));
			}
			for (uint16_t c = 0; c < this_read; c++) {
				Serial.print(F(" 0x"));
				Serial.print(cur[c] >> 4, HEX); //Prints 0 insted of nothing when byte is less than 8
				Serial.print(cur[c] & 0X0F, HEX); // Prints the remainder of the hex number
				Serial.print(F(","));
			}
			Serial.println();
		}
	Serial.println(F("};"));
	return true;
}
```
Z

beprecon, 2020-02-12:
Um.. Does this code perform the gyro runtime calibration without calling the Run-time calibration routine function or not ?

ZHomeSlice, 2020-02-12:
Yes DMP_FEATURE_GYRO_CAL is enabled in this firmware instance. 
I marked the features with an "X" that are enabled in the firmware instance.

> Motion Driver 6.12 – User Guide
> Document Number: AN-EMAPPS-0.0.6
> Revision: 1.2
> Release Date: 05/05/2015
> 
> 12.2 DMP Features
> The DMP features many functions as detailed in the Features Guide. These functions can be dynamically enabled and disabled. The main API is
> int dmp_enable_feature(unsigned char mask);
> This function takes the mask and indexes into the correct memory address in the DMP firmware to enable and disable the feature.
> 
>  Features are
> CONFIDENTIAL & PROPRIETARY 11 of 14 PRELIMINARY
>  #define DMP_FEATURE_TAP (0x001)
>  #define DMP_FEATURE_ANDROID_ORIENT (0x002)
>  #define DMP_FEATURE_LP_QUAT (0x004)
>  #define DMP_FEATURE_PEDOMETER (0x008)
> X #define DMP_FEATURE_6X_LP_QUAT (0x010)
> X #define DMP_FEATURE_GYRO_CAL (0x020)
> X #define DMP_FEATURE_SEND_RAW_ACCEL (0x040)
> X #define DMP_FEATURE_SEND_RAW_GYRO (0x080)
>  #define DMP_FEATURE_SEND_CAL_GYRO (0x100)

Z