jrowberg/i2cdevlib #515: dmp reset function

beprecon, 2020-02-17:
I'm in a project with mpu6050.

Actually, I'm not a sense or embedded expert.

After finishing first dmp initialization, I want to make a function to reset only dmp values(quaternion, gyro, accel..) at specific  point or time.

Because using dmpInitialize and calibrate functions takes 2 seconds or more. but I need the fast reset function that takes 0.5 sec or less.

So I made a function in MPU6050_6Axis_MotionApps_V6_12.h with MPU6050 register map.

uint8_t MPU6050::dmp_reset_value() { 

  I2Cdev::writeBit(devAddr,0x68, 2, 1);
  I2Cdev::writeBit(devAddr,0x68, 1, 1);
  I2Cdev::writeBit(devAddr,0x68, 0, 1);
  I2Cdev::writeBit(devAddr,0x6A, 2, 1);
  I2Cdev::writeBit(devAddr,0x6A, 1, 1);
  I2Cdev::writeBit(devAddr,0x6A, 0, 1);

}

but it doesn't make the values to be reset(0).

How can I make this function ?

ZHomeSlice, 2020-02-18:
DMP is a black box and while every ounce of the memory is available to access for crazy purposes like your reset were to do that is not documented what might work is to wait to start the execution of the DMP code until you wish to start the calculations.

Under the hood :) in PU6050_6Axis_MotionApps_V6_12.h starting at line 368

These lines load the DMP code into MPU6050's memory:
```
	if (!writeProgMemoryBlock(dmpMemory, MPU6050_DMP_CODE_SIZE)) return 1; // Loads the DMP image into the MPU6050 Memory // Should Never Fail
```
Set the DMP code starting point. 
Either here or in the next few lines the DMP is enabled and starts to calculate the yaw Pitch and Roll. Yaw starts at zero pitch and roll uses the accelerometer to start their calculations at the appropriate angle.
you could likely wait here or break here to achieve the proper starting point.

```
	I2Cdev::writeWords(devAddr, 0x70, 1, &(ival = 0x0400)); // DMP Program Start Address
	I2Cdev::writeBytes(devAddr,0x1B, 1, &(val = 0x18)); // 0001 1000 GYRO_CONFIG: 3 = +2000 Deg/sec
	I2Cdev::writeBytes(devAddr,0x6A, 1, &(val = 0xC0)); // 1100 1100 USER_CTRL: Enable Fifo and Reset Fifo
	I2Cdev::writeBytes(devAddr,0x38, 1, &(val = 0x02)); // 0000 0010 INT_ENABLE: RAW_DMP_INT_EN on
	I2Cdev::writeBit(devAddr,0x6A, 2, 1);      // Reset FIFO one last time just for kicks. (MPUi2cWrite reads 0x6A first and only alters 1 bit and then saves the byte)
```
the following line could be removed to keep the MPU active. 
```
  setDMPEnabled(false); // disable DMP for compatibility with the MPU6050 library
```

Z

beprecon, 2020-02-18:
Actually, Using registers, I want to reset only the measurement values like gyro, accel, quaternion values...

the specific time isn't considered in DMP code.