jrowberg/i2cdevlib #519: MPU6050 DMP freezes after random duration

hendrikcech, 2020-02-29:
Hi!

I'm trying to track down an issue that I am having with the MPU6050_DMP6 example. I'm using an Arduino MKR Zero (ARM Cortex M0+) with a GY-521 MPU6050 breakout board. The interrupt wire is attached to digital pin A1.

The example works fine at first (the DMP values are read and printed correctly) but eventually the sensor seems to lock-up and not deliver any new values after about 5-60 seconds.

Here is a list of things I tried and observed:
- When using the old MPU6050_DMP6 example (the one before commit bed91df), I observed that `mpu.getIntStatus()` almost always returns 3 (`MPU6050_INTERRUPT_I2C_MST_INT_BIT`) and sometimes 2 (`MPU6050_INTERRUPT_PLL_RDY_INT_BIT`) under normal operation. Once the sensor locks up, `mpu.getIntStatus()` constantly returns 0 (`MPU6050_INTERRUPT_DATA_RDY_BIT`). I can still read the DMP values once the sensor locks up but they don't change anymore.
- With the new example, `mpu.dmpGetCurrentFIFOPacket(fifoBuffer)` always returns 0 because of `if (!fifoC) return 0;` in `dmpGetCurrentFIFOPacket`.
- The non-DMP raw example works fine. The issue seems to be with the DMP.
- I switched the cables (which are about 15cm long), the position of the sensor on the breadboard, and the sensor itself.
- Using MotionApps20 vs. V6.12 makes no differences.
- Once the sensor freezes, measurements with a multimeter show that SDA stays low, SCL is high, an the interrupt wire is high as well.

Does anybody have suggestions on what I could try to further nail down the issue or resolve it entirely?
Thanks!

RobertSpann, 2020-04-10:
Have a look on github for other projects that have been recently updated. There are other libraries that improved Jeff's project. Search for MPU6050.

ZHomeSlice, 2020-04-10:
Locking MPU6050, While I haven't experienced it. I was working with another person here on GitHub and he had a similar problem his locking issues were caused by noise from motors. I would verify that this isn't the cause. 
Here's his blog[https://www.fpaynter.com/2019/11/imu-motor-noise-troubleshooting-part-ii/]( https://www.fpaynter.com/2019/11/imu-motor-noise-troubleshooting-part-ii/)
Z

henrik04, 2020-04-23:
+1 MPU6050 either on a ESP WrooM 32 or ESP8266 or Arduino Nano all freeze randomly without explanation. 
- MPU6050_6Axis_MotionApps_V6_12.h
- Tried to slow down the I2c to 100 KHz instead of 400 no difference!
- I2C bus pulled up 4.7K to 3.3V to raise levels.. No difference.
- Lock the LCD display connected on the I2C bus as well.
- two 1000uF and one 47uF tantalum decoupling capacitors on each devices (using the I2C bus) : Same!
Really annoying..

ZHomeSlice, 2020-04-23:
@henrik04, Checking to see what voltage you are supplying to  pin VCC
Z

orhanyor, 2020-06-13:
after some research i found out that, simply connecting AD0 to ground solves the problem. its easy to overlook so dont let it float like i did! Its the address pin for the mpu6050.

i was able to run DMP example for around 2-3 minutes max. after fixing AD0 to ground i was able to run the same example code for more than 2 hours without a single freeze. I can probably reproduce the issue if i disconnect the AD0 cable while the code is running.

chrisvolto, 2020-12-29:
Hello, the solution of orhanyor didn't worked for me but after digging I resolved my problem thanks to this thread.
https://gitmemory.com/issue/jrowberg/i2cdevlib/543/707336866

In shorten, add `Wire.setWireTimeout(3000, true);` in your setup.
For example, the MPU6050_DMP6 example will be :
```c++
    // join I2C bus (I2Cdev library doesn't do this automatically)
    #if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
        Wire.begin();
        Wire.setClock(400000); // 400kHz I2C clock. Comment this line if having compilation difficulties
        Wire.setWireTimeout(3000, true); //timeout value in uSec
    #elif I2CDEV_IMPLEMENTATION == I2CDEV_BUILTIN_FASTWIRE
        Fastwire::setup(400, true);
    #endif
```
instead of
```c++
    // join I2C bus (I2Cdev library doesn't do this automatically)
    #if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
        Wire.begin();
        Wire.setClock(400000); // 400kHz I2C clock. Comment this line if having compilation difficulties
    #elif I2CDEV_IMPLEMENTATION == I2CDEV_BUILTIN_FASTWIRE
        Fastwire::setup(400, true);
    #endif
```

broken link alt
[image]

0ktidwell, 2021-09-26:
I had same problem for days, after 3-5 minutes, hang. I changed my baud rate from 9600 to 115200 for other reasons, no hang since, after running for many hours????/

ZHomeSlice, 2021-09-27:
This might be a solution:
https://github.com/jrowberg/i2cdevlib/issues/543#issuecomment-707336866
"It sounds to me like you have run up against the classic I2C bus 'freeze' problem, and if so, it has nothing at all to do with the MPU6060. I2C bus freezes are a known problem with Arduino,"

also...
This was mentioned near the bottom of that thread:
To resolve this issue you need to increase the delay between mpu creation task and readings from mpu
vTaskDelay(2000 / portTICK_PERIOD_MS);
Maybe it was related to some delay while MPU setting up all the things under the hood

hendrikcech, 2021-09-27:
I can't test the solution posted by chrisvolto but since other people seem to be happy with it, I will close this issue for now.

closed 2021-09-27 by hendrikcech

RazaZaidi2802, 2022-03-13:
> Hello, the solution of orhanyor didn't worked for me but after digging I resolved my problem thanks to this thread. https://gitmemory.com/issue/jrowberg/i2cdevlib/543/707336866
>
> In shorten, add [...]

This works for me too. 6050 kept freezing randomly between 10 to 60 seconds. It's been running for over an hour now.

dhruv221, 2023-06-25:
still the issue is not resolved! why??