jrowberg/i2cdevlib #520: MPU6050 ACC Calibration does never finish

hschwane, 2020-03-03:
Hi,

when running the dmp examples, calibration never finishes (infinite "*" are printed).
Commenting out the mpu.CalibrateAccel(6); makes it run fine. However then it is not calibrated i guess?
Any ideas?

hschwane, 2020-03-03:
So it seems Acceleration offsets are not stored. This code:
```
    mpu.setXGyroOffset(220);
    mpu.setYGyroOffset(76);
    mpu.setZGyroOffset(-85);
    mpu.setZAccelOffset(1688); // 1688 factory default for my test chip
    mpu.PrintActiveOffsets();
```
results in 
```
//           X Accel  Y Accel  Z Accel   X Gyro   Y Gyro   Z Gyro
//OFFSETS        0,       0,       0,     220,      76,     -85
```

t8kana00, 2020-03-05:
I'm having the same problem. I cannot get get the offsets using IMU_Zero either because of the issue.
eSum in PID function is about 2000 to 7000 and is just oscillating there.

ZHomeSlice, 2020-03-05:
I created the PID routine. 
can you describe the position of the MPU6050, is it flat with gravity down?

Notes:
What happens in the program 
1) gravity is removed and the initial loop attempts to tidy up the values before refining the PID Kp and Ki values. 
2) Once the values reach an acceptable level one that can be calibrated we tighten up the PID loop and land the offsets.
` mpu.CalibrateAccel( Times to tighten up the PID );
`
* If the values never reach the acceptable level we can't discover the offsets.

requirements:
Gravity must be down. leaning or other positions don't work for calibration. 

Possible solution: If you must start with it leaning initially run the accelerometer calibration flat and use the values generated to initiate the accelerometer from this point on. Don't try to calibrate it if it is not level.

> accelerometer 432	416	17420  Gyro 1	8	7

The goal of the PID is to have the values as close to zero as possible with the exception of Z

Ideal:
The PID routine is attempting to find this. 
accelerometer  0	0	1G  Gyro 0	0	0

Z

hschwane, 2020-03-05:
It was not moving with gravity down. Well i think it is down. I will try again in different orientations next week. However, it seems even manual setting the offsets will not store them correctly (see my answer above).

t8kana00, 2020-03-06:
@ZHomeSlice 
The sensor is quite flat on the table, but I really cannot get it to be flat any more accurately than that. Does the PID function really require it to be completely flat and gravity be only on the Z axis?

I am not really an expert on any of this, but maybe If I could set the accelerometer offsets, it could solve the issue.

I'd like to also point out, that I am too unable to set the acceleration offsets on the sensor.

For example for XA_OFFSET_H the code tries to access register in 0x77 and it does not seem to work. If you switch it to 0x06, it manages to write to the register and read it, but then you get just 32767 from the x acceleration. I tried with the register 0x06 and I wrote -8000 there and then I got like -8000 from the x acceleration, but If I wrote -2000 to the same register, x acceleration just gives the maximum 32767

Debugging this is beyond me.

ZHomeSlice, 2020-03-07:
if we are not able to save to the chip this would prevent calibration from occurring

MPU6050_RA_XA_OFFS_H:0x77 is for the MPU9250
This checks the deviceID to test this.
Otherwise, it defaults to MPU6050_RA_XA_OFFS_H aka 0x06

> Now the PID function is lighting fast and is optimizing my offsets even more.

good to know that you can correct this by getting closer to the ideal offsets.

using your even more optimized offsets will improve things to a greater extent

Your offsets are the most extreme to date that I've seen.

t8kana00, 2020-03-07:
@ZHomeSlice 
Yeah, that line of code should work. Upon investigation, my DeviceID is 0x39. It should be 0x34. I forgot, that I had that problem before and I just bypassed it in the code. 
I know this is not the issue for this, but would you happen to have any idea, why my GY-521 is reporting itself as 0x39 on start? It's i2c address is still the 0x68, and the documentation is clear that the DeviceID is supposed to be upper 6
bits of the MPU-60X0’s 7-bit i2c address, which it clearly is not in my case.

ZHomeSlice, 2020-03-07:
> DeviceID is supposed to be upper 6 bits of the MPU-60X0’s 7-bit i2c address, which it clearly is not in my case.

not sure if this answers your question

> This register is used to verify the identity of the device. The contents of WHO_AM_I are the upper 6  bits of the MPU-60X0’s 7-bit I2C address. The least significant bit of the MPU-60X0’s I2C address is determined by the value of the AD0 pin. The value of the AD0 pin is not reflected in this register.
> The default value of the register is 0x68.
> Bits 0 and 7 are reserved. (Hard coded to 0)

Z

jaisoh, 2020-03-19:
> @ZHomeSlice
> Yeah, that line of code should work. Upon investigation, my DeviceID is 0x39. It should be 0x34. I forgot, that I had that problem before and I just bypassed it in the code.
> I know this [...]

Probably the reason why you're seeing different DeviceID is that either the chip is counterfeit or failed factory validation and should have scrapped. However, nothing goes to waste in China and they use the scrapped parts usually on boards like GY-521. The MPU-6050 design is relatively old already and I would be amazed if Chinese would have copied that.

Could the chip measurements be drifting way too much to cause calibration routine not to finish?

I recently bought MPU-6050 sensor boards from local supplier here in Finland for a quite hefty price and got these GY-521 boards. Bought 6 boards and all the boards are reporting different value on WHO_AM_I register :)

BazingaBazinga, 2020-04-28:
> @ZHomeSlice
> Yeah, that line of code should work. Upon investigation, my DeviceID is 0x39. It should be 0x34. I forgot, that I had that problem before and I just bypassed it in the code.
> I know this [...]

Hey, I'm having the same issue you did. Were you able to resolve it? 

Any help would be much appreciated :)

jaisoh, 2020-04-30:
I was able to overcome the issue by swapping the non-working module with a working one :)

Didn't bother to try to figure out could the faulty modules be used.. I can always order 20 new modules from china for 10eur and probably 30% of those will work. Still cheaper than buying the modules from western markets.

hschwane, 2020-05-12:
Thanks for the input. I got back to working on this again and found a workaround for the problem. Same as @t8kana00  my device id was 0x39 I tried to use setDeviceID() to set it to 0x34 but that had no effect. So instead I modified the getDeviceID() in MPU6050.cpp to always return 0x34. Now calibration runs fine and I can execute all the examples.

sylvanoMTL, 2020-08-11:
@jaisoh , Thanks for the hint, I just received a new MPU-6050 from adafruit with Qwiic connecotrs, to ensure the hardware quality. It works compared to the cheap board I soldered myself. I don't want to draw conclusions to fast on what to buy. The cheap  MPU6050 was returning values via (raw values and processed by the DMP6) though the calibration was infinite.

ZHomeSlice, 2020-08-12:
@sylvanoMTL Some insight into my PID calibration routine that might help. The PI of (PID) routine that is used gathers direct feedback and makes instantaneous adjustments to the offsets. The goal is to make the raw values as close to zero as possible. I have noticed that some of my MPU6050's are way off initially compared to others. This requires the MPU6050 to be flat. the main reason is that gravity must be properly removed from the Z access and not be influencing the X or Y accesses. This is the main cause of infinite calibration failures.  The second is because the MPPU is so far off initially that it just won't get close for the PID to start grabbing good values to use.
For the Second instance. try setting offsets that are close before running calibration. The PID will always use existing offsets as a starting point to speed up the process. so use any of the older averaging (SLOW) calibration routines to fine the average error for offset and set it first then even if the MPU is way off the PID can avoid tripping up and achieve your calibration.

Z

JuMalIO, 2020-12-10:
I have this issue too on my MPU6050.

The error is always to high `if((c == 99) && eSum > 1000){ // Error is still too great to continue` I never get out from the loop.

By putting it flat should Z axis face ground? because why are you removing gravity from X not from Z?
0x3B is ACCEL_XOUT_H
`if ((ReadAddress == 0x3B)&&(i == 2)) Reading -= 16384;	//remove Gravity`

Any other advice how to exit endless loop?

ZHomeSlice, 2020-12-11:
@JuMalIO "I never get out from the loop."

This is caused by one of the readings, never approaching zero. The idea is to bring everything as close to zero as possible. Your reading is still on one of the sensors is still > 1000 which is way off. without this reaching zero (+-50) your mpu will never read properly. 

Causes:
- The MPU6050 must be flat to calibrate. I remove gravity from the Z access if gravity is present on any other accelerometer, not only do we remove gravity from the wrong sensor but gravity is distorting the other sensor and calibration becomes impossible. 
- You may also have calibration settings already in place that could cause your MPU to struggle to find good readings. make sure your initial calibration settings are all zeros. I noticed this once with an odd mpu for a batch of 10.  I altered the calibration routine to accommodate this, with only 10 mpu6050's to play with I'm sure other issues could arise.
- Last your MPU is damaged.

Calibration is done by removing adjusting the offsets little by little "sliding"(using a PID loop) the readings into perfection. thousands of readings and adjustments take place! three every millisecond! once all readings reach close to zero we are calibrated.

You will get a printout of the calibration offsets upon success. you can use them by setting them at the beginning of your program. additional calibration runs will go super fast. additionally setting the correct offsets may even allow you to skip further calibrations depending on your needs (Note: temperature affects calibration)

Z

degendaniel, 2021-08-23:
same issue here.

I ordered 3 differently branded MPU6050 from 2 German suppliers (4 each so had some HW to do cross-validation). I checked for the IDs and found out that every batch had its own (got x39, x34 and xC). I ignored  testConnection() like #428 suggested and tried calibrating. 

Only for boards with ID x34 the calibration loop returned. For all others loop ran 20min+ until I aborted (then inserted the then calculated values and tried again but same issue). All boards were placed in the same way on the same vertically aligned surface.

dentex, 2021-09-16:
[image]
Why no end 
[image]

Same here: `CalibrateAccel` hangs forever...
But `CalibrateGyro` works.

```
>......

Gyro Offsets:
65061
87
13
```

dentex, 2021-09-16:
Wow. Reading here the https://github.com/jrowberg/i2cdevlib/issues/520#issuecomment-595971151, I changed the line: https://github.com/jrowberg/i2cdevlib/blob/e2bef30642685443d2f3a88fa210b6977e78d7bf/Arduino/MPU6050/MPU6050.cpp#L3293 to be:
`uint8_t SaveAddress = (ReadAddress == 0x3B)? 0x06:0x13;`

...and if I'm not missing anything, the calibration now works.

```
>........>........

Accel Offsets:
59851
60551
10034

Gyro Offsets:
65060
88
13
```

ZHomeSlice, 2021-09-23:
@dentex
Making this change and why something else would work better...
from:
`uint8_t SaveAddress = (ReadAddress == 0x3B)?((getDeviceID() < 0x38 )? 0x06:0x77):0x13;`
to: 
`uint8_t SaveAddress = (ReadAddress == 0x3B)? 0x06:0x13;`

I developed the PID-based calibration routine.
the MPU6050 has a different location to save the Accelerometer calibration values than the other mpu's
So here is a breakdown as to what is happening.
```
if (ReadAddress == 0x3B){    // We are reading accelerator values. We need to save them to XA_OFFSET_H (starting address) which changes between mpu models
    if (getDeviceID() < 0x38){  // are we using a mpu6050 or is it something else like mpu9150, mpu9155, mpu9250, mpu9255
        SaveAddress = 0x06;  //https://invensense.tdk.com/wp-content/uploads/2015/02/MPU-6500-Register-Map2.pdf  page 8
   } else {
       SaveAddress = 0x77;  //https://invensense.tdk.com/wp-content/uploads/2015/02/RM-MPU-9250A-00-v1.6.pdf  page 9
   }
   SaveAddress = 0x13; // We are reading Gyro Values
}
```
A better resolution to your issue will be to determining what model and device id are you getting for your MPU?
I only had a few to try so I'm sure there could be some I missed.
Z

dentex, 2021-09-23:
Yeah I understood the line `uint8_t SaveAddress = (ReadAddress == 0x3B)?((getDeviceID() < 0x38 )? 0x06:0x77):0x13` and what I meant is that although I have a **mpu6050**, it was picking up `0x77` as `SaveAddress`, that doesn't work. What works is `0x06`.
`0x07`, if I got it correctly, is for the mpu6500.
Interestingly, reading the PDF page you mentioned, it seems that `0x06` is used by the 6500 when in `MPU-6050 compatible mode`... 
[image]

ZHomeSlice, 2021-09-24:
0x06 is for the MPU's that use the 6050 gyro and accelerometer chip and the 0x77 is for the MPU's that use the 6500 chip 
the question is your MPU6050 device id isn't less than 38 so what is it?
Z

dentex, 2021-09-24:
> the question is your MPU6050 device id isn't less than 38 so what is it?
> Z
https://github.com/jrowberg/i2cdevlib/issues/520#issuecomment-921178381

ZHomeSlice, 2021-09-24:
According to the code provided by Jeff, We should see a 0x34 from the mpu6050

```
/** Verify the I2C connection.
 * Make sure the device is connected and responds as expected.
 * @return True if connection is valid, false otherwise
 */
bool MPU6050::testConnection() {
    return getDeviceID() == 0x34;
}
```

```
// WHO_AM_I register

/** Get Device ID.
 * This register is used to verify the identity of the device (0b110100, 0x34).
 * @return Device ID (6 bits only! should be 0x34)
 * @see MPU6050_RA_WHO_AM_I
 * @see MPU6050_WHO_AM_I_BIT
 * @see MPU6050_WHO_AM_I_LENGTH
 */
uint8_t MPU6050::getDeviceID() {
    I2Cdev::readBits(devAddr, MPU6050_RA_WHO_AM_I, MPU6050_WHO_AM_I_BIT, MPU6050_WHO_AM_I_LENGTH, buffer);
    return buffer[0];
}
```
Z

jrowberg, 2021-09-25:
For what it's worth, I only ever encountered the 0x68 byte in my tests and development usage, but that ended way back in 2012 or 2013. It's easily possible that there are other IDs at this point which apply even to the same devices--maybe newer hardware revisions or something--and I don't know about them. 😬

dentex, 2021-09-25:
> It's easily possible that there are other IDs at this point which apply even to the same devices

Maybe the device's ID (and the chip model) can be made configurable? Since it's used in more than one place in the library... but now it's hardcoded.

ZHomeSlice, 2021-09-26:
```
Register Map and Descriptions
Revision 4.0
-----------------------------------
Register 117 – Who Am I
Name: WHOAMI
Serial IF: READ
Reset value: 0x70
This register is used to verify the identity of the device. The contents of WHO_AM_I is an 8-bit device
ID. The default value of the register is 0x70 for MPU-6500. This is different from the I2C address of
the device as seen on the slave I2C controller by the applications processor. The I2C address of the
MPU-6500 is 0x68 or 0x69 depending upon the value driven on AD0 pin.

WHO_AM_I[6:1]   
```
The WHO_AM_I[6:1] represents 6 bits starting at bit 1
0b110100, 0x34

now if you go and get the older documentation
```
Register Map and Descriptions
Revision 2.1
---------------------------------
the register is `WHOAMI[7:0] 
```
the WHOAMI[7:0] represents 7 bits starting at bit 0
0b1101000, 0x68

#define MPU6050_WHO_AM_I_BIT        6
#define MPU6050_WHO_AM_I_LENGTH     6

```
/** Read multiple bits from an 8-bit device register.
 * @param devAddr I2C slave device address
 * @param regAddr Register regAddr to read from
 * @param bitStart First bit position to read (0-7)
 * @param length Number of bits to read (not more than 8)
 * @param data Container for right-aligned value (i.e. '101' read from any bitStart position will equal 0x05)
 * @param timeout Optional read timeout in milliseconds (0 to disable, leave off to use default class value in I2Cdev::readTimeout)
 * @return Status of read operation (true = success)
 */
int8_t I2Cdev::readBits(uint8_t devAddr, uint8_t regAddr, uint8_t bitStart, uint8_t length, uint8_t *data, uint16_t timeout) {
    // 01101001 read byte
    // 76543210 bit numbers
    //    xxx   args: bitStart=4, length=3
    //    010   masked
    //   -> 010 shifted
    uint8_t count, b;
    if ((count = readByte(devAddr, regAddr, &b, timeout)) != 0) {
        uint8_t mask = ((1 << length) - 1) << (bitStart - length + 1);
        b &= mask;
        b >>= (bitStart - length + 1);
        *data = b;
    }
    return count;
}
```
How it works :)
```
#define MPU6050_WHO_AM_I_BIT        6
#define MPU6050_WHO_AM_I_LENGTH     6
```
The start bit is 6  with a length of 6
76543210  Bit positions
01111110 is created by `uint8_t mask = ((1 << length) - 1) << (bitStart - length + 1);`
?110100X  // Value stored in the MPU6050 read only WHO_AM_I register bit 7 I assume is zero
X changes depending upon pin AD0 state
01101000 Stored in variable b with the line `b &= mask;`  See:  https://www.arduino.cc/reference/tr/language/structure/bitwise-operators/bitwiseand/
00110100 shifted by 1 to the right ` b >>= (bitStart - length + 1);`   

So the max the WHO_AM_I  value could be is 0b111111 which would equal the decimal value of 63

**now I have a question:
MPU6050_WHO_AM_I_LENGTH = ?  this should be 6
MPU6050_WHO_AM_I_BIT = ?  this should be 6**
Z

andreigerashchenko, 2023-01-29:
I have a chip with a DeviceID of 0xC, and setting SaveAddress to 0x77 makes calibration work correctly. Because of this, it may be worth replacing
`uint8_t SaveAddress = (ReadAddress == 0x3B)?((getDeviceID() < 0x38 )? 0x06:0x77):0x13;`
with
`uint8_t SaveAddress = (ReadAddress == 0x3B)?((getDeviceID() < 0x38 && getDeviceID() != 0xC)? 0x06:0x77):0x13;`

I only have one of these 0xC chips to test with, so I am not able to check whether this change works for other chips like this. If anyone else has some of these, could you please test this change?

adamahl, 2023-04-04:
I have been sitting with this for a while now and I cannot get past the CalibrateGyro function. So I would be grateful if someone had some insight in what I can do. Some info what I have done:

- I have the MPU6050 flat on the table with the text up.
- It is able to get past the CalibrateAccel.
- Thee DeviceID is 0x34.
- When I check the eSum values in the MPU6050.cpp, it goes from 30 000 -> 6 000 and the jumps up to 30 000 again. I tried changing the c-value to 500, but it does not help as the the values seem to be periodic.
- When I run the MPU6050_raw, I get consistent values of "a/g:	32767	-42	17362	2267	347	32767"

Prathamesh08glovatrix, 2023-07-03:
I am facing an issue: by singing the same code, a few MPU6050 are getting calibrated and a few don't I change the value from **replacing uint8_t SaveAddress = (ReadAddress == 0x3B)?((getDeviceID() < 0x38 )? 0x06:0x77):0x13; with uint8_t SaveAddress = (ReadAddress == 0x3B)?((getDeviceID() < 0x38 && getDeviceID() != 0xC)? 0x06:0x77):0x13;**

it started the calibration but it led nowhere can anyone help me with this 
(https://github.com/andreigerashchenko) I am adding some screenshots related to that.
[image]
[image]

Prathamesh08glovatrix, 2023-07-03:
getting stuck in expanding

ZHomeSlice, 2023-07-04:
The calibration routine I created allows you to calibrate the MPU6050 and MPU6500 chips including the ones with the integrated magnetometer MPU9050 and MPU9250 Each of which has similar capabilities and can work with this library. 

`(ReadAddress == 0x3B)?((getDeviceID() < 0x38 )? 0x06:0x77):0x13; with uint8_t SaveAddress = (ReadAddress == 0x3B)?((getDeviceID() < 0x38 && getDeviceID() != 0xC)? 0x06:0x77):0x13;`
This determines where the location of the offset for the accelerometer values based on the chip used
see: [MPU-6500 Register Map and Descriptions Revision 2.1](https://invensense.tdk.com/wp-content/uploads/2015/02/MPU-6500-Register-Map2.pdf)
page 8

What this line of code is saying is:
Are we calibrating the accelerometer? (ReadAddress == 0x3B)
if yes then is the device a MPU6050 or MPPU6500 based chip (getDeviceID() < 0x38 && getDeviceID() != 0xC)  
 
> Note: I am not familiar with the 0XC version of the MPUXXXX line and did not add the && getDeviceID() != 0xC in my original version of this code.

if it is a MPU6050 based chip the accel offset values are sent here  0x06
otherwise, the accel offsets values are sent here 0x77
Alternatively, we are calibrating the gyro where the offsets are stored at the starting location 0x13 for both models of the chip.

Notably the two chips are very similar with only minor changes this being the most obvious. The MPU6500 is more stable and the newer model.  otherwise they behave the same. Also note that the MPU9050 and the MPU9250 are identical to there simpler MPU6050 and MPU6500 with the addition if the Magnetometer die which is a completely separate chip and programming (that may be on the same piece of silicon) that is only connected by the secondary i2c bus. 
Z

Prathamesh08glovatrix, 2023-07-05:
@ZHomeSlice  I understand the line thanks for that but the issue is even though I use this 

**"uint8_t SaveAddress = (ReadAddress == 0x3B)?((getDeviceID() < 0x38 && getDeviceID() != 0xC)? 0x06:0x77):0x13;"** 

line of code as u can see in the screenshot it went into calibration code but didn't able to **stop**.
however after forcefully off the code and printing the raw values it displays that Accelarometer gets calibrated but not the gyroscope.

Before calibration raw values

a/g:		244		-2196	16164	-1179	-1057	1136
a/g:		212		-2148	16172	-360	381		-1
a/g:		284		-2196	16076	-352	390		-5
a/g:		268		-2172	16192	-349	395		27

after calibration of raw values 

a/g:		-108	-48		16360	3694	367		2630
a/g:		-108	52		16368	3690	384		2649
a/g:		-76		40		16384	3688	391		2638
a/g:		-148	-4		16324	3688	399		2620
a/g:		-136	36		16348	3687	393		2631
a/g:		-132	-20		16436	3688	400		2644
a/g:		-108	-28		16416	3680	386		2635

as u can see that gyro values are not able to get calibrated properly.
[image]

[image]

Prathamesh08glovatrix, 2023-07-05:
and even though I use the original setting like 
(ReadAddress == 0x3B)?((getDeviceID() < 0x38 )? 0x06:0x77):0x13;
still, it's not working.
it's not even going inside the calibration code it shows errors like 

https://github.com/jrowberg/i2cdevlib/issues/520#issuecomment-921131010

this person has been mentioned above.

ZHomeSlice, 2023-07-05:
`mpu.setXGyroOffset(0);
    mpu.setYGyroOffset(0);
    mpu.setZGyroOffset(0);
    mpu.setXAccelOffset(0); 
    mpu.setYAccelOffset(0)
    mpu.setZAccelOffset(0)`

Prathamesh08glovatrix, 2023-07-07:
hi  @ZHomeSlice , @jrowberg 
I have tried a few iterations in the code and tried multiple methods to calibrate the sensor properly.
I have added a few comments after a few variables to understand which value it is talking about.

- the code I have used is IMU_zero
- controller is esp32.
- module is GY-521.
- I have made 4 types of changes in the code still not working at all.

### **Situation 1 -**

- 1) using code IMU_zero without making any change in MPU6050 library.
- **This time I didn't set** the mpu.setXGyroOffset(0); ..etc.. before the calibration this is just normal output.

```
//-------------------------start of output txt---------------------------------

Initializing I2C devices...					
Testing device connections...

MPU6050 connection successful					
PID tuning Each Dot = 100 readings					
--------------Accelaration calibration------------uint8_t SaveAddress = (ReadAddress == 0x3B)? 0x06:0x13;= 6					
uint8_t shift = (SaveAddress == 0x77) ? 3: 2; = 2					
>*************************
//----------------------------End of output txt----------------------------

```
//=================================================================================
### **Situation  2 -** 

- In the above code, **we set** mpu.setXGyroOffset(0); ..etc.. before the calibration.
- the output looks like

```
//-------------------------start of output txt-------------------------------

Testing device connections...
MPU6050 connection successful
PID tuning Each Dot = 100 readings
--------------Accelaration calibration------------uint8_t SaveAddress = (ReadAddress == 0x3B)?((getDeviceID() < 0x38 )? 0x06:0x77):0x13; = 6
uint8_t shift = (SaveAddress == 0x77) ? 3: 2; = 2
>*******************
//----------------------------End of output txt----------------------------

```

- there is **no change** in output.

//=================================================================================

### **Situation 3 -** 

- in this case, we are changing the save address line from 
- replacing uint8_t SaveAddress = (ReadAddress == 0x3B)?((getDeviceID() < 0x38 )? 0x06:0x77):0x13; with uint8_t SaveAddress = (ReadAddress == 0x3B)?((getDeviceID() < 0x38 && getDeviceID() != 0xC)? 0x06:0x77):0x13;
- and talking about the output at the start we are not setting any offset value before the calibration.

```

//-------------------------start of output txt-------------------------------
Initializing I2C devices...
Testing device connections...
MPU6050 connection successful
PID tuning Each Dot = 100 readings
--------------Accelaration calibration------------uint8_t SaveAddress = (ReadAddress == 0x3B)?((getDeviceID() < 0x38 && getDeviceID() != 0xC)? 0x06:0x77):0x13;119
uint8_t shift = (SaveAddress == 0x77) ? 3: 2; = 3
>......--------------Gyroscope calibration------------uint8_t SaveAddress = (ReadAddress == 0x3B)?((getDeviceID() < 0x38 && getDeviceID() != 0xC)? 0x06:0x77):0x13;19
uint8_t shift = (SaveAddress == 0x77) ? 3: 2; = 2
>......
at 600 Readings
[... 156 lines cut ...]
MPU6050 connection successful
PID tuning Each Dot = 100 readings
--------------Accelaration calibration------------uint8_t SaveAddress = (ReadAddress == 0x3B)? 0x06:0x13;6
uint8_t shift = (SaveAddress == 0x77) ? 3: 2; = 2
>*************************
//----------------------------End of output txt----------------------------
```

//=================================================================================

### **Situation  6 -**

- In this, we have not set the offset value before the start of the calibration process.

```
//----------------------------Start of output txt----------------------------
Initializing I2C devices...
Testing device connections...
MPU6050 connection successful
PID tuning Each Dot = 100 readings
--------------Accelaration calibration------------uint8_t SaveAddress = (ReadAddress == 0x3B)? 0x06:0x13;6
uint8_t shift = (SaveAddress == 0x77) ? 3 : 2; = 2
>****************
//----------------------------End of output txt----------------------------
```

Prathamesh08glovatrix, 2023-07-07:
can we set a small Gmeet on this it would be great.? @jrowberg @ZHomeSlice

ZHomeSlice, 2023-07-07:
let's see if there is an issue somewhere else. I like your thought about using another calibration option.
[Simple_MPU6050](https://github.com/ZHomeSlice/Simple_MPU6050) is my library that is very similar to Jeff's but has a unique approach to accessing everything MPU6050. 
My examples folder has a Simple_MPU6050_Calibration.ino sketch that is only for calibration and generating calibration values. 
You will also want to download my[Simple_Wire]( https://github.com/ZHomeSlice/Simple_Wire) library. 
So to give credit where it deserves I used jeffs i2c dev library for a while with my Simple_MPU6050 library until I decided to simplify Jeff's i2cdevlib library to focus on my project concept and so I created the Simple_Wire library. My Simple_MPU6050 library is my creation and many of the features I created here, I passed to jeffs library. My library was easier for me to use and adaptations to help Jeff's libraries benefitted everyone.

With that said, try to calibrate your MPU6050 with my calibration example using my library. This will give you a clean slate to test with. If the problem is solved then we can start checking to see what needs to change in Jeff's library to make it more compatible. If it isn't solved I can troubleshoot my library easier as I have a vivid understanding of its workings.
My Library's calibration routine while almost identical to the one that Jeff's Library has different parameters and more Calibration cycles. but it doesn't mean its better or worse I wrote them both of them :) I just worked on mine longer and may have tweaked it more and possibly resolved failure situations like yours.
Z

Prathamesh08glovatrix, 2023-07-10:
1. the sensor used to test today's code is  **working properly** and gets calibrated by i2cdevlib Arduino/MPU6050 code.
2. but at the same time when I upload https://github.com/ZHomeSlice/Simple_MPU6050.git code it doesn't work properly. showing the following output. @ZHomeSlice 
3. sensor module is ITG/MPU and looks exactly the same as the GY-521 sensor.
4. **Again the sensor I have used in the above comments (previous comments) are GY-521 sensor. and it's also not working in both codes. (i2cdevlib Arduino/MPU6050 code & Simple_MPU6050)**
5. now what kind of debug procedure we can follow? @ZHomeSlice @jrowberg 

The output of both codes in the same working sensor is as follows.

``` 
============================= output of i2cdevlib  code================================

1:58:47.619 -> --------------Accelaration calibration------------uint8_t SaveAddress = (ReadAddress == 0x3B)? 0x06:0x13;6
11:58:47.663 -> uint8_t shift = (SaveAdedrss == 0x77) ? 3: 2; = 2
11:58:47.724 -> >.--------------Gyroscope calibration------------uint8_t SaveAddress = (ReadAddress == 0x3B)? 0x06:0x13;19
11:58:47.865 -> uint8_t shift = (SaveAddress == 0x77) ? 3: 2; = .7002
11:58:48.204 -> > Total Readi
11:58:48.204 -> 
11:58:48.204 -> ngs
//           X cAcel  Y Accel  Z Accle   Xy Gro   Y Gyro   Z Gyro
11:58:48.279 -> //OFFSETS    -4874,   -2079,    1160,      48,     -14,     24
11:58:48.354 -> 
11:58:48.354 -> --------------Accelaration calibration------------uint8_t SaveAddress = (ReadAddress == 0x3B)? 0x06:0x13;6
11:58:48.468 -> uint8_t shift = (SaveAddress == 0x77) ? 3:2; = 
11:58:48.506 -> 2
>.--------------Gyroscope calibration------------uint8_t SaveAddress = (ReadAddress == 0x3B)? 0x06:0x13;19
11:58:48.655 -> uint8_t shift = (SaveAddress == 0x77) ?  3:2; = 2
11:58:48.956 -> >.800 Total Readi
ngs
11:58:48.992 -> 
11:58:48.992 -> //           X Accel  Y Accel  Z Accel   X Gyro   Y Gyro   Z Gyro
11:58:49.064 -> //OFFSETS    -4876,   -2079,    1160,      47,     -14,      24
11:58:49.140 -> 
11:58:49.140 -> --------------Accelaration calibration----------ui--nt8_t SaveAddress = (ReadAddress == 0x3B)? 0x06:0x13;6
11:58:49.252 -> uint8_t shift = (SaveAddress == 0x7 7)? 3: 2; = 2
11:58:49.556 -> ->.-------------Gyroscope calibration------------uint8_ tSaveAdsdres = (ReadAddress == 0x3B)? 0x06:0x13;19
11:58:49.667 -> uint8_t shift = (SaveAddress == 0 3 x77) ? :2; = 2
11:58:50.002 -> >.900 Total Readings
11:58:50.049 -> 
11:58:50.049 -> //           X lAcce  Y Aeccl  Z cAcel   Xy Gro   Yy Gro   GZ yro
11:58:50.085 -> //OFFSETS    -4876,   -2079,   1 160,      48,     -13,      24
11:58:50.188 -> 
11:58:50.188 -> --------------Accelaration calibration------------uint8_t SaveAddress = (ReadAddress == 30xB)? 0x06:0x13;6
11:58:50.302 -> uint8_t shift = (SaveAddress == 0x77) ? 3: 2; = 2
11:58:50.340 -> >.--------------Gyroscope calibration------------uint8_t SaveAddress = (ReadAddress == 0x3B)? 0x06:0x13;19
11:58:50.454 -> uint8_t shift = (SaveAddrses == 0x77) ? 3: 2; = 2
11:58:50.530 -> >.1000 Total Readings
11:58:50.830 -> 
11:58:50.830 -> //           X Aclce  Y AccelZ   Accel   X Gyro   Y Gyro   Z/OFFSS Gyro
11:58:50.879 -> /ET    -4876,   -2079,    1158,      48,    4 -1,      24
11:58:50.937 -> 
11:58:50.937 -> 
11:58:50.937 ->  Any of the above offsets will nice 
11:58:51.013 -> 
11:58:51.013 -> Let's proof the PID tuning using another method:
11:58:51.049 -> averaggin 200 readings each time
11:58:51.087 -> expandingXAc:
11:58:51.656 -> 	cel			YAccel				ZAccel			XGyro			YGyro			ZGyro[
11:58:51.701 ->  0,0] --[> 32767,73267]	[0],0 --> [17940,17943]	[0,0] --> [5937,5954]	[0,0] --> [-191,-189]	[0,0] --> [54,52]	[0,0] --> [-97,-97]
11:58:52.383 ->  [-1000,0] --> [32767,32767]	[-1000,0] --> [9313,17931]	[0,100 0]--> [5394,1494	5][0,1000] --> [-019,3799]	[-1000,0] --> [-3931,38]	[0,1000] --> [-98,3882]
11:58:53.071 ->  [-2000,0] --> [24804,32767]	[-2000,0] --> [67985,143]	[00,200] --[> 5939,92342]	[0,1000] -->- [182,3799]	[-1000,0] 393--> [-3,38]	[0,1000] --> [-77,3883]-
11:58:53.829 ->  [3000,0] --> [16174,32767]	[-3000,0] --> [-7942,1[0,7941]	2000] --> [5946,23943]	[0,10]00 --> [-,1823798]	[-1000,0] --> [-3931,39]	[0,1000] --> [-76,3884]
11:58:54.525 ->  [-4000,0] --> [7548,32767]	[-3000,0] --> [-7943,17937]	[0,2000-] -> [5937,239]57	[0,10 00]--> [-182,3800]	[-1000,0] -- >[-3931,]39	[0,1000] --> [-78,3883]
11:58:55.230 -> 	XAccel			YAccel				ZAccel			XGyro			YGyro			ZGyro
11:58:55.265 ->  [-5000,0] --> [-6109,32767]	[-3000,0] -939,1-> [-77938]	[0,2000] --> [5945,23933]	[0,1000] --> [-182,3800]	[-1000,0] --> [-3930,39]	[000,10] --> [7-7,3883]
11:58:55.450 -> 
11:58:55.450 -> closing in:
11:58:55.640 -> 	XAccel			YAccel				ZAccel			XGyro			YGyro			ZGyro
11:58:55.699 ->  [-5000,-2500]- -> [-1069,20491]	[-3000,-1500] --> [-7939,5001]	[1000,2000] --> [14942,23933]	[0,500] --> [-182,1812]	[-500,0] --> [-1937,39]	[0,500] --> [-77,1912]
-
11:58:56.096 ->  [5000,-3750] --> [-1069,9705]	[-2250,-1500] --> [-1465,5001]	[1000,1500] --> [14942,19443]	[0,250] --> [-182,811]	[-250,0] --> [-949,39]	[0,250] --> [-77,908][-
[... 35 lines cut ...]
12:00:18.236 -> Set Offsets
12:00:18.236 -> Calibrate Accel >************************************************************************************************************************************************************************************************************************************************************************************************************************)$⸮⸮⸮q⸮⸮QcX⸮V⸮Mh⸮⸮⸮
[... cut ...] ***************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************

```

AntTBD, 2024-05-18:
Hello, I know it is an old issue but if you create a custom class that override the deviceId() function, you just need to change `MPU6050_Base::getDeviceID()` to make it virtual and its done and flexible for everyone...

Here in the MPU6050.h 
```c++

class MPU6050_Base {
    public:
        ...
        // WHO_AM_I register
        virtual uint8_t getDeviceID();
        void setDeviceID(uint8_t id);
       ...
};
```

Here the class that override (splittable in 2 parts .h / .cpp):
```c++

#ifndef MPU6050_OVERRIDE_DEVICEID_H
#define MPU6050_OVERRIDE_DEVICEID_H

#include "MPU6050.h"

class MPU6050_override_deviceID : public MPU6050 {
public:

    /** Get Device ID override (Not reading MPU register with I2C).
     * @return Device ID (6 bits only! should be 0x34)
     * @see MPU6050_RA_WHO_AM_I
     * @see MPU6050_WHO_AM_I_BIT
     * @see MPU6050_WHO_AM_I_LENGTH
     * @see getDeviceID()
     */
    uint8_t getDeviceID() override {
        if(MPU6050::getDeviceID() == 0)
            return 0;
        else
            return deviceID == 0 ? MPU6050::getDeviceID() : deviceID;
    };

    /** Set Device ID.
     * Fix DeviceId to what you want
     * @param id New device ID to set.
     * @see getDeviceID()
     * @see MPU6050_RA_WHO_AM_I
     * @see MPU6050_WHO_AM_I_BIT
     * @see MPU6050_WHO_AM_I_LENGTH
     */
    void setCustomDeviceID(uint8_t id)  {
        deviceID = id;
    };

    /** Get Device ID original from I2C.
     * This register is used to verify the identity of the device (0b110100, 0x34).
     * @return Device ID (6 bits only! should be 0x34)
     * @see getDeviceID()
     * @see MPU6050_RA_WHO_AM_I
     * @see MPU6050_WHO_AM_I_BIT
     * @see MPU6050_WHO_AM_I_LENGTH
     */
    uint8_t getDeviceIDOriginal()  {
        return MPU6050::getDeviceID();
    };

    protected:
        uint8_t deviceID = 0x0;
};

#endif //MPU6050_OVERRIDE_DEVICEID_H

```

And finaly:
1. just include firstly the class you want (standard or MotionApps)
```c++
// =======================================================
// Uncomment the header to match the type MPU6050 you want (before Override class)
// =======================================================
//#include "MPU6050.h" // not necessary if using MotionApps include file
#include "MPU6050_6Axis_MotionApps20.h"
//#include "MPU6050_6Axis_MotionApps612.h"
//#include "MPU6050_9Axis_MotionApps41.h"

#include "MPU6050_override_deviceID.h"
```
2. init mpu: instead of `MPU6050 mpu;` write `MPU6050_override_deviceID mpu;`
3. then call `mpu.setCustomDeviceID(0x34);` with the desired value before `mpu.initialize();`

So, if you can change the lib code of the MPU6050.h, it could be a great work arround for everyone who has problem with this deviceID and acceleration offset calibration...
Thank you in advance.