jrowberg/i2cdevlib #521: Communication problems using MPR121

mhusinsky, 2020-03-08:
hey there,

when trying to use an MPR121 (using the [Adafruit Breakout Board](https://www.adafruit.com/product/1982)) i failed using the code provided.
after some hours of debugging and comparing the code with the [library provided by adafruit](https://github.com/adafruit/Adafruit_MPR121) i found out that the culprit is the way I2C data is requested in I2Cdev.cpp

Changing this line:
https://github.com/jrowberg/i2cdevlib/blob/c08960d03ab36ef0cb07c618926bf9553d78157a/Arduino/I2Cdev/I2Cdev.cpp#L236

to 

`Wire.endTransmission(false);`

makes it work though. 

i'm not sure if this change has implications for other devices but it is necessary to make the MPR121 work.

mhusinsky, 2020-03-08:
i forgot to mention that this happend to me on an ESP32 based board

AshwiniCDAC, 2021-02-03:
Can you share your code here ? As the same setup is working for me for ESP32