jrowberg/i2cdevlib #529: Arduino 1.8.12 won't include current I2Cdev library

JayReynoldsFreeman, 2020-04-11:
I am running Arduino 1.8.12 on a rather old Mac Pro (tower machine) with OS X 10.11.6. I downloaded and unzipped i2cdevlib-master, then used "add .zip library" to get the I2Cdev folder into my Arduino libraries folder. Then I tried to use "manage libraries" to include the library in my sketch, but it did not come up in the list of available libraries. This failure is repeatable. even after quitting and restarting the Arduino app, and also after deleting and reinstalling the I2Cdev folder in my Arduino libraries folder.

The same steps applied to the MPU6050 library resulted in a successful include.

As a sanity check, I cloned the repository for i2cdevlib and used diff -r to compare the files in the cloned I2Cdev directory with the ones in my Arduino library. They were the same. I also checked permissions on the files in my Arduino library and they were unremarkable.

jrowberg, 2020-04-17:
The repository structure as-is is not meant for a direct full import into an Arduino libraries folder. You have to copy/move only the contents of `/i2cdevlib/Arduino` into your libraries folder or project path. I just realized that the [instructions on the website](https://www.i2cdevlib.com/usage) were incorrect in this regard and have been for...a decade? So I fixed them. 🤦

Adding a .zip library directly using the master .zip file within the IDE won't work, for the same reason. Yes, there are many, many ways this collection of libraries [could be improved](https://community.perilib.io/t/perilib-i2cdevlib-reborn/15)...

msaine, 2020-04-19:
Jeff,

Would it make more sense to have two libraries, one for 8 bit CPU’s and one for 32 bit CPU’S. It seems from these discussions that there are considerable work arounds to support both with a single library.  Is there a way to make two libraries that possibly use common routines (or templates) to some extent from a third common group that would cut down on the complications? It seems to me that maintaining two libraries is a lot more work but maybe the 8 bit one would be a more basic system? I don’t know the mechanics of maintaining such a system so I may be way off base. Just a thought.

Monty

playrobotics-alex, 2020-05-11:
Hi Jeff,

Arduino library manager now require library.properties file in order to import the library.
They explain the change here:
https://arduino.github.io/arduino-cli/library-specification/

I am using Arduino Create, but I guess same problem will happen in new version of the Arduino IDE.
I just created such file locally so I can import the library, but please if possible just add such file to the folder, I am sure it will help many people...

Thank you!

jrowberg, 2021-09-26:
Closing this issue as it’s impossible to do correctly with a single repository containing all core libraries and device libraries for multiple platforms. If this changes, we’ll try to correct the properties file for all new sub repos.

closed 2021-09-26 by jrowberg