jrowberg/i2cdevlib #541: MPU6050 change port to ESP32

pantalok, 2020-07-21:
Thankyou for sharing your work.

I am a new user and experimenting with MPU6050 on ESP32 using Arduino IDE. I attempted to use the sample MPU6050_DMP.ino to experiment with Pich,row, Yaw angle however not able to compile this example

It seems your changes in MPU6050.cpp in Arduino folder are not ported over to ESP32 folder and missing these two class definition
- dmpGetCurrentFIFOPacket()
- PrintActiveOffsets ()

I tried to cut/paste the code and I am not much familiar with the code organization but it appears there are deeper dependencies in other i2c macro such as BUFFER_SIZE etc that are absent in ESP32 definition

Can you take a look when you get a chance to port the MPU6050_DMP.ino for ESP32. Thanks again

jrowberg, 2021-09-26:
If you are using the Arduino IDE and an ESP32 board, these problems should now be resolved.

closed 2021-09-26 by jrowberg

krasney, 2022-01-01:
Perhaps I'm making a mistake somewhere, but I don't see these added in ESP32 and am having the same issue.