jrowberg/i2cdevlib #542: Support for ICM20X48

shreyasgokhale, 2020-07-24:
According to TDK/ Mouser MPU6050 is [no longer recommend](https://eu.mouser.com/ProductDetail/TDK-InvenSense/MPU-6050?qs=u4fy%2FsgLU9O14B5JgyQFvg==) for new designs. The ICM20648 and ICM 20689 are successors for MPU6050, which also have DMP and similar structure. 

So far I am only able to find [source code](https://invensense.tdk.com/developers/login/?message=software-downloads&redirect_to=developers/software-downloads/) for embedded MotionDriver which works successfully with STM32 and uses DMP. But I couldn't make it work with this library for ESP32. Are there any plans to upgrade this library to support these IMUs or using a native library + wrapper for different microcontrollers like #512 ?

jrowberg, 2020-07-26:
I don't have any plans at this point to add more devices (or port existing ones) to I2Cdevlib, due to the extremely limiting structure of the core library and its lack of portability. Some time ago, I started working on a restructured implementation of the core library which addresses as many inherent problems as I could identify:

https://community.perilib.io/t/perilib-i2cdevlib-reborn/15

You can see the set of public APIs for register-based operations here:

https://github.com/perilib/perilib-arduino-core/blob/master/src/RegisterInterface.h

...which are implemented on the platform of choice with exactly two HAL-specific read and write methods, which are really almost the only portion of code that would have to be ported to different platforms:

https://github.com/perilib/perilib-arduino-core/blob/master/src/hal/TwiRegisterInterface_ArduinoWire.cpp

Notice lines 26-38 which solve the issue of varying internal TWI/Wire buffer lengths across multiple platforms (AVR, SAMD, ESP32, etc.) which have plagued many with the original libraries. And here's an example showing a variety of these APIs in use:

https://github.com/perilib/perilib-arduino-core/blob/master/examples/Register_TWI_read_write/Register_TWI_read_write.ino

And finally an example of direct communication to the MPU-6050 with the new core library, demonstrating its simplicity and power:

https://github.com/perilib/perilib-arduino-core/blob/master/examples/Register_TWI_MPU6050_direct/Register_TWI_MPU6050_direct.ino

I haven't got as far as building enough device-specific code to have incorporated the embedded MotionDriver for the new ICM lines, since I've been focused on the core architecture and my day job and side projects have made it difficult to find time for more. But this is definitely the direction new development should go for maximum benefit.

anilgit90, 2021-06-16:
Hi @shreyasgokhale @jrowberg, I can take this up as I am currently working on ICM-20689. @jrowberg shall I follow the same documentation that you have set up for MPU6050, Or do you have any specific contribution guidelines that I can follow.

jrowberg, 2021-06-19:
Hi @anilgit90, thanks for volunteering. I don't have any official guidelines anywhere, so following the conventions that you see in existing code/docs is the best plan.

closed 2021-09-26 by jrowberg