jrowberg/i2cdevlib #554: MPU6050.h file compatibility with SAMD

jaggzh, 2020-11-11:
I put a fix into my own project for this, but need some input for public use:

With a Feather M0, dtostrf.h is not included, and MPU6050.c also relies on a "BUFFER_LENGTH" defined.

I went ahead and added these lines to MPU6050.h, just based on stuff I found on the web:

```
#ifdef ARDUINO_SAMD_ZERO // For Feather M0, et. al
#include <avr/dtostrf.h>
#define BUFFER_LENGTH 64  // For ?
#endif

```

ZHomeSlice, 2020-11-11:
dtostrf.h is for my macro to print formatted floating-point numbers with proper spacing 

Line 3349 of MPU6050.cpp
#define printfloatx(Name,Variable,Spaces,Precision,EndTxt) { Serial.print(F(Name)); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(dtostrf((float)Variable,Spaces,Precision ,S));}Serial.print(F(EndTxt)); }//Name,Variable,Spaces,Precision,EndTxt

usage:
float Temp = 38.64275;
float OtherTemp = 112.444;
printfloatx("Temperature is ",Temp ,7,2," Deg C");
printfloatx("Temperature is ",OtherTemp ,7,2," Deg C");
would echo.
"Temperature is    38.64 Dec C"
"Temperature is  112.44 Dec C"

decimal points would be aligned.

------ 
#define BUFFER_LENGTH 64
or 
#define BUFFER_LENGTH 32 for UNO
these represent the ideal buffer size for your i2c bus. matching the i2c bus buffer size will allow for fewer calls to i2c bus helps to empty it quickly. 
have this match whatever you processors i2c bus uses.

Z

jrowberg, 2021-09-23:
Although it makes the debug output formatting slightly worse, I've recently removed the dependency on `dtostrf()` for better cross-platform compatibility, and I improved the BUFFER_LENGTH code to be much safer with a set of compiler macros in `I2Cdev.h`. These problems should be resolved now.

closed 2021-09-23 by jrowberg