jrowberg/i2cdevlib #557: Using FIFO without DMP?

piyushkansalongit, 2020-12-11:
I have a MPU6050 which i am using with an Arduino nano. I am interested in reading the raw values instead of the DMP processed values but also would like to use FIFO for burst reads. Is there any suitable example that anyone could point me to?

ZHomeSlice, 2020-12-12:
FIFO is every 10 ms slower than raw data
If you want to see raw sensor readings I was able to sample new readings every 1 ms

a simple function that may work for you (Example only this has not been tested):

```
uint16_t ReadingsPerSecond() {  // returns the number of readings retrieved in a single second. all 6 readings are gathered.
  uint16_t Counter;
  int16_t  GyroData[3];
  int16_t  AccelData[3];
  GyroReadAddress = 0x43;
  AccesReadAddress = 0x3B;
  unsigned long SecondTimer;
  SecondTimer = millis();
  while ((millis() - SecondTimer) <= (1000)) {
    Counter++
    mpu.readWords(devAddr, GyroReadAddress , 3, (uint16_t *)&GyroData); // reads 3  16 bit integers (Words) from the Gyro raw data Registers
    mpu.readWords(devAddr, AccesReadAddress , 3, (uint16_t *)&AccelData); // reads 3  16 bit integers (Words) from the Accel raw data Registers

    // do something with the data (spamming this to the serial port is slow)
    
    delay(1); // reading faster than 1 ms may give duplicate readings
  }
  return Counter;
}
```

Z

piyushkansalongit, 2020-12-12:
@ZHomeSlice Thank you for your comment.

Just to make sure I understand you, Do you suggest to not use FIFO if I don't intend to use DMP for it doesn't provide the stated benefits of burst reads?

closed 2020-12-17 by piyushkansalongit