jrowberg/i2cdevlib #559: MotionApps_V6_12.h DMP interrupts are wack

thijses, 2020-12-15:
i took a look at the example sketch
MPU6050_DMP6_using_DMP_V6.12 
and i tested some for myself, and i've noticed that the new motionapps implementation has broken interrupts:
the DMP interrupt triggers twice per packet, and at only one of them the FIFO is actually filled to packetSize, at the other (superfluous) interrupt, the FIFOcount is 0, sometimes the MPU6050_INTERRUPT_DATA_RDY_BIT is high, but not always i think.
I dont know how the DMP program works, and from what i understood about dmpInitialize(), nothing is obviously wrong as far as i can see, so the flaw must be in the DMP program(?). It is of course possible that this is by design, but i find this very odd; reading out any byte (like interrupt status, or FIFOcount) takes a significant amount of time (200-350us on esp32) which i wouldnt have to spend if the interrupt just triggered only when there's data available, but in MotionApps V6.12, the smaller packet size saves time on that FIFOread (from 1020us to 780us), so it's a tradeoff.
What could you tell me about this, if anything?

JuMalIO, 2020-12-20:
I have the same issue. Why do we get 2 interrupts?
[image]

App sets only MPU6050_INTERRUPT_DMP_INT_BIT
Why do we gat 2 interupts on 0 and 28 FIFO count? How to get only one Interrupt on full 28 FIFO?

ZHomeSlice, 2020-12-20:
This is something I've struggled with for hours. The interrupt triggers twice for every FIFO packet that is created.
I've tried every combination to remote this. It is set with one byte each not activated an interrupt trigger. It seems that data ready is triggered and fifo packet ready is triggered.  Data ready is related to other raw data not associated with fifo packets. I believe that fifo data consists of two data ready reads per every one packet which are merged together to get an accurate stable output in the fifo buffer. If you or someone finds a way to only get 1 interrupt trigger this would be amazing. 

Z

thijses, 2020-12-20:
it seems like the regular interrupt control byte is set to disable data-ready interrupts, and i think that motionApps V2 did not have this issue, so i imagine the DMP 'code' (mysterious large binary) program (asuming that's what it is!) that is transferred over might be the issue; is there any documentation on that by chance? (i have some free time to puzzle it over)

ZHomeSlice, 2020-12-20:
yes there is:
https://invensense.tdk.com/developers/login/
After getting registered you can download the official 6.12 eDMP library you may also want to get any documentation for the 5.1.3 version of eDMP as it may contain additional documentation

inside the zip files
motion_driver-6.12.zip
Documents folder
App Note 1 - Motion Driver 6.12 Getting Started.pdf
App Note 2- Motion Driver 6.12 Features Guide.pdf
App Note 3- Motion Driver 6.12 Porting Guide.pdf
MPU HW Offset Registers 1.2.pdf
Orientation Matrix Transformation chart.pdf

and
motion_driver-5.1.3.zip
Orientation Matrix Transformation chart.pdf
eMD-5_1_1 release note.pdf
MotionDriver_Tutorial_12212018 CB.pdf

among other pdf documents and example code.

you may find these files elsewhere online by googling the file names :)
Z

JuMalIO, 2020-12-23:
I understood why its happening. Its because sample rate is set to 1, so it skips one sample. If you change sample rate to n you will get n interrupts with 0 data and the last one will be with data. But how do we get rid of those useless interrupts?

ZHomeSlice, 2020-12-23:
I would assume this is possible. but with all my experimentation and completely re-writing the entire function using macros I was not able to achieve 1 sample 1 interrupt.

My code and Jeff's code are equal if function. I use macros to access every register in the mpu. I also use lambda functions to simplify (or make it more complex... lol) 
My version of the MPU6050 library.
https://github.com/ZHomeSlice/Simple_MPU6050

as you can see it is re-written in such a way as to totally go over each and every attribute in detail. Each macro is focused on a single register to the bit level.

I will assist but in order for me to help better we will need to test with my simple_mpu library 

--------------------
using simple_mpu6050

the macros for the interrupts are as follows
```

#define INT_PIN_CFG_WRITE_ACTL(Data)                    MPUi2cWrite(0x37, 1, 7, (uint8_t)Data)		//   1  The logic level for int16_t pin is active low. 0  The logic level for int16_t pin is active high.
#define INT_PIN_CFG_WRITE_OPEN(Data)                    MPUi2cWrite(0x37, 1, 6, (uint8_t)Data)		//   1  int16_t pin is configured as open drain. 0  int16_t pin is configured as push-pull.
#define INT_PIN_CFG_WRITE_LATCH_INT_EN(Data)            MPUi2cWrite(0x37, 1, 5, (uint8_t)Data)		//   1  int16_t pin level held until interrupt status is cleared. 0  int16_t pin indicates interrupt pulses is width 50us.
#define INT_PIN_CFG_WRITE_INT_ANYRD_2CLEAR(Data)        MPUi2cWrite(0x37, 1, 4, (uint8_t)Data)		//   1  Interrupt status is cleared if any read operation is performed. 0  Interrupt status is cleared only by reading INT_STATUS register
#define INT_PIN_CFG_WRITE_ACTL_FSYNC(Data)              MPUi2cWrite(0x37, 1, 3, (uint8_t)Data)		//   1  The logic level for the FSYNC pin as an interrupt is active low. 0  The logic level for the FSYNC pin as an interrupt is active high.
#define INT_PIN_CFG_WRITE_FSYNC_INT_MODE_EN(Data)       MPUi2cWrite(0x37, 1, 2, (uint8_t)Data)		//   1  This enables the FSYNC pin to be used as an interrupt.
#define INT_PIN_CFG_WRITE_BYPASS_EN(Data)               MPUi2cWrite(0x37, 1, 1, (uint8_t)Data)		//   When asserted, the i2c_master interface pins(ES_CL and ES_DA) will go into bypass mode when the i2c master interface is disabled.

#define INT_ENABLE_WRITE_FF_EN(Data)                    MPUi2cWrite(0x38, 1, 7, (uint8_t)Data)		//   1  Enable interrupt for
#define INT_ENABLE_WRITE_WOM_EN(Data)                   MPUi2cWrite(0x38, 1, 6, (uint8_t)Data)		//   1  Enable interrupt for wake on motion to propagate to interrupt pin. 0  function is disabled.
#define INT_ENABLE_WRITE_ZMOT_OFLOW_EN(Data)            MPUi2cWrite(0x38, 1, 5, (uint8_t)Data)		//   1  Enable interrupt for
#define INT_ENABLE_WRITE_FIFO_OFLOW_EN(Data)            MPUi2cWrite(0x38, 1, 4, (uint8_t)Data)		//   1  Enable interrupt for fifo overflow to propagate to interrupt pin. 0  function is disabled.
#define INT_ENABLE_WRITE_FSYNC_INT_EN(Data)             MPUi2cWrite(0x38, 1, 3, (uint8_t)Data)		//   1  Enable (I2C_MST_INT_BIT) Fsync interrupt to propagate to interrupt pin. 0  function is disabled.
#define INT_ENABLE_WRITE_RAW_PLL_RDY_INT_EN(Data)       MPUi2cWrite(0x38, 1, 2, (uint8_t)Data)		//   1  Enable
#define INT_ENABLE_WRITE_RAW_DMP_INT_EN(Data)           MPUi2cWrite(0x38, 1, 1, (uint8_t)Data)		//   1  Enable DMP interrupt
#define INT_ENABLE_WRITE_RAW_RDY_EN(Data)               MPUi2cWrite(0x38, 1, 0, (uint8_t)Data)		//   1  Enable Raw Sensor Data Ready interrupt to propagate to interrupt pin.  0  function is disabled.

```
My default setup which is the same as Jeffs

```
MPUi2cWriteByte(0x37, 0x32);				// 0010 0010 INT_PIN_CFG: ACTL The logic level for int pin is active low. and interrupt status bits are cleared on any read
MPUi2cWriteByte(0x38, 0x02);				// 0000 0010 INT_ENABLE: RAW_DMP_INT_EN on
```

Hope this helps

Z

JuMalIO, 2020-12-23:
To achieve 1 sample 1 interrupt you need to set MPU6050_DMP_FIFO_RATE_DIVISOR to zero. But for my use case I dont want to get that much data, so I use MPU6050_DMP_FIFO_RATE_DIVISOR 32 which means I will get 32 interrupts with zero data and 33rd interrupt will be with data.

I wrote this and it seems to work:

    static volatile uint8_t SampleCount = MPU6050_DMP_FIFO_RATE_DIVISOR;
    uint16_t MPU6050_OnDmpInterupt(uint8_t* fifoBuffer)
    {
    	if (SampleCount >= MPU6050_DMP_FIFO_RATE_DIVISOR)
    	{
    		uint16_t count;
    		ReadWord(MPU6050_ADDRESS, MPU6050_RA_FIFO_COUNTH, &count);
    		if (count == MPU6050_DMP_PACKET_SIZE)
    		{
    			I2C_Mem_Read_DMA(&hi2c1, MPU6050_ADDRESS, MPU6050_RA_FIFO_R_W, 1, fifoBuffer, MPU6050_DMP_PACKET_SIZE);
    			SampleCount = 0;
    		}
    		else if (count > MPU6050_DMP_PACKET_SIZE)
    		{
    			WriteByte(MPU6050_ADDRESS, MPU6050_RA_USER_CTRL, 0xC4);
    			SampleCount = 0;
    		}
    	}
    	else
    	{
    		SampleCount++;
    	}
    }

thijses, 2020-12-23:
then i think this issue can be closed, right?
in my own tests i was getting somewhat inconsistent interrupts, but it was (shitty) testcode, so if you agree that the number of interrupts is consistent, unwavering and just needs to be counted, then (slap this into the comments of the DMP sketch) and call it a day :)
as for me, i think i'll take the road of much data and set the divisor to 0, because a few milliseconds between measurements is all the ESP32 needs to do the rudimentary filtering i need, plus there's always the second CPU core.
thank you for your help, although i still dont know (will read documentation soon) why the old MotionApps didnt have this 'issue' (my current understanding is that this is a feature with a slight flaw, but a feature nonetheless, to reduce the datapoints sent to the main MCU without reducing the datapoints sent to the DMP (complete guess))

ZHomeSlice, 2020-12-24:
try something like:

#define MPU6050_DMP_FIFO_RATE_DIVISOR 0x01 // The New instance of the Firmware has this as the default
in setup as usual
`attachInterrupt(digitalPinToInterrupt(INTERRUPT_PIN), dmpDataReady, RISING);`

```
// ================================================================
// ===               INTERRUPT DETECTION ROUTINE                ===
// ================================================================

volatile int mpuInterrupt = MPU6050_DMP_FIFO_RATE_DIVISOR +1;     // Counts up from 2 until we get a good read then is set to zero
void dmpDataReady() {
  mpuInterrupt ++;
}

// ================================================================
// ===                    MAIN PROGRAM LOOP                     ===
// ================================================================
void loop() {
  // read a packet from FIFO
// this will heavily read the i2c buss until it detects a packet then only read every 2 interrupts exactly when new data arrives.
  if ((mpuInterrupt  > MPU6050_DMP_FIFO_RATE_DIVISOR) &&  mpu.dmpGetCurrentFIFOPacket(fifoBuffer)) { // Get the Latest packet 
    mpuInterrupt = 0 ;  // We got a good read now set this to zero so we can skip an interrupt

// got a good packet so do something with it.
  }
}

```

Z

thijses, 2020-12-24:
(i'd change 
`mpuInterrupt = 0`
to be
`mpuInterrupt -= (MPU6050_DMP_FIFO_RATE_DIVISOR+1)`
or something, because if the rate is very high, the interrupt counter might go not 1 but 2 above MPU6050_DMP_FIFO_RATE_DIVISOR before the main loop has time to handle it

jrowberg, 2021-09-25:
Closing old issues for this repo to clean up. Comment if further discussion is required.

closed 2021-09-25 by jrowberg

pstumbler, 2022-03-12:
I found a solution for this issue described directly in the source code from InvenSense motion_driver_6.12:
in file main.c: 
```
/*
 * Known Bug -
 * DMP when enabled will sample sensor data at 200Hz and output to FIFO at the rate
 * specified in the dmp_set_fifo_rate API. The DMP will then sent an interrupt once
 * a sample has been put into the FIFO. Therefore if the dmp_set_fifo_rate is at 25Hz
 * there will be a 25Hz interrupt from the MPU device.
 *
 * There is a known issue in which if you do not enable DMP_FEATURE_TAP
 * then the interrupts will be at 200Hz even if fifo rate
 * is set at a different rate. To avoid this issue include the DMP_FEATURE_TAP
 *
 * DMP sensor fusion works only with gyro at +-2000dps and accel +-2G
 */
```

and in file inv_mpu_dmp_motion_driver.c: 
```
/* Send gesture data to the FIFO. */
if (mask & (DMP_FEATURE_TAP | DMP_FEATURE_ANDROID_ORIENT))
    tmp[0] = 0x20;
else
    tmp[0] = 0xD8;
mpu_write_mem(CFG_27,1,tmp);  // #define CFG_27 (2742)

```
After writing the `20h` to offset `2742` in DMP memory, the interrupt follows exactly the fifo-rate!

If you work with real hardware interrupts, this reduces the interrupt load a lot...

thijses, 2022-03-12:
hot damn, very nice! thank you

ZHomeSlice, 2022-03-12:
Fixed it directly on the eDMP Image.
see
https://github.com/jrowberg/i2cdevlib/pull/674
#674 opened 44 minutes ago by [ZHomeSlice](https://github.com/jrowberg/i2cdevlib/issues?q=is%3Apr+is%3Aopen+author%3AZHomeSlice)

jrowberg, 2022-03-13:
I love this community.

pstumbler, 2022-03-14:
My bad - I only looked at the interrupts... :-(
If you read a bit ahead in the file `inv_mpu_dmp_motion_driver.c`, you see the following:

```
if (mask & (DMP_FEATURE_TAP | DMP_FEATURE_ANDROID_ORIENT))
    dmp.packet_length += 4;
```
This implies the following changes in the code to increase the buffer to 32 bytes:
```
/* ================================================================ *
 | Default MotionApps v6.12 32-byte FIFO packet structure:          |
 |                                                                  |
 | [QUAT W][      ][QUAT X][      ][QUAT Y][      ][QUAT Z][      ] |
 |   0   1   2   3   4   5   6   7   8   9  10  11  12  13  14  15  |
 |                                                                  |
 | [GYRO X][GYRO Y][GYRO Z][ACC X ][ACC Y ][ACC Z ] [TAP0 ... TAP4] |
 |  16  17  18  19  20  21  22  23  24  25  26  27  28  29  30  31  |
 * ================================================================ */
```
and 
```
/*
    dmpPacketSize += 16;//DMP_FEATURE_6X_LP_QUAT
    dmpPacketSize += 6;//DMP_FEATURE_SEND_RAW_ACCEL
    dmpPacketSize += 6;//DMP_FEATURE_SEND_RAW_GYRO
    dmpPacketSize += 4;//DMP_FEATURE_TAP
*/
    dmpPacketSize = 32;
```
To help with the `MPU6050_DMP_FIFO_RATE_DIVISOR` which is fixed in the moment to 0x01, I put something like this before the memory dmp:
```
#define DMP_DIV_LOW    0x09  // calculation is: uint16_t div = (200 / rate(Hz)) - 1; ==> e.g. rate = 20Hz ==> div = 0x09)
#define DMP_DIV_HIGH   0x00  // high byte of rate
```
and the memory dump is changed in bank 2 to set the FIFO rate before dmp download and don't need to be altered afterwards:
```
/* bank # 2 */
0x00, 0x28, 0x00, 0x00, 0xFF, 0xFF, 0x45, 0x81, 0xFF, 0xFF, 0xFA, 0x72, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x44, DMP_DIV_HIGH, DMP_DIV_LOW, 0x00, 0x05, 0xBA, 0xC6, 0x00, 0x47, 0x78, 0xA2,

```

pstumbler, 2022-03-14:
@ZHomeSlice 
Yes, but the proper functions for changing the FIFO rate in the DMP memory are missing in I2Cdevlib
A nice function like you have it in your Simple_MPU6050 code (`Set_DMP_Output_Rate_Hz(float rate)`) would be good ...

ZHomeSlice, 2022-03-15:
I thought I brought that over... 
I didn't test this I just quickly rough drafted the additions to jeffs code. This should give you an idea of what to do... and it should work. I'll look for your responses and shortly set up a Pull request with the changes to Jeffs's library after I test them... this may be a week out.

Here's a start brought from my Simple_MPU6050 version ported to Jeffs:

Add to MPU6050.h 
near top

`#define D_0_22    (22+512) // eDMP memory address V6.12 only -- Modify the Firmware Chunk for DMP output Rate `

add about line 817

`bool writeDMPMemory((uint8_t *data, uint16_t  address, uint16_t  dataSize);`

before private: at line 145
 
```
uint8_t DMP_Output_Rate[2] = {0x00,0x01}; 
void Set_DMP_Output_Rate(uint16_t value = 0x01); // 100Hz Default
void Set_DMP_Output_Rate_Hz(float rate = 100); // 100Hz Default
void Set_DMP_Output_Rate_Seconds(float rate = 1); // 1Hz Default
void Set_DMP_Output_Rate_Minutes(float rate = 1); // 1 minute Default

```
Add to MPU6050.cpp 
add about line 3099

```
bool MPU6050_Base::writeDMPMemory((uint8_t *data, uint16_t  address, uint16_t  dataSize) {
        setMemoryStartAddress(address); // BANK_SEL_WRITE(mem_addr);
        I2Cdev::writeBytes(devAddr, MPU6050_RA_MEM_R_W, dataSize, data, wireObj); //DMP_MEM_WRITE(length, data);
	return true;
}
```

 at the end
```
MPU6050_Base::Set_DMP_Output_Rate(uint16_t value){  // 100 HZ Default
	DMP_Output_Rate[0] = (uint8_t)((value & 0xFF00) >> 8);
	DMP_Output_Rate[1] = (uint8_t)(value & 0x00FF);
	
}
MPU6050_Base::Set_DMP_Output_Rate_Hz(float rate){  // 100 HZ Default

    rate = (rate>200) ? 200 : rate;
    rate = (rate<0) ? 0.0032 : rate;
	uint16_t div = (uint16_t)( 200 / rate - 1);
	Set_DMP_Output_Rate(div);
	
}
MPU6050_Base::Set_DMP_Output_Rate_Seconds(float rate){  // 1Hz Default
    rate = (rate>300) ? 300 : rate;
    rate = (rate<0) ? 0 : rate;
	uint16_t div = (uint16_t)( 200 / (1/rate) - 1);
	Set_DMP_Output_Rate(div);

}
MPU6050_Base::Set_DMP_Output_Rate_Minutes(float rate){  // 1 minute Default
    rate = (rate>5) ? 5 : rate;
    rate = (rate<0) ? 0 : rate;
	Set_DMP_Output_Rate_Seconds(rate/60);
	
}
```

Add to MPU6050_6Axis_MotionApps612.cpp
add right after line 365 "if (!writeProgMemoryBlock....."

`writeDMPMemory(DMP_Output_Rate, D_0_22, 2);      // Modify the Firmware Chunk for DMP output Rate  `

Now right before `devStatus = mpu.dmpInitialize();` in your program, you need to run one of the above functions to set the DMP Rate 

Z

pstumbler, 2022-03-15:
Thanks a lot.
But the point is, that without changing the fifo length to  `dmp.packet_length = 32` or reverting to the previous version the V6.12 in the repository is not working.

ZHomeSlice, 2022-03-15:
look for 
`0xA3, 0xF1, 0xB0, 0x87, 0xB5, 0x9A, 0x20, 0xF3, 0x9B, 0xA3, 0xA3, 0xDC, 0xBA, 0xAC, 0xDF, 0xB9, //change 2742 from 0xD8 to 0x20 Including the DMP_FEATURE_TAP -- known issue in which if you do not enable DMP_FEATURE_TAP then the interrupts will be at 200Hz even if fifo rate`

The 7th byte in is what changed from 0x20 to 0xD8
change it back to:                             
`0xA3, 0xF1, 0xB0, 0x87, 0xB5, 0x9A, 0xD8, 0xF3, 0x9B, 0xA3, 0xA3, 0xDC, 0xBA, 0xAC, 0xDF, 0xB9, //change 2742 from 0xD8 to 0x20 Including the DMP_FEATURE_TAP -- known issue in which if you do not enable DMP_FEATURE_TAP then the interrupts will be at 200Hz even if fifo rate`

This will restore functionality
I'll adjust the FIFO packet size to work with the additional bytes and I will also resolve other things 

Thanks for the catch

ZHomeSlice, 2022-03-15:
I also make a pull request to revert back until I can dig into this further

ZHomeSlice, 2022-03-15:
you could also try increasing the dmp.packet_length from  32  to 36

if (mask & (DMP_FEATURE_TAP | DMP_FEATURE_ANDROID_ORIENT))
        dmp.packet_length += 4;

dmp.packet_length = 36

This could be a solution.