jrowberg/i2cdevlib #564: MPU6050 and Arduino: always freezes after a while, ok with Electronic Cats fork

candardo, 2021-01-06:
I spent some time trying to get my MPU-6050 breakout board to work (GY-521). Raw data is fine but I need ypr angles and I want to take advantage of the DMP onboard so I tested i2cdevlib libraries and included examples. It works pretty well and data seem stable and correct but all the examples freeze after a while (some seconds to a few minutes).
I read some FAQ, forums and issues but nothing fixed the problem:
* https://www.i2cdevlib.com/forums/topic/27-fifo-overflow/ FIFO rate divider
* https://www.i2cdevlib.com/forums/topic/414-freezing-problem/ AD0 to GND
* https://github.com/jrowberg/i2cdevlib/issues/556 Fastwire (compile error here)

As a last resort a tried the [library ](https://github.com/ElectronicCats/mpu6050)from Electronic Cats. I didn't expect much since it's still based on i2cdevlib (AFAIK) and I expected it to be less maintained. Anyway I installed it via the Arduino IDE library manager and flashed the example MPU6050_DMP6.ino. With that version no freezes occur, and I tested it for hours! 
Nonetheless MPU6050_DMP6_using_DMP_V6v12.ino from that library still fails.

Is there some sort of regression in the code? Are there solutions I couldn't find? Is it an hardware related problem?

sylvanoMTL, 2021-01-10:
Are you using the interrupt pin? if so, do you have the pin interrupt linked to the correct GPIO of your board?

candardo, 2021-01-10:
I am using the interrupt on pin 2 of an Arduino Uno board. As I said the code does work but it freezes after a while.

msaine, 2021-01-10:
Has anyone done a serious check for a memory leak? This seems to me to be an ongoing problem, running for a while then freezing.
A ISR that doesn’t clean up after itself or a Malloc not matched with an associated free?? This running for a while then quitting report has been running thru this thread for a couple of years it seems.

Just my 2cents worth

orhanyor, 2021-01-10:
i had the same freeze issue, i could not run the MPU6050 for more than 2 minutes. somewhere i saw a little detail which fixed the freezing issues. i just tied AD0 from the module to ground and it shouldnt be left floating. that was causing 6050 to freeze. can you try that if you left AD0 floating?

orhanyor, 2021-01-10:
have you tried this https://github.com/jrowberg/i2cdevlib/issues/519#issuecomment-752023021

trademark18, 2021-01-16:
@orhanyor That worked for me.