jrowberg/i2cdevlib #569: how to change the unit of real world acceleration and gyro degree

nices0325, 2021-02-11:
[image]

this is my serial monitor shows the  acceleration values divided by 8192.0 but I don't think this is correct value... cause the Z value goes up to 3 if i rotate 90 degree . It should be x or y value should be 1 otherwise Z value should be 0..

does anybody know how to change the unit to g(m/s^2)
thank you for reading and sorry for my bad english

ZHomeSlice, 2021-02-11:
OP "acceleration values divided by 8192.0" 
why is this? the output of the accelerometer is a value between 0 and 65353 or 1111 1111 1111 1111B this is an unsigned integer with 16 bits and with 1/2 of 65353 or 32767 is always 0g no matter what the scale is. once it arrives we need to manipulate it to get our acceleration. first of all, we can make it a signed integer ±32767.  We now need to scale it to match the sensitivity setting of the MPU  ±2g, ±4g, ±8g, and ±16g. 
±2g,  = (4) we need to use 2 bits 11b for the whole number and the remaining 14 for precision so divide by  16384 or 11 1111 1111 1111b
±4g,  = (8) we need to use 3 bits  111b for the whole number and the remaining 13 for precision so divide by  8192 or 1 1111 1111 1111b
±8g, = (16) we need to use 4 bits  1111b for the whole number and the remaining 12 for precision so divide by   4096 or  1111 1111 1111b
±16g. = (32) we need to use 5 bits  11111b for the whole number and the remaining 11 for precision so divide by   2048 or 111 1111 1111b

Note
When you divide by one of these numbers 2048, 4096, 8192, 16384 you are moving the decimal point 1111 1111 1111 1111B / 8192 = 0111B = 8 or ±4

To see which number to divide by you will want to look at the values in Register 28 or 1C hex – Accelerometer Configuration
bit 3 and 4 set the scale.
Full Scale Range
00B ± 2g
01B ± 4g
10B ± 8g
11B ± 16g

Z

nices0325, 2021-02-12:
I heard that mpu6050 DMP has a fixed scale.. First of all the raw real world accel (Z) shows close to 8192 value so I devided it by 8192 so that I can get the g_force value. Is it the wrong way?? Actually I'm not a programmer so It would be a bit difficult to understand ... Would you let me know how to change the scale and get a g_force value if you could or if you don't mind
thanks for your reply !! I do appreciate that !

closed 2021-09-26 by jrowberg