jrowberg/i2cdevlib #570: Compiler Error uint8_t Trash[BUFFER_LENGTH] not declared

volkerpetersen, 2021-02-12:
when I compile my program using the library function mpu.dmpGetCurrentFIFOPacket(fifoBuffer) I get the compiler errors:
E:\VolkerPetersen\Dropbox\ProgramCode\MicroProcessorProjects\libraries\MPU6050\src\MPU6050.cpp:2763:32: error: 'BUFFER_LENGTH' was not declared in this scope
                  uint8_t Trash[BUFFER_LENGTH];

and
E:\VolkerPetersen\Dropbox\ProgramCode\MicroProcessorProjects\libraries\MPU6050\src\MPU6050.cpp:2769:39: error: 'Trash' was not declared in this scope
                          getFIFOBytes(Trash, (uint8_t)RemoveBytes);

Any suggestions?

ZHomeSlice, 2021-02-12:
#define BUFFER_LENGTH 32 // I'm assuming your processors i2c buffer is 32 bytes in length

you need this before you include your libraries

Z

volkerpetersen, 2021-02-12:
Thx!

 

Volker

507-837-9868

bill-richards, 2021-04-29:
> #define BUFFER_LENGTH 32 // I'm assuming your processors i2c buffer is 32 bytes in length
> 
> you need this before you include your libraries
> 
> Z

Where is this documented please?

jrowberg, 2021-09-25:
Errors related to undefined `BUFFER_LENGTH` have now been resolved.

closed 2021-09-25 by jrowberg