jrowberg/i2cdevlib #573: MKR ZERO BOARD COMPILE ERROR

nices0325, 2021-02-15:
아두이노:1.8.13 (Windows 10), 보드:"Arduino MKRZERO"

C:\Users\nices\Documents\Arduino\ELMDUINO_SD_TFT_GYRO_NANO\ELMDUINO_SD_TFT_GYRO_NANO.ino: In function 'void setup()':

C:\Users\nices\Documents\Arduino\ELMDUINO_SD_TFT_GYRO_NANO\ELMDUINO_SD_TFT_GYRO_NANO.ino:520:42: warning: 'uint32_t FsVolume::blocksPerCluster()' is deprecated [-Wdeprecated-declarations]

     freeKB *= sd.vol()->blocksPerCluster() / 2;

                                          ^

In file included from C:\Users\nices\Documents\Arduino\libraries\SdFat-master\src/FsLib/FsLib.h:31:0,

                 from C:\Users\nices\Documents\Arduino\libraries\SdFat-master\src/SdFat.h:35,

                 from C:\Users\nices\Documents\Arduino\ELMDUINO_SD_TFT_GYRO_NANO\ELMDUINO_SD_TFT_GYRO_NANO.ino:8:

C:\Users\nices\Documents\Arduino\libraries\SdFat-master\src/FsLib/FsVolume.h:86:12: note: declared here

   uint32_t blocksPerCluster() __attribute__ ((deprecated)) {return sectorsPerCluster();} //NOLINT

            ^~~~~~~~~~~~~~~~

C:\Users\nices\Documents\Arduino\libraries\MPU6050\MPU6050.cpp: In member function 'int8_t MPU6050::GetCurrentFIFOPacket(uint8_t*, uint8_t)':

C:\Users\nices\Documents\Arduino\libraries\MPU6050\MPU6050.cpp:2776:32: error: 'BUFFER_LENGTH' was not declared in this scope

                  uint8_t Trash[BUFFER_LENGTH];

                                ^~~~~~~~~~~~~

C:\Users\nices\Documents\Arduino\libraries\MPU6050\MPU6050.cpp:2776:32: note: suggested alternative: 'REG_DSU_LENGTH'

                  uint8_t Trash[BUFFER_LENGTH];

                                ^~~~~~~~~~~~~

                                REG_DSU_LENGTH

C:\Users\nices\Documents\Arduino\libraries\MPU6050\MPU6050.cpp:2782:39: error: 'Trash' was not declared in this scope

                          getFIFOBytes(Trash, (uint8_t)RemoveBytes);

                                       ^~~~~

exit status 1

This happens to my board... Is there any way to solve this problem?? plz reply thanks....

nices0325, 2021-02-15:
OMG I solved this problem on my own..

I added this code on my mpu6050.cpp file

#ifdef __AVR__
#include <stdlib.h> //samd 타입 ,아두이노 due 경우 추가
#include <avr/dtostrf.h>
#endif

#ifdef __arm__
#include <stdlib.h> //samd 타입 ,아두이노 due 경우 추가
#include <avr/dtostrf.h>
#endif

#if defined(I2C_BUFFER_LENGTH)
// Arduino ESP32 core Wire uses this
#define I2CDEVLIB_WIRE_BUFFER_LENGTH I2C_BUFFER_LENGTH
#elif defined(BUFFER_LENGTH)
// Arduino AVR core Wire and many others use this
#define I2CDEVLIB_WIRE_BUFFER_LENGTH BUFFER_LENGTH
#elif defined(SERIAL_BUFFER_SIZE)
// Arduino SAMD core Wire uses this
#define I2CDEVLIB_WIRE_BUFFER_LENGTH SERIAL_BUFFER_SIZE
#else
// should be a safe fallback, though possibly inefficient
#define I2CDEVLIB_WIRE_BUFFER_LENGTH 32
#endif

#ifndef BUFFER_LENGTH
// band-aid fix for platforms without Wire-defined BUFFER_LENGTH (removed from some official implementations)
#define BUFFER_LENGTH 32
#endif

plz add another codes for the another board...  thank you

jrowberg, 2021-09-25:
These issues have been fixed with a couple of recent commits. Sorry it took so long to address them.

closed 2021-09-25 by jrowberg