jrowberg/i2cdevlib #574: Why it is incompatible with arduino uno r3? and when I tried it with esp32 it is giving me error buffer length is not defined in this scope

Shikhar0708, 2021-02-17:


Shikhar0708, 2021-02-17:
why is it incompatible with arduino unpo r3?

ZHomeSlice, 2021-02-17:
UNO uses 
#define BUFFER_LENGTH 32
esp32 **may** have the same i2c buffer length but the Arduino ide doesn't define this for you you have to.
add the #define line before your includes and it should work.

Shikhar0708, 2021-02-17:
Okay and can you please help me with why i cant compile it for arduino uno

nices0325, 2021-02-18:
> 좋아, 왜 내가 arduino uno를 위해 컴파일 할 수 없는지 도와 주시겠습니까?

all most all of the atmega328 chip included arduino worked well..  redownload and refesh all the files and rerun it

Shikhar0708, 2021-02-18:
would you please have a look at my code?
#include "IMU.h"

#define NUM_SAMPLES 30
#define NUM_AXES 3
// sometimes you may get "spikes" in the readings
// set a sensible value to truncate too large values
#define TRUNCATE_AT 20

double features[NUM_SAMPLES * NUM_AXES];

void setup() {
    Serial.begin(115200);
    imu_setup();
}

void loop() {
    float ax, ay, az;

    imu_read(&ax, &ay, &az);

    ax = constrain(ax, -TRUNCATE_AT, TRUNCATE_AT);
    ay = constrain(ay, -TRUNCATE_AT, TRUNCATE_AT);
    az = constrain(az, -TRUNCATE_AT, TRUNCATE_AT);

    Serial.print(ax);
    Serial.print('\t');
    Serial.print(ay);
    Serial.print('\t');
    Serial.println(az);
}

jrowberg, 2021-09-25:
Errors related to undefined `BUFFER_LENGTH` have now been resolved with an addition to `I2Cdev.h` detecting and/or defining a suitable length for many platforms.

closed 2021-09-25 by jrowberg