jrowberg/i2cdevlib #575: "MPU6050.cpp" Variable Scope

XuhuaHuang, 2021-02-19:
When compiling for Arduino 33 IoT, I get the following error: 
`MPU6050.cpp:3353:154: error: 'dtostrf' was not declared in this scope`

I believe `dtostrf` is a function with parameter list `(float)Variable,Spaces,Precision,S)` and return something printable to the serial monitor? Is the function deleted accidentally?

Please look at this issue and thanks in advance!

ZHomeSlice, 2021-02-19:
The function is part of the Arduino Uno's sounds. It is missing in the ESP 32 ... but I believe can be included.
Try adding
#include <ave/dtostrf.h>

Z

XuhuaHuang, 2021-02-19:
I found the function definition on one of the Arduino's Forum. Function definition is attached as below:
// sorry for the poor format I am still trying to figure out the code block for GitHub comment editing
I also add the function prototype to the class and prepend the function name with `MPU6050::`

Thank you for replying! Have a nice weekend!

`char* MPU6050::dtostrf(double number, signed char width, unsigned char prec, char* s) {

    if (isnan(number)) {
        strcpy(s, "nan");
        return s;
    }
    if (isinf(number)) {
        strcpy(s, "inf");
        return s;
    }
    if (number > 4294967040.0 || number < -4294967040.0) {
        strcpy(s, "ovf");
        return s;
    }
    char* out = s;
    // Handle negative numbers
    if (number < 0.0) {
        *out = '-';
        ++out;
        number = -number;
    }

    // Round correctly so that print(1.999, 2) prints as "2.00"
    double rounding = 0.5;
    for (uint8_t i = 0; i < prec; ++i)
        rounding /= 10.0;

    number += rounding;

    // Extract the integer part of the number and print it
    unsigned long int_part = (unsigned long)number;
    double remainder = number - (double)int_part;
    out += sprintf(out, "%d", int_part);

    // Print the decimal point, but only if there are digits beyond
    if (prec > 0) {
        *out = '.';
        ++out;
    }

    while (prec-- > 0) {
        remainder *= 10.0;
    }
    sprintf(out, "%d", (int)remainder);

    return s;
}`

closed 2021-03-01 by XuhuaHuang