jrowberg/i2cdevlib #578: MPU6050 connection failed

redmlr, 2021-03-23:
Hellou,

I'm trying to use the i2cdev library for my mpu6050, but it does not seem to be working. No matter which example from the mpu6050 I take, it's saying always the same: `MPU6050 connection failed`.  (Sketches involving the library)

I'm using an Arduino Uno and a MPU6050 (GY-521). I connected `VCC to 3.3V`, `GND to GND`, `SCL to A4`, `SDA to A5`. I also ran an ic2-scanner to find the right address which is in my case: `0x68`.
When I try to get values using sketches that don't involve i2cdev, it works. But since I need the offset values, I need to use i2cdev.

flavioipiranga, 2021-09-12:
Hi,

What is the return value of `uint8_t MPU6050::getDeviceID()``?

redmlr, 2021-09-12:
Thank you for answering me @flavioipiranga! I can get you the output value in a couple of hours, I'm currently busy:(