jrowberg/i2cdevlib #579: MPU6050 problem, (ax) -32768 and (ay) 32767 is always at the in maximum  point

Yeremiy, 2021-03-27:
Hello experts. Sorry for bad English. I'm new to arduino. I don't know much, while I'm just trying to correct other people's sketches. There is MPU 6050 (GP-521). The problem is this.
SerialPort:     a / g: -32768 32767 18232 -417 1851 59
 ax -32768 ay 32767 is always at the in maximum point. Calibration does not help, it is calibrated indefinitely. I tried dozens of sketches for output to the port, always the same thing. What is the problem? Is this a physical or software problem? How can this be fixed? Thank.

Hasi123, 2021-04-08:
What values do you get if you do not apply any calibration? Also is the gyro output ok?

Yeremiy, 2021-04-08:
Hello. thanks for the help. calibration takes place on g(xyz) and when it goes to a(xyz) it is calibrated infinitely. But as I already said now, and a(xyz) at high points, I don't know why the gyroscope also malfunctioned.

meneraing, 2021-09-29:
Have you tried setting the offsets to 0 and then get some raw readings? Maybe calibration isn't giving you adequate values.

chevalfromage, 2023-05-01:
Hello, I got the same issue did you find the answers?

ZHomeSlice, 2023-05-01:
The calibration routine uses a PID loop (its only using Proportional and Integral) to tune the Accelerometer. Since we have 1 g here on earth I subtract 1g from one of the a(z) reading to eliminate that offset. The calibration loops once every millisecond an adjusts the offsets so that a(x) and a(y) == 0 and a(z) == 1g or as close to this as possible. when you are getting offsets slammed to the extreme such as -32768 32767 -32768  this implies that the accelerometer is not on the proper plane and that a(z) is not in the vertical position getting the full influence of gravity. The calibration routine is simple each loop it takes a portion of the error from zero (setpoint) and uses this to influence the offsets in the correct direction. since adjusting the offsets provides and instantaneous result we take a new reading 1ms later to adjust it again. as we reach an error of zero the influence applied to the offsets dwindles to nothing. 
with that sail if you are still getting offsets of -32768 32767 -32768 I am concerned that the accelerometer is damaged and not able to reach a balanced state.
Z
P.S. I created this method of calibrating the MPU6050 and provided this portion of the code to Jeffs library.

chevalfromage, 2023-05-01:
If i try to get the datas (with this code)
`    // MPU-6050 Short Example Sketch
    // By Arduino User JohnChi
    // August 17, 2014
    // Public Domain
    #include<Wire.h>
    const int MPU_addr=0x68;  // I2C address of the MPU-6050
    int16_t AcX,AcY,AcZ,Tmp,GyX,GyY,GyZ;
    void setup(){
      Wire.begin();
      Wire.beginTransmission(MPU_addr);
      Wire.write(0x6B);  // PWR_MGMT_1 register
      Wire.write(0);     // set to zero (wakes up the MPU-6050)
      Wire.endTransmission(true);
      Serial.begin(9600);
    }
    void loop(){
      Wire.beginTransmission(MPU_addr);
      Wire.write(0x3B);  // starting with register 0x3B (ACCEL_XOUT_H)
      Wire.endTransmission(false);
      Wire.requestFrom(MPU_addr,14,true);  // request a total of 14 registers
      AcX=Wire.read()<<8|Wire.read();  // 0x3B (ACCEL_XOUT_H) & 0x3C (ACCEL_XOUT_L)    
      AcY=Wire.read()<<8|Wire.read();  // 0x3D (ACCEL_YOUT_H) & 0x3E (ACCEL_YOUT_L)
      AcZ=Wire.read()<<8|Wire.read();  // 0x3F (ACCEL_ZOUT_H) & 0x40 (ACCEL_ZOUT_L)
      Tmp=Wire.read()<<8|Wire.read();  // 0x41 (TEMP_OUT_H) & 0x42 (TEMP_OUT_L)
      GyX=Wire.read()<<8|Wire.read();  // 0x43 (GYRO_XOUT_H) & 0x44 (GYRO_XOUT_L)
      GyY=Wire.read()<<8|Wire.read();  // 0x45 (GYRO_YOUT_H) & 0x46 (GYRO_YOUT_L)
      GyZ=Wire.read()<<8|Wire.read();  // 0x47 (GYRO_ZOUT_H) & 0x48 (GYRO_ZOUT_L)
      Serial.print("AcX = "); Serial.print(AcX);
      Serial.print(" | AcY = "); Serial.print(AcY);
      Serial.print(" | AcZ = "); Serial.print(AcZ);
      Serial.print(" | Tmp = "); Serial.print(Tmp/340.00+36.53);  //equation for temperature in degrees C from datasheet
      Serial.print(" | GyX = "); Serial.print(GyX);
      Serial.print(" | GyY = "); Serial.print(GyY);
      Serial.print(" | GyZ = "); Serial.println(GyZ);
      delay(333);
    }
`

i get this:

`AcX = 32767 | AcY = -296 | AcZ = 32767 | Tmp = 27.73 | GyX = 4796 | GyY = 1037 | GyZ = 1659

AcX = 32767 | AcY = -164 | AcZ = 32767 | Tmp = 27.64 | GyX = 4830 | GyY = 1041 | GyZ = 1657

AcX = 32767 | AcY = -296 | AcZ = 32767 | Tmp = 27.73 | GyX = 4803 | GyY = 1034 | GyZ = 1652`

everything look ok except AcX and AcZ

ZHomeSlice, 2023-05-03:
it looks like AcY is calibrating but AcX and AcZ are not. This could be because AcX for some reason is seeing gravity. but it is very curious as to why you are having so much trouble. 

I'm Curious if using my Simple_MPU6050 calibration routine will give you any good results.
https://github.com/ZHomeSlice/Simple_MPU6050
found in the examples 
https://github.com/ZHomeSlice/Simple_MPU6050/tree/master/Examples/Simple_MPU6050_Calibration
you will also  need my Simple_Wire Library to compile the Simple_MPU6050 Calibration example.
https://github.com/ZHomeSlice/Simple_Wire

I may be able to help you better as my calibration library was what I used to create Jeffs calibration routine for his library.
While they should be the same or very similar I may have added a fix or make other tweaks to work with oddities in calibration. 
Z

chevalfromage, 2023-05-07:
Hi, so i tried what you said and then i tried to use an other GY-521 and it seem like the one i was using is just broken. Thanks for the help