jrowberg/i2cdevlib #580: 'struct MPU6050' violates the C++ One Definition Rule in Arduino IDE on MPU650_DMP6 example

belousa, 2021-03-28:
Simply compiling the example produces a lot of nasty warning, starting with 

```
...\Documents\Arduino\libraries\MPU6050/MPU6050.h:436:7: warning: type 'struct MPU6050' violates the C++ One Definition Rule [-Wodr]
 class MPU6050 {
       ^
...\Documents\Arduino\libraries\MPU6050\MPU6050.h:436:7: note: a different type is defined in another translation unit
 class MPU6050 {
       ^
...\Documents\Arduino\libraries\MPU6050/MPU6050.h:1036:18: note: the first difference of corresponding definitions is field 'dmpPacketBuffer'
         uint8_t *dmpPacketBuffer;
                  ^
...\Documents\Arduino\libraries\MPU6050\MPU6050.h:436:7: note: a type with different number of fields is defined in another translation unit
 class MPU6050 {

```

I think that the issue is that compiling the example involves compiling two .cpp files: `...\Documents\Arduino\libraries\MPU6050/MPU6050.h` and `...\Documents\Arduino\libraries\MPU6050\examples\MPU6050_DMP6_using_DMP_V6.12.ino`. 

The difference comes from 

```
    #if defined(MPU6050_INCLUDE_DMP_MOTIONAPPS20) or defined(MPU6050_INCLUDE_DMP_MOTIONAPPS41)
        uint8_t *dmpPacketBuffer;
        uint16_t dmpPacketSize;
    #endif
```

The macro MPU6050_INCLUDE_DMP_MOTIONAPPS20 is defined in the `.ino` file and not defined in the `.cpp` file, hence the difference.

The way the class is used, this issue likely does not truly matter. And I am not sure how to fix it either. But I wanted to put it out there...

MathewAloisio, 2021-03-28:
What you need to do is find any function defined and implemented in a header file and prefix them with 'inline'.

belousa, 2021-03-29:
I doubt very much that this will help. The problem is not an issue with same symbol found twice (which is what this suggestion usually solves). The problem is that two different compilation uints have different (datat) sizes for the same class name. The correct solution is to derive an MDP class from tehe raw class, adding the needed members. But that, I assum, would break too many existing sketches.

MathewAloisio, 2021-03-29:
Oh yeah you're right, I didn't notice you only were getting the warning about the class violation rule.

jrowberg, 2021-09-28:
This should be fixed with the most recent commit.

closed 2021-09-28 by jrowberg