jrowberg/i2cdevlib #591: Custom define MPU6050 DMP setting is important in the MPU6050_6Axis_MotionApps_V6_12

williamesp2015, 2021-08-20:
In dmpInitialize() function the Accelerometer scale is fixed in 2g and gyro in 2000 dpm. Is anyone made an effort to customize the Acc and gyro sampling rate and scale and Low pass filter?

eadf, 2021-09-26:
I could have sworn I made an experiment with template specializations that solved that problem (some 5+ years ago)
But I can't find the source anymore. Maybe the idea is still valid?

ZHomeSlice, 2021-09-27:
So This is the reason the rates are where they are at:

Refference: 
Motion Driver 6.12 – Features User Guide
Document Number: AN-EMAPPS-0.0.6
Revision: 1.1
Release Date: 05/05/2015
Page 4 Reads:

4 The DMP features
The DMP is a fast, low power, programmable, embedded lightweight processor in the MPU devices. It is design to offload functionality, like sensor fusion and gesture recognition, from the MCU to save overall power in the system.
The DMP has many features which can be dynamically turned off and on at run-time. Individual feature can also be disabled while leaving others running. All DMP data is outputted to the FIFO except for pedometer. The DMP can also be programmed to generate an interrupt via gesture or if data ready. For details on flashing and enabling the DMP please read the porting guide.

 3 Axis Low Power Quaternions – ...... not used here

 6 Axis Low Power Quaternions – gyro and accel quaternions. Similar to the 3-axis LPQ, integrates the accel and gyro at 200Hz sampling rates will outputting to the FIFO at the user requested rates. The 3-axis LPQ and 6-axis LPQ are mutually exclusive and should not be run concurrently. If enabled the 6-axis quaternions can be pushed into the MPL library and the MPL will handle the compass integration for 9-axis.

DMP is the pre-compiled proprietary code that is embedded into the MPU it has minimal configuration options that mostly turn on and off features. 
Something more interesting is the MPL library another proprietary pre-compiled library that I haven't gotten to work with the Arduino IDE :) but that's for another discussion.

Download location
https://invensense.tdk.com/developers/
you will need an account to get to the downloads
look for "eMD 6.12"

Z

williamesp2015, 2021-09-27:
Dear @ZHomeSlice you haven't respond to the issue regarding custom Acc and Gyro measurement range  not fixed 2g and 2000 Degree/sec.

wrickout, 2024-06-04:
DMP sensor fusion works only with gyro at +-2000dps and accel +-2G