jrowberg/i2cdevlib #593: HMC5883L.h under STM32 problems on microprocessor

Escrich, 2021-09-06:
Dear all as I'm trying to use your good work to connect an HMC5883 accelerometer under an STM32F41XXX, I'm getting the error:

compilation terminated.
exit status 1
stm32f10x.h: No such file or directory

 I took a file with same name from
https://github.com/espruino/Espruino/blob/master/targetlibs/stm32f1/lib/stm32f10x.h
But I get the new error:

compilation terminated.
exit status 1
Error compilando para la tarjeta Generic STM32F4 series.

Inside the file i can not choose from different microcontrollers or such, and any way, even in case of it works with BluePills processors, what is it intended to happen with Black Pills?

Until now I'm still trying without success.

flavioipiranga, 2021-09-13:
Hi @Escrich, 

Check this file: https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f4xx.h

Instead using stm32f10x.h try use the stm32f4xx.h. After that you will need to change this line: https://github.com/jrowberg/i2cdevlib/blob/e2bef30642685443d2f3a88fa210b6977e78d7bf/STM32/I2Cdev.h#L37

The change will be from:
`#include <stm32f10x.h>`
To that:
`#include <stm32f4xx.h>`

Escrich, 2021-09-14:
Dear all, i made all modifications sent to me ,but at end i still have same error:

 

exit status 1

'HMC5883L' does not name a type

Any suggestion about to how to solve it?

 

This is the sketch I’m testing:

 

// I2C device class (I2Cdev) demonstration Arduino sketch for HMC5883L class

// 10/7/2011 by Jeff Rowberg ***@***.***>

// Updates should (hopefully) always be available at https://github.com/jrowberg/i2cdevlib

//

// Changelog:

//     2013-05-04 - Added Heading Calculation in degrees

//     2011-10-07 - initial release

 

/* ============================================

I2Cdev device library code is placed under the MIT license

Copyright (c) 2011 Jeff Rowberg

 

Permission is hereby granted, free of charge, to any person obtaining a copy

of this software and associated documentation files (the "Software"), to deal

in the Software without restriction, including without limitation the rights

to use, copy, modify, merge, publish, distribute, sublicense, and/or sell

copies of the Software, and to permit persons to whom the Software is

furnished to do so, subject to the following conditions:

 

The above copyright notice and this permission notice shall be included in

all copies or substantial portions of the Software.

 

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR

IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,

FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE

AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER

LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,

OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN

THE SOFTWARE.

===============================================

*/

 

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation

// is used in I2Cdev.h

#include "Wire.h"

 

// I2Cdev and HMC5883L must be installed as libraries, or else the .cpp/.h files

// for both classes must be in the include path of your project

#include "I2Cdev.h"

#include "HMC5883L.h"

 

// class default I2C address is 0x1E

// specific I2C addresses may be passed as a parameter here

// this device only supports one I2C address (0x1E)

HMC5883L mag;

 

int16_t mx, my, mz;

 

#define LED_PIN PC13

bool blinkState = false;

 

void setup() {

    // join I2C bus (I2Cdev library doesn't do this automatically)

    Wire.begin();

 

    // initialize serial communication

    // (38400 chosen because it works as well at 8MHz as it does at 16MHz, but

    // it's really up to you depending on your project)

    Serial.begin(38400);

 

    // initialize device

    Serial.println("Initializing I2C devices...");

    mag.initialize();

 

    // verify connection

    Serial.println("Testing device connections...");

    Serial.println(mag.testConnection() ? "HMC5883L connection successful" : "HMC5883L connection failed");

 

    // configure Arduino LED pin for output

    pinMode(LED_PIN, OUTPUT);

}

 

void loop() {

    // read raw heading measurements from device

    mag.getHeading(&mx, &my, &mz);

 

    // display tab-separated gyro x/y/z values

    Serial.print("mag:\t");

    Serial.print(mx); Serial.print("\t");

    Serial.print(my); Serial.print("\t");

    Serial.print(mz); Serial.print("\t");

    

// To calculate heading in degrees. 0 degree indicates North

    float heading = atan2(my, mx);

    if(heading < 0)

      heading += 2 * M_PI;

    Serial.print("heading:\t");

    Serial.println(heading * 180/M_PI);

 

    // blink LED to indicate activity

    blinkState = !blinkState;

    digitalWrite(LED_PIN, blinkState);

}

 

 

 

Thanks a lot for your help

 

 

Jose M. Escrich

Arganda del Rey 

Madrid 28500

Spain

Enviado el: martes, 14 de septiembre de 2021 0:55
Para: jrowberg/i2cdevlib
CC: Jose M. Escrich; Mention

flavioipiranga, 2021-09-14:
Hey @Escrich,
Since you mentioned the BlackPill board I assumed that you were using STM32CubeIDE.

At Arduino IDE 1.8.16 using the code that you mentioned in the last comment, I achieve a successful build.

I did the following configs:

1. Add `https://github.com/stm32duino/BoardManagerFiles/raw/master/package_stmicroelectronics_index.json` to additional boards Manager URLs;
2. Download the STM32 board package;
3. Select the correct Board(Generic STM32F4) and Board Number(BlackPill F411CE);
4. Download i2cdevlib source code;
5. Include libraries I2Cdev and HMC5883L using Sketch->Include Library-> Add .ZIP library (select the folder correspondent to each  library);
6. Build and smile!

Escrich, 2021-09-15:
Sorry but it’s not being so easy, of course I use Cube, and I did everything related to libraries as you said.

First of all as far as I have other libraries for same device on other different microcontroller chips, I need to include the complete path to reach the right library, your library:

 

Once done, asks me about library Ic2_stm32f4xx.h, then I was included it changing a little the stm32f4xx.h one:

Please look at the header tabs

 

Selection made inside the file

 

 

Once done all of this, finally I get the same error than in the beginning:

 

 

And really i dont know what to change to mede it valid.

 

Thanks for all your valuable help.

 

Regards:

 

Jose M. Escrich

Arganda del Rey 

Madrid 28500

Spain

Enviado el: martes, 14 de septiembre de 2021 23:33

flavioipiranga, 2021-09-15:
Well, looks like you included some print screen images. But I can't see any print screen, could you confirm if this is the case?

flavioipiranga, 2021-09-15:
@Escrich just to confirm, are you trying to build the example code using STM32CubeIDE?