jrowberg/i2cdevlib #631: MPU6050 Not staying connected with Arduino Due

cswoods, 2021-11-01:
I have been using the I2Cdevlib with great success with an Arduino Mega 2560. I wanted to get faster DAQ, so I got a Due. The issue I am having is that after a while of reading and printing values, (~25 secs), the DAQ gets much slower(>200Hz to <1Hz) and prints the same values. I believe the accelerometers are losing their connection to the Due. When I rerun the code, or close the Serial Monitor and reopen it, the MPU6050 connection fails and I only get 0s. My full hardware setup is a Due, multiplexer, and four MPU6050s. I have the ADO pin wired to the ground (0x68). I have also tried wiring the ADO pin to the 3.3V (0x69). I have the MPUs and multiplexer wired to the 3.3v. The thing that does not make any sense is that every now and then, it will work wonderfully, but then after it ran for about 25 seconds, the connection will be lost and it will not reconnect for awhile. Attached is my code and I am using an up to date I2Cdevlib, (2 weeks ago).  Any help is appreciated
`#include "I2Cdev.h"
#include "MPU6050.h"
#include "Wire.h"
// ------------------------ Multiplexer CONFIG --------------------------
#define TCAADDR 0x70
bool foundIMU[8] = {false};
unsigned long timer = 0;
void tcaselect(uint8_t i){
    if (i > 7) return; 
    Wire.beginTransmission(TCAADDR);
    Wire.write(1 << i);
    Wire.endTransmission();  
    
}
MPU6050 accelgyro(0x69);
int16_t ax, ay, az;
int16_t gx, gy, gz;
int iter;
#define ARRAY_LEN 3
#define OUTPUT_READABLE_ACCELGYRO
#define LED_PIN 13
bool blinkState = false;

void setup() {

    // join I2C bus (I2Cdev library doesn't do this automatically)
    #if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
        Wire.begin();
    #elif I2CDEV_IMPLEMENTATION == I2CDEV_BUILTIN_FASTWIRE
        Fastwire::setup(400, true);
    #endif
    Serial.begin(115200);

    // Wait until Serial communication isdetected
    while(!Serial) {}; 
    
    InitMPU(0);
    InitMPU(1);
    InitMPU(2);
    InitMPU(3);
}

void loop() {
    // read raw accel/gyro measurements from device
    //accelgyro.getMotion6(&ax, &ay, &az, &gx, &gy, &gz);
    while (millis() < 30000){

      Serial.print(millis());
      Serial.print("\t");
      readIMU(0);
      readIMU(1);
      readIMU(2);
      readIMU(3);
 
    }

  delay(5000);
}

void InitMPU(uint8_t muxID){
  //
  // Switch to Assigned Multiplexer Channel
  tcaselect(muxID);

  accelgyro.initialize();

  Serial.println("Testing device connections...");
  Serial.println(accelgyro.testConnection() ? "MPU6050 connection successful" : "MPU6050 connection failed");
}

void readIMU(uint8_t muxID){
   tcaselect(muxID);
//   accelGyro.getMotion6(&ax, &ay, &az, &gx, &gy, &gz);
   accelgyro.getAcceleration(&ax ,&ay, &az);  

   Serial.print(ax); Serial.print("\t");
   Serial.print(ay); Serial.print("\t");
   Serial.print(az); 
     
   if(muxID==3){
    Serial.println("");
   }
   else{
   Serial.print("\t");
   }
}`

jrowberg, 2021-11-02:
I'm not sure I follow the logic here; with the `while (!Serial) {}` in `setup()`, you'll eat up a few seconds after initial boot/reset, and then your `loop()` code will stop reading and generating output after less than 30 seconds--your estimate of 25 sounds reasonable given the wait-for-USB behavior in `setup()`:

```
void loop() {
    while (millis() < 30000){
        // NOTE THAT millis() ONLY RESETS TO 0 AFTER LOOPING AT 2^32
        // SO THIS IS PRACTICALLY EQUIVALENT TO RUNNING EXACTLY ONCE
    }

  delay(5000); // <-- why?
}
```

What if you get rid of the `while (...)` condition and remove the `delay(5000)` at the same time?

ZHomeSlice, 2021-11-02:
Thought that came to mind:
Did you modify the MPU6050 Breakout board to properly work with 3.3V?
[Schematic](https://lh3.googleusercontent.com/proxy/V4hcBKmCmqdrBcY0-1c7HhOFvVqbiz3J0l9utwP73F3uAg7UEXzZaxV5DLXsZp8S8yy8Ya0rLAGr_fr7WMNni-vBn82za6Ss29oy5ldZGzknXY4d6hFi3_TVkzE4YNg5)
Note that the breakout board has a 5V regulator and supplying the 5V regulator with 3.3V will only further reduce the MPU6050 voltage likely causing it to fail.
Z

eadf, 2021-11-02:
That schematic link does not work for me, but you can use a [bi-directional level shifter](https://randomnerdtutorials.com/how-to-level-shift-5v-to-3-3v/) to get around that problem.
 
Be careful with 5V and Due. The pins on a Due are *not* [5V tolerant.](https://store.arduino.cc/products/arduino-due)

cswoods, 2021-11-02:
The problem is that the the connection to the accelerometers seem to close for some reason, because I will still get readings before the 30 secs run out, however, they will be much slower and the same numbers over and over. The weird thing is that sometimes it will run the whole 30 secs and sometimes it will only run for like 10 seconds. Then If I close the Serial Monitor and reopen it, it says the connections have failed and the only values I get are zeros. If I re-upload the code and then open the Serial Monitor, the same thing happens as when I just closed and reopened the Serial Monitor. I do not have this problem on the Mega 2560, so I figure it has something to do with the ARM chip on the Due.
Also, I found a work around. When the connection to the accelerometers fails, I can unplug the Due for a few seconds and then plug it back in and open the Serial Monitor and it works. I don't know if it is just a Due thing or what.

eadf, 2021-11-02:
I've also noticed that the Due is more finicky with the i2c pull down resistors when the 5V Atmegas can sometimes do without (i.e only use the internal pd)

ZHomeSlice, 2021-11-02:
Sorry the link was temporary... here's the updated link [Schematic](https://aws1.discourse-cdn.com/arduino/original/4X/7/4/6/74631f82563988409a5cddf32fdacbfee3e4707f.jpeg)
Note that the i2c Buss is tied on the 3.3V side to the pull-up resistors. 5V is not the issue on the i2c bus side.
It's the fact that if you don't provide 5V to the voltage regulator on the breakout board it will further drop the voltage provided to the i2c pull-up and the mpu6050. I recommend bypassing the regulator if you are only using a 3.3v system so that you don't have this issue. This requires a bit of modification to the breakout board.

Z

closed 2021-11-09 by cswoods