jrowberg/i2cdevlib #632: ESP32 ESP-IDF example gives almost the same value for yaw and pitch

pnn16004, 2021-11-08:
## Issue template

- Hardware description: ESP32
- Library: ESP32_ESP-IDF
- Installation type: Use as a component for micro-ROS
- Compilation: micro-ROS ESP-IDF component

#### Steps to reproduce the issue
Combine the provided example with the int32_publisher_custom_transport example in the micro-ROS ESP-IDF component.

#### Expected behavior
Get roll, pitch, yaw.

#### Actual behavior
Yaw and pitch have pretty much the same value (but not exactly the same).

#### Additional information
I changed no functionality of the original example, and everything else works as expected, but yaw and pitch give almost the same values. Can someone confirm that the original ESP32_ESP-IDF example works correctly for them?

eadf, 2021-11-08:
Are you able to test your MPU6050 with another controller (Arduino or something)?
Just to make sure it is ok.

pnn16004, 2021-11-09:
@eadf That is a good idea. Unfortunately I don't have any other controllers, but I instead tries this Arduino code with Platform.IO (VS Code): https://github.com/ElectronicCats/mpu6050, which is based on i2cdevlib.

I now get three different values for yaw, pitch, roll, so the IMU seems to work properly (MPU6050_DMP6 example).

The codes are very similar, so I'm not sure why the ESP32 ESP-IDF example isn't working.

ZHomeSlice, 2021-11-09:
@eadf I've created an alternative library version for accessing the MPU6050. In doing so I have used Jeff's i2cdev library as my base and extended it to encompass the entire mpu6050 registry map using macros rather than individual functions. with that said I am involved with Jeff's library improving it as I find things. 

Try my library out with your code and let me know if it has the same issues. This will allow me to help fix jeffs code.

[https://github.com/ZHomeSlice/Simple_MPU6050](https://github.com/ZHomeSlice/Simple_MPU6050)

I believe most examples if not all are now compatible with the ESP series of controllers.
Z

pnn16004, 2021-11-09:
@ZHomeSlice I'm going to assume you meant me. To me it looks like your library is for using Arduino only. I need to use ESP-IDF since I want to use it together with micro-ROS. If you simply want me to test it, I can do it. Which examples should I test?

ZHomeSlice, 2021-11-09:
@pnn16004 I am interested in seeing if this version is compatible. as well as if it has the same error as the MPU6050 library you are experiencing.
I'm interested in working with the ESP32 more and I appreciate the pointers to ESP-IDF and micro-ROS Where would I find out more information on these.
Thanks
My library and Jeffs are almost == when it comes to performance as we end up doing almost the same thing. Jeffs is function-driven and my code is macro-based which creates functions at compile time.

Z

pnn16004, 2021-11-11:
I tried the ESP32 ESP-IDF example alone with Platform.IO and it works. There is something wrong elsewhere with the micro-ROS code.

closed 2021-11-11 by pnn16004