jrowberg/i2cdevlib #633: Manage gyro and acc offset on calibration process before interrupt

MaxESP, 2021-11-13:
Dear M Jrowberg,

Thanks a lot for your MPU6050 library is the best one i have tried ; i have achieve a tilt compass on esp32 with couple mpu6050 and lsm303dlh.

It's work great about 3 degrees max  at -+90° pitch or roll inclination .I going to work to improve the little error.....

I want to achive a automatic calibration using memory but i was not able to manage offset data of gyro and acc on the calibration process on the sketch mpu6050 DMP6 v6v12.  Can you please provid the link of the fonction that print those values on serial monitor.

yours faithfully

ZHomeSlice, 2021-11-13:
`mpu.PrintActiveOffsets();`
does the printing of the offsets to serial
Z

MaxESP, 2021-11-13:
thank a lot M JROWBERG i am going to try manage those values before interrupt and put them in memory

MaxESP, 2022-09-10:
Thank M ZhomeSlice i was able to store them in memory thanks a lot 

how can I manage 
mpu.setXGyroOffset(42) ;
    mpu.setYGyroOffset(99);
    mpu.setZGyroOffset(79);
    mpu.setXAccelOffset(-919); 
    mpu.setYAccelOffset(2015); 
    mpu.setZAccelOffset(1697); 

to put the value memory in mpu.setXGyroOffset(?) ;

MaxESP, 2022-09-11:
I want to presise my previous post.

1 The function  printf the values of the array 
2 I am able to store those value as a json format in spiffs
3 Now i have to set a x value for int 16-t  for mpu.setXGyroOffset(x) ;   x =0;
4 imput the new  int16_t value(mpu.setXGyroOffset(x)) from json declare the new value of x;

 I am stuck in point 4 to convert int value x from json to int16_t value declaration