jrowberg/i2cdevlib #635: Git Pull updates but leaves MPU9150 folder Red

paynterf, 2021-11-13:
I just did a Git->Pull of the i2cdevlib folder tree into my Windows 10 PC with Git 2.32.0.2 installed , but the i2cdevlib\Arduino\MPU9150 folder still shows the red exclamation point icon instead of the green checkmark. I redid the 'Pull' several times, and I looked through the folder on GitHub, but could find no reason for this.  Am I missing something?

TIA,

Frank
[image]

eadf, 2021-11-22:
I have no experience with git on windows, but perhaps you are using something like this: <https://tortoisegit.org/docs/tortoisegit/tgit-dug-wcstatus.html>. It's indicating that the file is modified locally. 
What does `git status` say?

paynterf, 2021-11-22:
Git status shown below, along with the results of a diff at the folder level.  AFAICT, the local folder and the remote folder are identical.  I did see some other discussions that indicated that the difference could be due to a capitalization difference in a file or folder name, where windows ignores case but Github doesn't. Could that have happened somewhere in the MP9150 folder tree?
[image]
[image]

eadf, 2021-11-22:
I found this comment: "We dropped case-insensitive matching with TortoiseGit 2.3.7, ..." <https://gitlab.com/tortoisegit/tortoisegit/-/issues/2654> (and that was four years ago)
So it could be as you say. 

You could try and install a dedicated git GUI app, like [SourceTree](https://www.sourcetreeapp.com) and see if it finds any issues. Sourcetree is free and really excellent imho.

paynterf, 2021-11-23:
I'm pretty happy with Tortoise Git, so I think I'll just put up with the
red '!' icon overlay on the MPU9150 folder ;-).

Frank