jrowberg/i2cdevlib #636: MPU6050A DMP init failed

Kanken6174, 2021-11-14:
I have 3 MPU6050 modules, from the same vendor. But only 1 out of the 3 work with this library (this one being the genuine 6050), even though they all work when i'm pulling data directly from the register.

I am using the unmodified MPU6050DMP6 sketch of the lib, but 2 of my modules report their WHOAMI register as 0x98 instead of 0x68 as it should be. i've tried bypassing this issue by modifying the `testConnection()` function of the lib, but even after adding the new address as a valid one, so clearly there are more underlying issues.

The counterfeit 6050A either fail to init the DMP with code 1, or loop forever while prininting "*" in the loading bar.

ZHomeSlice, 2021-11-15:
The *** thing is the Calibration routine.
Make sure your MPI is flat or just skip the calibration.
Also, you can skip the test connection if it communicates it works if it doesn't It fails.

Z

Kanken6174, 2021-11-15:
after skipping the test it fails. It seems that this library just won't work with the 6050A chips, only the 6050, even though they're supposed to be functionnally the same.
(i can pull the raw values from the registers just fine, so it's not that the 6050A are completely dead, there's just something wrong wwith their DMP.)

ZHomeSlice, 2021-11-17:
Can you get me a link to the Registery map and datasheets? I did not know of a version of the MPU6050 that has the A designation. could it be an early Alpha chip...

could you try it with my version of the library to see if it works here? The results should be exactly the same I am just more familiar with my library.
[https://github.com/ZHomeSlice/Simple_MPU6050](https://github.com/ZHomeSlice/Simple_MPU6050)

Z

Kanken6174, 2021-11-18:
[image]

This is your lib, running the Seimpl_MPU6050_Basic_Example on the 6050A (it goes on forever)

[image]
And this is the genuine 6050 running the same code after 3 seconds

Kanken6174, 2021-11-18:
[image]
Original log of the 6050, as you can see the WHOAMI address is different.

ZHomeSlice, 2021-11-19:
************ forever
Calibration is failing to complete. 
Remove calibration routines soon setup and try again.
Z

Kanken6174, 2021-12-05:
I have found the underlying issue, the mpu6050A is actually not an invensense MPU6050, but a TDK ICM-20689, which does indeed have a WHOAMI register value of 152 and not 104 like the 6050. The library is failing because the config register isn't at the same address, unlike the data registers which are in the same place.

It's another chip entriely. Your code is not at fault here.