jrowberg/i2cdevlib #637: need help using a tilted 6050

mark-hahn, 2021-11-21:
I have designed a two-wheel toy.  When the prototypes arrived I was dissapointed to find it could only move in a drunkenly fashion.  The two cheap motors on the wheels behaved differently with the same drive current.  Even with fine control from dragging a finger on the cellphone I coudn't drive in a straight line.  I want it to go in a straight line when you lift your finger.

So I dropped a 6050 on the control board and got in new pcbs.  I started researching code to use the 6050 and was surprised at how complicated it is.  My main problem is the control board is tilted at a 60 degree angle such that the 6050 Z-axis tilts and points mostly to the forward direction.

I only need yaw on the toy's Z-axis for turning, and acceleration on the toy's X-axis to control speed.  The toy will only be used on level floors so gravity will be parallel to the toy's Z-axis.  There will be no pitch or roll.

Acceleration will be easy to code.  The tilt is only forward so the rotation is around the y-axis.  I'll measure gravity to get the angle and do a simple 2D rotation in the ZX plane.  Then the rotated raw x-axis acceleration will let me control speed.  I can ignore the resulting raw Z and Y values and I don't even have to subtract gravity.

However, I am confused about how to get the yaw for the toy reference frame.  When the toy turns there will be rotation on the chip's Z and X axes (not on Y) .  I need to calculate the toy's yaw from those two raw values.  I wouldn't think that I need to do an integrating AHRS algorithm but I could be wrong.

Can someone give me a hint on how to do this calculation?

ZHomeSlice, 2021-11-21:
It sounds like you are doing something similar but simpler to my balancing bot. I did a fly-by-wire direction control so that it would drive in a straight line while balanced all using the mpu6050. here is the example video driving on all types of surfaces grass slopes and even hopping off of a sidewalk curb.
[https://youtu.be/JI7Pp7kWgmg](https://youtu.be/JI7Pp7kWgmg)

Let me know if this is correct I would be happy to share my code. It can use Jeffs mpu6050 library
I use the DMP because it automatically tracks quaternion calculations and is simple to implement. the balancing bot only needed readings acquired at 10ms increments to operate successfully using 2 PID loops to control balance and direction. 

Z

mark-hahn, 2021-11-21:
That is super cool!  I'm impressed.  Is it going to be a product?  If not it should be.

My bot is simpler in that is has a ball as a third leg and doesn't require balancing.  The only tricky thing it does is motion based on finger gestures.  And if the navigation accuracy is good enough it will detect an invisible finish line in races when it gets near the starting location.  Its main claim to fame is the video it provides on the phone for driving when the unit is not visible to the driver and for general fun.

Of course I'd like to read your code.  It's the main way I learn everything.  I assume it's open source and I can steal from it?

ZHomeSlice, 2021-11-22:
The MPU6050 library is almost identical in function with jeffs although it is written differently so if mine is confusing you can get the same functionality with Jeffs mpu6050 library. 

[https://github.com/ZHomeSlice](https://github.com/ZHomeSlice)

Additionally the RC_V1r3 has tank drive code as well as PID fly by wire code in it.
A brief note on how it works 
everything is interrupt driven including the MPU6050.
Whe the Radio RC receiver spits out a pulse the interrupt is triggered on rise and fall and the time is stored between 0 Left and 1000 right with 500 as center this is stored for later use.
The MPU6050 triggers a interrupt when data is ready and this in turn automatically triggers a call for the data to be retrieved. when this data is captured the Simple_MPU class triggers a callback function that we choose in setup
`mpu.on_FIFO(MPU_Math);`
MPU_Math must be setup with the following variables to allow for the data to be passed
`void MPU_Math (int16_t *gyro, int16_t *accel, int32_t *quat, uint32_t *timestamp) `
each time and only when the MPU spitts out data this function is ran.
here is where the magic happens and where you should look 

This is the script that brings them all together as a tank or yaw only control without the balancing bot stuff:
```
//#include "MPU_ReadMacros.h"
//#inc;ude "MPU_WriteMacros.h"
#include "Simple_MPU6050.h"
#include <RC_V1r3.h>

Simple_MPU6050 mpu;
RC RCRadio;

/*             _________________________________________________________*/
//               X Accel  Y Accel  Z Accel   X Gyro   Y Gyro   Z Gyro
//#define OFFSETS     1133,     -44,    1056,      50,       9,      20 // My Last offsets. you will want to use your own as these are only for my specific MPU6050
#define OFFSETS    -1482,   -1386,    1126,      59,     -46,     -24

/*             _________________________________________________________*/

// RC Radio Input Pins
#define  Xpin           10
#define  Ypin           A0
#define  Throttlepin    12

//Arc controller Version ? Pins
#define  LeftPWMin      6
#define  LeftForward    7
#define  LeftReverse    8
#define  RightPWMin     5
#define  RightForward   4
#define  RightReverse   3
uint32_t TimeoutTimer;
//***************************************************************************************
//******************              Callback Funciton                **********************
//***************************************************************************************
#define printfloatx(Name,Variable,Spaces,Precision,EndTxt) print(Name); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(dtostrf((float)Variable,Spaces,Precision ,S));}Serial.print(EndTxt);//Name,Variable,Spaces,Precision,EndTxt
#define PRINTSTIMER(t)    for (static uint32_t SpamTimer; (uint32_t)(millis() - SpamTimer) >= (t); SpamTimer = millis())

void MPU_Math (int16_t *gyro, int16_t *accel, int32_t *quat, uint32_t *timestamp) {
  PRINTSTIMER(1) {
    Quaternion q;
    VectorFloat gravity;
    float ypr[3] = { 0, 0, 0 };
    float yprDeg[3] = { 0, 0, 0 };
    mpu.GetQuaternion(&q, quat);
    mpu.GetGravity(&gravity, &q);
    mpu.GetYawPitchRoll(ypr, &q, &gravity);
    mpu.ConvertToDegrees(ypr, yprDeg);
    Serial.printfloatx(F("Yaw")  , yprDeg[0], 6, 1, F(" ")); //printfloatx is a Helper Macro that works with Serial.print that I created (See #define above)
    int16_t LeftDrive;
    int16_t RightDrive;
    int16_t X;
    int16_t Y;
    int16_t XDZ;
    int16_t YDZ;
    int16_t FbyW_Y;
    int16_t Throttle;
[... 89 lines cut ...]
  Serial.println(F("Using Offsets"));
  mpu.load_DMP_Image(OFFSETS); // loads the offsets defined above
#else
  mpu.CalibrateMPU(); // Calibrate the MPU
  mpu.load_DMP_Image();// Does it all for you 
#endif  
  mpu.on_FIFO(MPU_Math);
  RCRadio.pciSetupRC(14); // Throttle
  RCRadio.pciSetupRC(15); // Aleron/Roll
  RCRadio.pciSetupRC(16); // Elivator/Pitch
  RCRadio.pciSetupRC(17); // Rudder
  RCRadio.pciSetupRC( 9); // Aux1
  RCRadio.pciSetupRC(10); // Gear
  RCRadio.Deadzone(150); // Power Output dead zone where the motors have no drive ability
  RCRadio.onRCIsAlive(Radio_Alive);
  RCRadio.onRCIsNotAlive(Radio_Failed);
  Serial.print(F("Setup Complete"));
}

void loop() {
  mpu.dmp_read_fifo();// Must be in loop
}
```
Hope this helps
Z

mark-hahn, 2021-11-22:
Thanks mucho.  Will peruse it asap.

mark-hahn, 2021-11-28:
I'm just starting to look at your code.  A quick question:  do you calibrate the sensor every time you power it up or do you save the numbers somewhere?  Everyone seems to calibrate often, but I'm shipping a toy and it would be hard to ask a 7-year-old to hold it still each time they want to play.  :-)  I could have the UI start calibration from a menu item, but even parents wouldn't have the patience or inclination to do it.  Does calibrating make much difference for your robot?  If everything is using feedback then wouldn't the feedback adjust for everything automatically?

ZHomeSlice, 2021-11-28:
I use the #define to create a list of offsets

```
/*             _________________________________________________________*/
//               X Accel  Y Accel  Z Accel   X Gyro   Y Gyro   Z Gyro
//#define OFFSETS     1133,     -44,    1056,      50,       9,      20 // My Last offsets. you will want to use your own as these are only for my specific MPU6050
#define OFFSETS    -1482,   -1386,    1126,      59,     -46,     -24

/*             _________________________________________________________*/

```

then load the offsets at this point:
  `mpu.load_DMP_Image(OFFSETS); // loads the offsets defined above`

you could also do this:
  `mpu.load_DMP_Image(1133,     -44,    1056,      50,       9,      20); // loads the offsets defined above`

Z

closed 2021-12-04 by mark-hahn