jrowberg/i2cdevlib #638: Are there any advantages to using the DMP other than off-loading the cpu?

mark-hahn, 2021-11-21:
The subject line pretty much says it all.  Can I get the same results using the cpu, esp32 in my case, instead of the DMP?  I've done AHRS calculations on the esp32 and I can get 400 to 500 samples per second.  How many samples per second can the 6050 do with the DMP?  The rest of the application on the esp32 requires almost no cpu load since the wifi comm is on its own core.

ZHomeSlice, 2021-11-21:
Simply gives you quaternion calculations that can represent your MPU in a 3d environment with little to no load on your processor. Remember you will need to capture the rate over time to calculate the position of the MPU and doing this accurately with time seems to be the most difficult part. letting the MPU6050 do the work simplifies everything I'm my opinion. otherwise, if you can capture the data directly and do the math outside the MPU6050 then there is no reason why it won't work that way also.
Z

mark-hahn, 2021-11-21:
Thanks for the quick reply.

I forgot to mention that I'm 90% sure I'm using cloned 6050s.  My chinese supplier has something like 100,000 in stock and invensense doesn't make 6050s any more, right?  Can I rely on the chip id to indicate whether it's a clone?

I'm concerned that the DMP on a clone won't work right.  Does anyone have any experience with cloned DMPs?

ZHomeSlice, 2021-11-21:
the cloned 6050s Shouldn't matter much as long as the calibration works it should be good. worst case is that they will drift more than normal but for driving a bot you youldn't see much of a change.
Z