jrowberg/i2cdevlib #658: MPU6050 library problem with esp-idf > 4.2.2

scoulondre, 2022-01-16:
Hi, just FYI, I have a project using MPU6050 working perfectly with esp-idf 4.1.2, but not with 4.2.2 or 4.3.2. 
It might as well be a problem with esp-idf, but I did not try the idf pre-releases yet.
I am not stuck by this fortunately, but I thought this workaround may be helpful to others.
The init stage works well, but the mpu.getIntStatus() never returns the appropriate value and no data is received.
I therefore tried the example provided in the current repository i2cdevlib/ESP32_ESP-IDF/ and got the very same problem so I can rule out a bug on my side.
After changing the MPU chip, the ESP32 WROOM board (even with another generic brand), and basically all the possible hardware, it's still the same, so I guess you can reproduce it.

ZHomeSlice, 2022-01-17:
Have you tried it with this mpu6050 library?
[https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/MPU6050](https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/MPU6050)
I believe it has resolved the esp32 issues.

in MPU6050.h
```
#ifdef __AVR__
#include <avr/pgmspace.h>
#elif defined(ESP32) // <<< HERE >>>
    #include <pgmspace.h>
#else
//#define PROGMEM /* empty */
//#define pgm_read_byte(x) (*(x))
//#define pgm_read_word(x) (*(x))
//#define pgm_read_float(x) (*(x))
//#define PSTR(STR) STR
#endif
```
you will also want to use the Arduino[ i2cdev library ](https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/I2Cdev)as it has the esp adjustments and is compatible with the Arduino MPU6050 library

```
#ifndef I2CDEVLIB_WIRE_BUFFER_LENGTH
    #if defined(I2C_BUFFER_LENGTH)
        // Arduino ESP32 core Wire uses this <<< HERE >>>
        #define I2CDEVLIB_WIRE_BUFFER_LENGTH I2C_BUFFER_LENGTH
    #elif defined(BUFFER_LENGTH)
        // Arduino AVR core Wire and many others use this
        #define I2CDEVLIB_WIRE_BUFFER_LENGTH BUFFER_LENGTH
    #elif defined(SERIAL_BUFFER_SIZE)
        // Arduino SAMD core Wire uses this
        #define I2CDEVLIB_WIRE_BUFFER_LENGTH SERIAL_BUFFER_SIZE
    #else
        // should be a safe fallback, though possibly inefficient
        #define I2CDEVLIB_WIRE_BUFFER_LENGTH 32
    #endif
#endif
```

Z

scoulondre, 2022-01-17:
Thanks for you answer. I'd like to avoid using Arduino as an ESP-IDF component for various reasons. But that's go to know there is this workaround too.

ZHomeSlice, 2022-01-18:
What ESP-IDF are you using?
I'm Interested... This is just a hobby for me and I didn't know that there were alternatives.
Thanks
Z

scoulondre, 2022-01-18:
There are basically 2 different frameworks for ESP32:
- Arduino framework + Arduino core for the ESP32 https://github.com/espressif/arduino-esp32 which uses the Arduino setup() and loop() entry points. It's a kind of legacy development support for ESP32 for people who do not want to learn a new framework. However it also eliminates access to some of the ESP32 more advanced features, and only one of the two ESP32 cores are used.
- ESP-IDF native framework https://github.com/espressif/esp-idf which has nothing to do with the Arduino ecosystem, and relies on the appmain() entry point. Using an Arduino library is sometimes possible with some kind of wrapping ("Arduino as a ESP-IDF component"). The benefit is that it provides the highest fidelity against the ESP32 hardware.

The MPU6050 is primarily developed for the Arduino framework, and worked well for the ESP-IDF framework until version 4.1.2, that's what https://github.com/jrowberg/i2cdevlib/tree/master/ESP32_ESP-IDF provides.

gchauhan852, 2023-02-06:
@scoulondre I am beginner and have worked with ESP-IDF framework only. I am going to work on a hobby project where I'll be using a sensor (MP6050) for the 1st time. Online I could find references of how to install library and work with MP6050 in Arduino SDK framework but couldn't find any reference w.r.t ESP-IDF. Could you please help me by sharing some references.

Thanks