jrowberg/i2cdevlib #669: MPU6050 connection failed

Dudelsack3HD, 2022-02-08:
When I open Serial monitor it says:

Testing device connections...
MPU6050 connection failed
Send any character to begin DMP programming and demo: 

When I then press a random button it outputs:

Initializing DMP...
DMP Initialization failed (code 1)

I am using gy-521 and have connected pin A4 to SDA, A5 to SCL, 5V to VCC and GND to GND. When I run MPU6050_Base::getDeviceID() it outputs 0. Any suggestions how to get the sketch working?

ZHomeSlice, 2022-02-09:
See if you can find an address with this code
```
// --------------------------------------
// i2c_scanner
//
// Version 1
//    This program (or code that looks like it)
//    can be found in many places.
//    For example on the Arduino.cc forum.
//    The original author is not know.
// Version 2, Juni 2012, Using Arduino 1.0.1
//     Adapted to be as simple as possible by Arduino.cc user Krodal
// Version 3, Feb 26  2013
//    V3 by louarnold
// Version 4, March 3, 2013, Using Arduino 1.0.3
//    by Arduino.cc user Krodal.
//    Changes by louarnold removed.
//    Scanning addresses changed from 0...127 to 1...119,
//    according to the i2c scanner by Nick Gammon
//    http://www.gammon.com.au/forum/?id=10896
// Version 5, March 28, 2013
//    As version 4, but address scans now to 127.
//    A sensor seems to use address 120.
//
//
// This sketch tests the standard 7-bit addresses
// Devices with higher bit address might not be seen properly.
//
bool blinkState = false;
#include <Wire.h>

void setup()
{
  pinMode(9, OUTPUT);
  digitalWrite(9, LOW);
  Wire.begin();

  Serial.begin(115200);
  Serial.println("\nI2C Scanner");
  pinMode(3, OUTPUT);
  for (int x = 0; x < 2; x++) pinMode(8 + x, OUTPUT);
  for (int c = 0; c < 2; c++) {
    for (int x = 0; x < 2; x++) digitalWrite(8 + x, HIGH);
    digitalWrite(8 + c, LOW);
  }
}

void loop() {
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for (address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address < 16)
        Serial.print("0");
      Serial.print(address, HEX);
      Serial.println("  !");
      blinkState = !blinkState;
      digitalWrite(3, blinkState);
      nDevices++;
    }
    else if (error == 4)
    {
      Serial.print("Unknow error at address 0x");
      if (address < 16)
        Serial.print("0");
      Serial.println(address, HEX);
    }
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");
 // digitalWrite(9, !digitalRead(9));
  delay(5000);           // wait 5 seconds for next scan
    digitalWrite(8, !digitalRead(8));
    digitalWrite(9, !digitalRead(9));
  
}
```

yukiloh, 2025-04-07:
I also have this problem, and it will output "MPU6050 connection failed".
However, when it shows "Send any character to begin DMP programming and demo:"
After I input something, the MPU can obtain the Euler data normally.  
I've tried the code `// i2c_scanner`, and the output is 0x68. In fact, I've also connected it to the GND.