jrowberg/i2cdevlib #670: Updating to Arduino ESP32 2.02 breaks this library!

kjamesDBW, 2022-02-09:
Hi,
Couldn't figure it out at first but updating to **ESP32 2.02** in the Arduino Boards Manager breaks this library. It never manages to get past the .initialize() function:

// initialize device
    Serial.println("Initializing I2C devices...");
    accelgyro.initialize(); **<---- dies here**

Downgraded to **ESP32 2.0.0** and all back to working correctly. Phew!

Hope this save someone some time fiddling and swapping bits out like I have this evening...

mahboud, 2022-03-05:
I see the same issue with ESP 2.0.0 and 2.0.1 in addition to 2.0.2. I don't see the problem with 1.0.6, and that works for my other Adafruit ESP32, but the Adafruit ESP32 Feather v2 won't compile with the 1.0.6.

If you have a trick to make 2.0.0 work, I'd be interested in hearing it.

listout, 2022-04-03:
Same issue. Don't why it's still not fixed. If I have some spare time I'll sending a PR.

mp-se, 2022-05-06:
I have the same issue, looks like the problem is in the wire implementation in espressif32 for arduino. Have not dont any more deeper investigation but there are tickets open.

igcxl, 2022-07-07:
I have the same issue when updating to Arduino ESP32 2.0.3

mp-se, 2022-07-07:
Try to remove line 285 in i2cdev.cpp, that worked for me. If it works I can send in a PR. 

            for (int k = 0; k < length; k += min((int)length, I2CDEVLIB_WIRE_BUFFER_LENGTH)) {
                useWire->beginTransmission(devAddr);
                useWire->write(regAddr);
                useWire->endTransmission();
                // See: https://github.com/espressif/arduino-esp32/issues/6674
                // useWire->beginTransmission(devAddr); <-- Remove this line...
                useWire->requestFrom((uint8_t)devAddr, (uint8_t)min((int)length - k, I2CDEVLIB_WIRE_BUFFER_LENGTH));

brainwind-software, 2022-09-28:
@jrowberg Will here be a fix-commit for this issue?
I triefd for days, only to find out, there must not be begin/endTransmission around any requestForm,
and now I have a manually fixed version of this lib in my project. But I would like to see it in the official.
Its the same fix for issue #672

closed 2022-10-14 by jrowberg

jrowberg, 2022-10-14:
> @jrowberg Will here be a fix-commit for this issue?

Sorry it took me so long to get to such a mind-numbingly simple issue that's been in the library from the very beginning. There was one small part that was already fixed some time ago, but most of the offending code was still intact. It's now all fixed.

commit 98a3b4ec (2022-10-14): Remove extra begin/endTransmission calls during reads

Fixes #670 and #672