jrowberg/i2cdevlib #672: Hang in i2cdevlib when using ESP32 Board Manager v 2.0.2

mahboud, 2022-03-05:
I'm not sure if this is a bug in the i2clib or in the board manager code.

I only installed the minimum set of libraries I needed in my Arduino folder:

hardware:
    espressif

libraries:
    ArduinoJson I2Cdev      MPU6050

And in the board manager, I picked the ESP32 by Espressif board but at version 1.0.6.

I build my project and it works. This is what the console looks like when things work:

> Hello!!!
> prep MPU
> in prep: 
> in prep: b4
> Begin prepareMPU
> getIntEnabled: 64
> getIntMotionStatus: 0
> getIntMotionStatus should be reset by previous read: 0
> Testing device connections...MPU6050 
> connection successful
> Writing mode: 0
> end of prep
> MPUBoot number: 1
> Wakeup was not caused by deep sleep: 0
> Setup ESP32 to sleep for every 20 Seconds
> About to sleep
> Going to sleep now
> 

Now, I go and change the ESP32 by Espressif board support to version 2.0.2.

I build and run, and this is what I get in the console:

> 
> Hello!!!
> prep MPU
> in prep: 
> in prep: b4
> 

Ok, so where is it stuck? This is what the code looks like:
```

#include <MPU6050.h>
#include <Wire.h>

MPU6050 accelgyro;
int ledPin = 27;

void prepareMPU() {

  Wire.begin();
  Serial.print("in prep: ");

  pinMode(ledPin, OUTPUT);
  Serial.println("in prep: b4 ");
  accelgyro.initialize();
  Serial.println("Begin prepareMPU");

  bool mode = accelgyro.getInterruptMode();
  if (mode) {
    Serial.println("interrupt mode set");
  }
```

MPU6050 is the library from Jeff Rowberg. Same with I2Cdev library. Could this be a bug in either of those libraries? Or is this a bug in the 2.0.2 board support?

The hang is at 

`accelgyro.initialize();`

this is the code from initialize():

```
void MPU6050_Base::initialize() {
    setClockSource(MPU6050_CLOCK_PLL_XGYRO);
    setFullScaleGyroRange(MPU6050_GYRO_FS_250);
    setFullScaleAccelRange(MPU6050_ACCEL_FS_2);
    setSleepEnabled(false); // thanks to Jack Elston for pointing this one out!
}
```

Tracing it, I go through `setCLockSource()` and then`writebits()` and then `readByte()` and then `readBytes()`: 

> i1I2C (0x68) reading 1 bytes from 0x6B...

It gets stuck in this loop: 
```
for (int k = 0; k < length; k += min((int)length, I2CDEVLIB_WIRE_BUFFER_LENGTH)) {
                useWire->beginTransmission(devAddr);
                useWire->send(regAddr);
                useWire->endTransmission();
                useWire->beginTransmission(devAddr);
                useWire->requestFrom((uint8_t)devAddr, (uint8_t)min((int)length - k, I2CDEVLIB_WIRE_BUFFER_LENGTH));

                for (; useWire->available() && (timeout == 0 || millis() - t1 < timeout); count++) {
                    data[count] = useWire->receive();
                    #ifdef I2CDEV_SERIAL_DEBUG
                        Serial.print(data[count], HEX);
                        if (count + 1 < length) Serial.print(" ");
                    #endif
                }

                useWire->endTransmission();
            }
```

Specifically it gets stuck in                 
`useWire->requestFrom((uint8_t)devAddr, (uint8_t)min((int)length - k, I2CDEVLIB_WIRE_BUFFER_LENGTH));`

I believe the code for requestFrom() is in the Espressif hardware folder. I don't know if I should go tracing in there, or whether the problem isn't that deep.

If I switch back to board support 1.0.6, it starts to work again. I'd love to know where I should start poking around to figure out what is happening. I'd prefer to use the latest board support if at all possible.

i164j9, 2022-05-09:
As posted here: https://github.com/espressif/arduino-esp32/issues/6674

Run pio upgrade --dev to the latest developer version and then change platform.ini
platform = espressif32@3.4.0

If you are in fact using an esp32 board.

Armageddit, 2022-05-15:
i use an ESP32-C3
this ist not supportet in 3.4.0
for normal ESP32 `platform = espressif32@3.5.0` also woks fine