jrowberg/i2cdevlib #675: How to calibrate compass in dmp config?

Hortay, 2022-03-13:
Hello! I am trying to calibrate compass and setup mpu9250 with 9Axis_MotionApps41. When I change values in Dmp config nothing changes.

ZHomeSlice, 2022-03-13:
@Hortay 
Use the 6.12 version it has everything the older 4.1 version has with all the bug fixes. If there are functions missing let us know they should transfer form 4.1 to the 6.12 version without any issues.

The embedded digital motion processing eDMP (Binary file we upload to the mpu9250 for 6 Degrees of freedom) doesn't have the ability to integrate the compass internally.

Invensense has provided a library to do this... But it is pre-compiled, and I cannot figure out how to implement it into the Arduino architecture and it isn't part of Jeff's libraries "Yet"!

I want to make this happen... but I need help. if somehow we can figure out the process.

Where to download the files
The manufacturer's site is located here: https://invensense.tdk.com/
The link to follow is:
[eMD 6.12](https://invensense.tdk.com/developers/login/?message=software-downloads&redirect_to=developers/software-downloads/)
You will need to log in. just request access and they should give it to you.
Download the zip file from the above link once you get permission to log in.

The files you are looking for are in the following directory structure
motion_driver_6.12\msp430\eMD-6.0\core\mpl\
The Library file name is libmplmpu.lib 
Opening this .lib file provides us with nothing useful...

There are several .h files in there but the one you are likely looking most for is fusion_9axis.h.
The fusion_9axis.h file gives you access to the following compiled functions:
```
/*
 $License:
    Copyright (C) 2011-2012 InvenSense Corporation, All Rights Reserved.
    See included License.txt for License information.
 $
 */

/******************************************************************************
 *
 * $Id$
 *
 *****************************************************************************/

#ifndef MLDMP_FUSION9AXIS_H__
#define MLDMP_FUSION9AXIS_H__

#include "mltypes.h"

#ifdef __cplusplus
extern "C" {
#endif

    void inv_init_9x_fusion(void);
    inv_error_t inv_9x_fusion_state_change(unsigned char newState);
    inv_error_t inv_enable_9x_sensor_fusion(void);
    inv_error_t inv_disable_9x_sensor_fusion(void);
    inv_error_t inv_start_9x_sensor_fusion(void);
    inv_error_t inv_stop_9x_sensor_fusion(void);
    inv_error_t inv_9x_fusion_set_mag_fb(float fb);
    inv_error_t inv_9x_fusion_enable_jitter_reduction(int en);
    inv_error_t inv_9x_fusion_use_timestamps(int en);

#ifdef __cplusplus
}
#endif

#endif // MLDMP_FUSION9AXIS_H__
```
Hope this helps and keep us informed of your progress...
Someone must know how to do this :)

Z

Hortay, 2022-03-13:
Thanks for the answer. It's hard to understand quickly. But I'll try to get into the question.
And can i ask about some registers in .h file?

ZHomeSlice, 2022-03-13:
@Hortay 
I'll do my best. What questions do you have?
Z