jrowberg/i2cdevlib #683: Can't compile after recent commits

CaptainJeroen, 2022-04-12:
After recent commits into the master branch I can't seem to compile the MPU6050_raw example. Older versions compile and run fine. Sadly I'm no C++ expert so I have no clue how to solve the error. 

This is the error:

C:\...\MPU6050\MPU6050.cpp:3374:11: error: prototype for 'int16_t* MPU6050_Base::GetActiveOffsets()' does not match any in class 'MPU6050_Base'
 int16_t * MPU6050_Base::GetActiveOffsets() {
           ^~~~~~~~~~~~

In file included from C:\...\MPU6050\MPU6050.cpp:39:0:
C:\...\MPU6050\MPU6050.h:835:11: error: candidate is: int16_t MPU6050_Base::GetActiveOffsets()
   int16_t GetActiveOffsets();
           ^~~~~~~~~~~~~~~~

exit status 1
Error compiling for board Arduino Uno.

eaglehail, 2022-04-14:
I opened pull request #685 to address this compile error. It worked for me locally running the MPU6050 library on an Arduino Micro.

eaglehail, 2022-04-14:
Should be fixed now.