jrowberg/i2cdevlib #688: esp 32 mpu6050 with dmp not working in arduino ide

abhinmajix, 2022-05-18:
esp 32 mpu6050 with dmp not working with latest esp 32 boards 2.0.3. but it works with older 1.0.3 version. when using 2.0.3 dmp example able to upload but nothing coming up in serial monitor. is there anything to be initialized to work with esp 32 boards.

yufuping, 2022-05-23:
I encounter the same issue

isai17, 2022-05-27:
Same problem here

ZHomeSlice, 2022-05-29:
This may help:
https://github.com/jrowberg/i2cdevlib/issues/672#issuecomment-1120611043
I have not tested this.

yufuping, 2022-05-31:
I use Arudino IDE with esp32 for Arduino 2.03, it doesn't work.

igcxl, 2022-07-07:
I encounter the same issue

GiveSomeColor, 2022-07-12:
hello, i encounter this issue too, and i take some time look into it, i found out the root cause，
in the new version of  esp32 for Arduino lib，they add semaphore lock to TwoWire driver, the  beginTransmission funtion
will take the semaphore and in the  endTransmission function it will give the taken semaphore, 。
But in  I2Cdev::readBytes function the i2c reading procedure is as follow ，before the requestFrom() the semphore is taken by
beginTransmission function , so requestFrom will always pending on the semphore. 

 I2Cdev::readBytes
  useWire->beginTransmission(devAddr);
  useWire->write(regAddr);
  useWire->endTransmission();
  **useWire->beginTransmission(devAddr);**
  useWire->requestFrom((uint8_t)devAddr, (uint8_t)min((int)length - k, I2CDEVLIB_WIRE_BUFFER_LENGTH));

There is two approach to fix , 1. comment the sencond beginTransmission function in both  I2Cdev::readBytes and I2Cdev::readWords;(it is ok for ESP32 but i dont if it will break the other hardware platform )
or 2. add a definition #define CONFIG_DISABLE_HAL_LOCKS 1 in  esp32 for Arduino lib to disable the semaphore lock.

ArtrixTech, 2022-07-18:
> 

Worked fine for me!!!!! Thanks a lot for solving this!

ZHomeSlice, 2022-07-18:
should we consider adding something like: 
`#if not defined(ESP32)
`

Example:
```
I2Cdev::readBytes
useWire->beginTransmission(devAddr);
useWire->write(regAddr);
useWire->endTransmission();
#if not defined(ESP32)
useWire->beginTransmission(devAddr);
#endif
useWire->requestFrom((uint8_t)devAddr, (uint8_t)min((int)length - k, I2CDEVLIB_WIRE_BUFFER_LENGTH));
```

jrowberg, 2022-07-18:
No, that spurious `beginTransmission()` call should just be removed. The `requestFrom()` call does the whole read transaction atomically. I thought this was gone already from other implementation, but it seems it's still there. I suspect that the duplicate operation is silently ignored in the AVR/SAM/SAMD core code, but it appears to cause a problem here. It can (and should) come out of all implementations.

There's a good explanation of what each of the main Wire library transaction functions do here, for reference:

https://github.com/Koepel/How-to-use-the-Arduino-Wire-library/wiki/Explanation-of-the-functions-of-the-Wire-library

chufusheng, 2022-09-27:
> hello, i encounter this issue too, and i take some time look into it, i found out the root cause， in the new version of esp32 for Arduino lib，they add semaphore lock to TwoWire driver, the beginTransmission funtion will take the semaphore and in the endTransmission function it will give the taken semaphore, 。 But in I2Cdev::readBytes function the i2c reading procedure is as follow ，before the requestFrom() the semphore is taken by beginTransmission function , so requestFrom will always pending on the taken semphore and no one will release it.
> 
> I2Cdev::readBytes useWire->beginTransmission(devAddr); useWire->write(regAddr); useWire->endTransmission(); **useWire->beginTransmission(devAddr);** useWire->requestFrom((uint8_t)devAddr, (uint8_t)min((int)length - k, I2CDEVLIB_WIRE_BUFFER_LENGTH));
> 
> There is two approach to fix , 1. comment the sencond beginTransmission function in both I2Cdev::readBytes and I2Cdev::readWords;(it is ok for ESP32 but i dont if it will break the other hardware platform ) or 2. add a definition #define CONFIG_DISABLE_HAL_LOCKS 1 in esp32 for Arduino lib to disable the semaphore lock.

#633 

 #define CONFIG_DISABLE_HAL_LOCKS 1

Aarpp, 2023-04-04:
Has this " solution" been edited in the library? after searching through I2CDev.h /.cpp  & mpu6050.h/.cpp
I am unable to find the above-mentioned portion to attempt the edit.