jrowberg/i2cdevlib #695: Strange line in esp-idf mpu6050 example

brand17, 2022-06-25:
The file `example.cpp` has the following line:

`	    } else if (mpuIntStatus & 0x02) {
`

 to check for DMP data ready interrupt.

But I see here https://invensense.tdk.com/wp-content/uploads/2015/02/MPU-6000-Register-Map1.pdf (page 28) that the zero bit should be checked instead:

`	    } else if (mpuIntStatus & 0x01) {
`

jrowberg, 2022-06-26:
The DMP data ready interrupt is not the same as the general data ready interrupt, so nope. They just don't document it in the official register map section...argh.

closed 2022-06-26 by jrowberg

brand17, 2022-06-26:
Thanks, Jeff)

I am running the standard example https://github.com/jrowberg/i2cdevlib/blob/master/ESP32_ESP-IDF/main/example.cpp. I only had to swap PIN_SDA and PIN_CLK to make it working (based on https://www.instructables.com/ESP32-Internal-Details-and-Pinout/). But the bit 1 is always zero. 

What could be a reason? 

I've checked that DMP initialized correctly and enabled.

If I try to check the zero bit instead - I am getting wrong sensor orientation (looking like random figures).

I also ensured that the sensor is ok: your example https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/MPU6050/examples/MPU6050_DMP6 works like a charm from Arduino IDE.

Please note that I don't use the sensor interrupt pin - I don't see use of that in the example (interrupt PIN or smth like that).