jrowberg/i2cdevlib #703: MPU6050 Tap feature

kanimaru, 2022-07-31:
Hi guys what have to be done to get the Tap feature that the datasheet is promoting?
Probably another version of [dmpMemory](https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/MPU6050/MPU6050_6Axis_MotionApps612.cpp#L135) but where you get it in the first place?
And then I'm confused what to do next? Shouldn't be there a documentation how the DMP communicate these information?

My use case: 
I have an ESP32 connected to GY512 as a portable device. To save power I just want to activate the device when it get tapped. 
If there is a better Idea it would be also OK otherwise how the heck this works :).

Much thanks in advance.

TROFER, 2022-10-01:
Hi Kanimaru,

I am also looking to get tap detection working. The MPU6050 does support it according to the spec but this library doesn't have that support yet. Did you manage to find any alternatives?

kanimaru, 2022-10-08:
Was looking for other modules but didn't find any and currently I have no time for that anymore. Sorry.

asuslennikov, 2024-02-04:
You know, although this library is good, it's not the only one, which supports MPU6050. The official driver available almost for 10 years now. You can download it yourself from manufacture's website: https://invensense.tdk.com/developers/software-downloads/#mpu-en

Also, there is several repos you can check as an example:
- https://github.com/VoxMi/MPU-9150
- https://github.com/greedyhao/MotionDriver2RTT
- https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library

The SparkFun library already has tap example: https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/blob/master/examples/MPU9250_DMP_Tap/MPU9250_DMP_Tap.ino. You only need to define the correct sensor version (`#define MPU6050`)

If for some reasons, you highly coupled with this library, it is possible to reproduce the same behavior here (mix MPU6050_DMP6_using_DMP_V6.12 with SpurkFun example).

```
const int16_t mpuAddress = 0x68;  //It can be 0x68 or 0x69
MPU6050 mpu(mpuAddress);
bool dmpReady = false;               // set true if DMP init was successful
uint8_t fifoBuffer[64];              // FIFO storage buffer
volatile bool mpuInterrupt = false;  // indicates whether MPU interrupt pin has gone high

void dmpDataReady() {
  mpuInterrupt = true;
}

void enableMpuInterrupts() {
  attachInterrupt(digitalPinToInterrupt(INTERRUPT_PIN), dmpDataReady, RISING);
}

void mpuWriteMemory(unsigned short memAddress, unsigned short length, unsigned char *data) {
  mpu.writeMemoryBlock(data, length, memAddress >> 8, memAddress & 0xFF);
}

void setTwoByteValue(int value, unsigned char *data) {
  data[0] = value >> 8;
  data[1] = value & 0xFF;
}

void configureTaps() {
  // Array to store register values
  unsigned char tmp[4];

  // Add taps to FIFO buffer (i.e. DMP output)
  // see dmp_enable_feature method
  // https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/blob/master/src/util/inv_mpu_dmp_motion_driver.c#L1005
  tmp[0] = 0x20;
  mpuWriteMemory(2742, 1, tmp);  // CFG_27

  // Enable taps
  // see dmp_enable_feature method
  // https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/blob/master/src/util/inv_mpu_dmp_motion_driver.c#L1030
  tmp[0] = 0xF8;
  mpuWriteMemory(2224, 1, tmp);  // CFG_20

  // set FIFO rate
  // see dmp_set_fifo_rate method
  // https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/blob/master/src/util/inv_mpu_dmp_motion_driver.c#L660
  setTwoByteValue((DMP_SAMPLE_RATE / 10 - 1), tmp);
  mpuWriteMemory(534, 2, tmp);  // D_0_22
  const unsigned char regs_end[12] = { 0xfe, 0xf2, 0xab, 0xc4, 0xaa, 0xf1, 0xdf, 0xdf, 0xBB, 0xAF, 0xdf, 0xdf };
  mpuWriteMemory(1106, 12, (unsigned char *)regs_end);  // CFG_6

  // Set threshold (0-1600) for each axis, for example 30
  // see dmp_set_tap_thresh method
  // https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/blob/master/src/util/inv_mpu_dmp_motion_driver.c#L727
  // Divide threshold by sample rate (200)
  // And then multuply by current accelerometer LSB (16384, 8192, 4096, 2048)
  // For some reason also save 0.75 value (12288, 6144, 3072, 1536)
  // X-Axis
  setTwoByteValue((30 * 16384 / DMP_SAMPLE_RATE), tmp);
  mpuWriteMemory(468, 2, tmp);  // DMP_TAP_THX
  setTwoByteValue((30 * 12288 / DMP_SAMPLE_RATE), tmp);
  mpuWriteMemory(292, 2, tmp);  // D_1_36
[... 200 lines cut ...]

void mpuWriteMemory(unsigned short memAddress, unsigned short length, unsigned char *data) {
  // mpu.writeMemoryBlock(data, length, memAddress >> 8, memAddress & 0xFF);
 // instead of actual write to the sensor - just update the array
  for (int i = 0; i <= length; i++) {
    dmpMemory[i + memAddress] = data[i];
  }
}

void configureTaps() {
   // see above
}

void setup() {
  Serial.begin(115200);
  configureTaps();
  printDmpMemory();
}

void loop() {
  // do nothing
}

```
But I didn't test it. You can produce the `dmpMemory` output on simulator (), for example https://wokwi.com/projects/new/esp32 (esp32 just for memory requirements)