jrowberg/i2cdevlib #716: missing members?

hellooutthere2, 2022-11-21:
Compiling an older program which used i2cdevlib and devices. Missing members from MPU_6050_Base. enablePassThrough(), magTriggerMeasurement(), and getMag(int16, int16, int16). Any help would be appreciated. Thanks

hellooutthere2, 2022-11-21:
I'm guessing I'm missing some information here... The 6050 doesn't have a magnetometer. I'll have to dig into the code some more.

ZHomeSlice, 2022-11-21:
The magnetometer is connected to the secondary i2c bus that is part of the mpu6050 (the standard breakout boards available do not have these pins exposed). it is simpler to directly connect the magnetometer to the primary i2c bus as the embedded DMP Firmware doesn't do anything with the magnetometer data when it comes to 9 degrees of freedom. It requires an external .lib file to provide the magnetometer integration.
If you were to add a Magnetometer to the secondary i2c buss of the mpu6050 you could do one of 2 things. you could pass through the buss allowing you to directly access it as if it were on the primary i2c bus or you could have the mpu6050 get the data and store it in the registers. these registers are accessed when you want the data through the primary bus. 
I hope this helps.
Z