jrowberg/i2cdevlib #718: Bug in i2cdevlib/RP2040/I2Cdev/I2Cdev.cpp I2Cdev::readWords() does not work

alphatech56, 2022-11-24:
in I2Cdev::readWords() line 183 code J+2 
should be j += 2

alphatech56, 2022-11-24:
I have fix, but can't push.

jrowberg, 2022-11-24:
You're the second person in a week or so who has noted that they can't push a change. Normally, this works fine for this repo, and I haven't changed anything about the repo permissions recently. Can you clarify what steps you have tried, and what errors you're receiving? (Obviously the fix you've noted here is easy enough for me to make, but I'd like to figure out why you can't do a merge request.)

alphatech56, 2022-11-25:
This is a transcript from my attempt. I have my public key in github.

BTW, if I have changes that need review. What is the procedure? Some
changes are bug fixes, but some of them my  personal preference, not
necessarily what should be in the public repo.

 ***@***.***:~/depot/i2cdevlibArduino/i2cdevlib $ git pull
Current branch master is up to date.
***@***.***:~/depot/i2cdevlibArduino/i2cdevlib $ git remote -v
origin  ***@***.***:jrowberg/i2cdevlib.git (fetch)
origin  ***@***.***:jrowberg/i2cdevlib.git (push)
***@***.***:~/depot/i2cdevlibArduino/i2cdevlib $ git push
ERROR: Permission to jrowberg/i2cdevlib.git denied to alphatech56.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
***@***.***:~/depot/i2cdevlibArduino/i2cdevlib $

Thanks. Naum

jrowberg, 2022-11-25:
Ah, I see the problem. You can't push directly to someone else's repository unless you're added by the repo owner as a contributor with write permissions. This page describes the standard Github contribution workflow:

https://jarv.is/notes/how-to-pull-request-fork-github/

...which goes something like this:

1. Fork the original repo
2. Clone your new forked repo
3. Create a branch in your repo specifically for your fix(es)
4. Apply/commit/push your changes to that branch
5. Submit a pull request so the original repo owner can review and merge your changes

It takes a bit of effort the first couple of times, but it makes sense after you step through it, and it gives you credit for your changes if your commits are merged.