jrowberg/i2cdevlib #733: PlatformIO registry package outdated / critical bug

PhilippMolitor, 2023-02-23:
The package of I2Cdevlib is severely outdated on the PlatformIO registry. I used the Arduino variant, which has a bug in there that renders all `read` methods unusable as they invoke `Wire->beginTransmission();` right before trying to read from the device, which results in nothing being read.

Please consider to release an update.

jrowberg, 2023-02-24:
Hil @PhilippMolitor,

Can you confirm which code you actually have? The incorrect calls to `beginTransmission()` were removed from all remaining parts of the code (as far as I know) last October, in this commit:

https://github.com/jrowberg/i2cdevlib/commit/98a3b4ec838223fd5316de54203a6dc88718cd00

My understanding is that PlatformIO pulls directly from the Github repo rather than maintaining a separate copy of any code, so if you have the latest version installed, it should be working. @ivankravets was the one who actually created the PlatformIO definition for these libraries (I wasn't really aware of PIO at the time), so if there's some version/code caching going on, maybe he can clarify.

The structure of the I2Cdevlib repository--a terrible mistake in retrospect--makes it challenging to do any kind of sane versioning and release tagging, so this may be contributing to the issue. I've pretty extensively described all the flaws and how it should have been done instead on the [Perilib intro post](https://community.perilib.io/t/perilib-i2cdevlib-reborn/15).

PhilippMolitor, 2023-02-24:
This is what the header says:

```cpp
// I2Cdev library collection - Main I2C device class
// Abstracts bit and byte I2C R/W functions into a convenient class
// 2013-06-05 by Jeff Rowberg <jeff@rowberg.net>
//
// Changelog:
//      2021-09-28 - allow custom Wire object as transaction function argument
//      2020-01-20 - hardija : complete support for Teensy 3.x
//      2015-10-30 - simondlevy : support i2c_t3 for Teensy3.1
//      2013-05-06 - add Francesco Ferrara's Fastwire v0.24 implementation with small modifications
//      2013-05-05 - fix issue with writing bit values to words (Sasquatch/Farzanegan)
//      2012-06-09 - fix major issue with reading > 32 bytes at a time with Arduino Wire
//                 - add compiler warnings when using outdated or IDE or limited I2Cdev implementation
//      2011-11-01 - fix write*Bits mask calculation (thanks sasquatch @ Arduino forums)
//      2011-10-03 - added automatic Arduino version detection for ease of use
//      2011-10-02 - added Gene Knight's NBWire TwoWire class implementation with small modifications
//      2011-08-31 - added support for Arduino 1.0 Wire library (methods are different from 0.x)
//      2011-08-03 - added optional timeout parameter to read* methods to easily change from default
//      2011-08-02 - added support for 16-bit registers
//                 - fixed incorrect Doxygen comments on some methods
//                 - added timeout value for read operations (thanks mem @ Arduino forums)
//      2011-07-30 - changed read/write function structures to return success or byte counts
//                 - made all methods static for multi-device memory savings
//      2011-07-28 - initial release
```

Last update in september, though this is the same header as in the master branches current state. The `beginTransmission()` calls are definitely in there.
On PlatformIO, the last release is 9 months ago: https://registry.platformio.org/libraries/jrowberg/I2Cdevlib-Core/installation

It seems like it is tied to a custom release strategy - it says version 1.0.0 on the registry website, but I see not git tag / release in this repo. Also, the last change is 2 days ago acording to the registry, so there seems to be some kind of awareness in regard to git activity.

ivankravets, 2023-02-24:
@jrowberg, thanks for pinging me. I'm working on this issue. Let me back soon with updates.

ivankravets, 2023-02-24:
The [1.0.0](https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/I2Cdev/library.json#L3) version is pinned to the "Jun 3, 2022". Do instruct PlatformIO to update all libraries, @jrowberg, please update the `version` field in all `library.json` files.

For example, from `1.0.0` to `1.0.1` (meaning path version) or `1.1.0` meaning the version is fully compatible with the previous 1.0.0 API but adds new functionalty.

PhilippMolitor, 2023-02-25:
From what I can tell from the header changelig, no relevant features have changed since then. @jrowberg would you like to update the package version to 1.0.1 as described by @ivankravets so the working version is live?

jrowberg, 2023-02-25:
Thanks for the info. The impact of the version on the way PlatformIO handles things makes sense. I've just updated it to 1.0.1 in `library.json`.

closed 2023-02-25 by jrowberg

ivankravets, 2023-02-25:
Thanks, @jrowberg! I see new version has been picked by the crawler https://registry.platformio.org/libraries/jrowberg/I2Cdevlib-Core