jrowberg/i2cdevlib #743: I get no output in Serial Monitor Wesmos D1 Mini

F1L1Pv2, 2023-04-24:
`

#include "I2Cdev.h"

#include "MPU6050_6Axis_MotionApps20.h"
//#include "MPU6050.h" // not necessary if using MotionApps include file

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
// is used in I2Cdev.h
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
    #include "Wire.h"
#endif

// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69
MPU6050 mpu;
//MPU6050 mpu(0x69); // <-- use for AD0 high

// uncomment "OUTPUT_READABLE_QUATERNION" if you want to see the actual
// quaternion components in a [w, x, y, z] format (not best for parsing
// on a remote host such as Processing or something though)
//#define OUTPUT_READABLE_QUATERNION

// uncomment "OUTPUT_READABLE_EULER" if you want to see Euler angles
// (in degrees) calculated from the quaternions coming from the FIFO.
// Note that Euler angles suffer from gimbal lock (for more info, see
// http://en.wikipedia.org/wiki/Gimbal_lock)
//#define OUTPUT_READABLE_EULER

// uncomment "OUTPUT_READABLE_YAWPITCHROLL" if you want to see the yaw/
// pitch/roll angles (in degrees) calculated from the quaternions coming
// from the FIFO. Note this also requires gravity vector calculations.
// Also note that yaw/pitch/roll angles suffer from gimbal lock (for
// more info, see: http://en.wikipedia.org/wiki/Gimbal_lock)
#define OUTPUT_READABLE_YAWPITCHROLL

// uncomment "OUTPUT_READABLE_REALACCEL" if you want to see acceleration
// components with gravity removed. This acceleration reference frame is
// not compensated for orientation, so +X is always +X according to the
// sensor, just without the effects of gravity. If you want acceleration
// compensated for orientation, us OUTPUT_READABLE_WORLDACCEL instead.
//#define OUTPUT_READABLE_REALACCEL

// uncomment "OUTPUT_READABLE_WORLDACCEL" if you want to see acceleration
// components with gravity removed and adjusted for the world frame of
// reference (yaw is relative to initial orientation, since no magnetometer
// is present in this case). Could be quite handy in some cases.
//#define OUTPUT_READABLE_WORLDACCEL

// uncomment "OUTPUT_TEAPOT" if you want output that matches the
// format used for the InvenSense teapot demo
//#define OUTPUT_TEAPOT

#define INTERRUPT_PIN D5  // use pin 2 on Arduino Uno & most boards
#define LED_PIN 13 // (Arduino is 13, Teensy is 11, Teensy++ is 6)
bool blinkState = false;

// MPU control/status vars
bool dmpReady = false;  // set true if DMP init was successful
uint8_t mpuIntStatus;   // holds actual interrupt status byte from MPU
uint8_t devStatus;      // return status after each device operation (0 = success, !0 = error)
uint16_t packetSize;    // expected DMP packet size (default is 42 bytes)
uint16_t fifoCount;     // count of all bytes currently in FIFO
uint8_t fifoBuffer[64]; // FIFO storage buffer

// orientation/motion vars
Quaternion q;           // [w, x, y, z]         quaternion container
VectorInt16 aa;         // [x, y, z]            accel sensor measurements
[... 178 lines cut ...]
            Serial.print("\t");
            Serial.println(aaWorld.z);
        #endif
    
        #ifdef OUTPUT_TEAPOT
            // display quaternion values in InvenSense Teapot demo format:
            teapotPacket[2] = fifoBuffer[0];
            teapotPacket[3] = fifoBuffer[1];
            teapotPacket[4] = fifoBuffer[4];
            teapotPacket[5] = fifoBuffer[5];
            teapotPacket[6] = fifoBuffer[8];
            teapotPacket[7] = fifoBuffer[9];
            teapotPacket[8] = fifoBuffer[12];
            teapotPacket[9] = fifoBuffer[13];
            Serial.write(teapotPacket, 14);
            teapotPacket[11]++; // packetCount, loops at 0xFF on purpose
        #endif

        // blink LED to indicate activity
        blinkState = !blinkState;
        digitalWrite(LED_PIN, blinkState);
    }
}

`

F1L1Pv2, 2023-04-24:
i can upload it just fine but i get no output mpu6050 works and wesmos is alright too because i tested them in other sketches

aribftap, 2023-04-24:
have you done use IMU_Zero on the example? or allready upload it and get the offset? i'm using esp8266 too and the board is nodemcu

if you use example MPU6050_DMP6 you have to do
1. change this #define INTERRUPT_PIN 15  // use pin 2 on Arduino Uno & most boards because i see on the file MPU6050_DMP6_WiFi the default is #define INTERRUPT_PIN 15 // use pin 15 on ESP8266
2. one the code 
    void dmpDataReady() {
    mpuInterrupt = true;
} change too
IRAM_ATTR void dmpDataReady() {
    mpuInterrupt = true;
}
3. change the offset on below this code to your offset that you get it from file IMU_Zero
// supply your own gyro offsets here, scaled for min sensitivity
    mpu.setXGyroOffset(220);
    mpu.setYGyroOffset(76);
    mpu.setZGyroOffset(-85);
    mpu.setZAccelOffset(1788); // 1688 factory default for my test chip
 
 Sorry for my bad english i hope you solved it

F1L1Pv2, 2023-04-25:
from imu_zero i got these results

```XAccel			            YAccel			ZAccel			         XGyro			YGyro			ZGyro
 [-743,-742] --> [-3,16]	[107,108] --> [-2,16]	[523,524] --> [16375,16393]	[-101,-100] --> [0,4]	[-22,-21] --> [-2,3]	[-25,-24] --> [-1,2]
-------------- done --------------```
how to put it into offsets? which values

F1L1Pv2, 2023-04-25:
[image]
no output serial monitor D:

aribftap, 2023-04-25:
For using the offset that you get from IMU_Zero, you can visit this;
https://www.instructables.com/MPU6050-Setup-and-Calibration-Guide/

check his youtube to understand how to use offset

and for MPU6050_DMP6, on your ESP8266 / Wemos D1 Mini you can push the button reset if output not shown up from Serial Monitor 

*Note for MPU6050_DMP6 you already set up the offset or your MPU6050 and follow my tutorial on first comment

Hope you solved it this time

aribftap, 2023-04-25:
> nevermind im just stupid i forgot to hardreset using button. still thanks for help

your welcome