jrowberg/i2cdevlib #777: Read Absolute Orientation

CristyRO9, 2025-01-04:
Hi! I used the MPU6050_DMP6 code and  it's working very well, but I have a problem, I seem to be reading relative orientation, I'm more interested in PITCH to be absolute and YAW to be relative. How can I do that? Thank you in advance!

jrowberg, 2025-01-05:
Pitch and roll should already be absolute, since orientation is computed according to gravity. Yaw is the only one that can drift without an absolute magnetic heading reference. Do you have the IMU chip itself mounted on a PCB such that it's flat/level relative to the ground? Or is it at 90 degrees in any direction?

CristyRO9, 2025-01-05:
Well I have the MPU6050 module placed on a breadboard alongside the Pro Mini 3.3V board, but whenever I power the setup it will take it's initial position as reference, something I don't want. I have a model glider for which PITCH axis will be used only at the launch for a few seconds and then the YAW axis will be used to guide the glider to a specific direction, set by a potentiometer, I don't need the true azimuth, just a reference. Only for the PTCH axis I want absolute readings relative to earth, because on the field I can't position the glider perfectly level when powering on. I will post a link to a [video ](https://drive.google.com/file/d/1SyFEGeR_Em_CkjzhsBhncIvdAw96B2qm/view?usp=sharing)to see what's happening. Thank you!

CristyRO9, 2025-01-06:
How can I read the absolute orientation regarding earth?

CristyRO9, 2025-01-07:
Please help me... :(