jrowberg/i2cdevlib #786: Potential crash on I2C communication failure due to ESP_ERROR_CHECK

devprofile98, 2025-08-09:
Hi, Thanks for the great library. 

I am using i2cdev and esp-idf for an ESP32 project to connect to MPU6050, everything is ok in debug build, but in production, I think the ESP_ERROR_CHECK is misused, for example, the `i2c_master_cmd_begin` function can fail because the bus is busy (`ESP_ERR_TIMEOUT`), In debug environments this result in calling `aborrt()`, which could be ok in many cases, but in production the `ESP_ERROR_CHECK` macro became empty, and the [`writeBytes`](https://github.com/jrowberg/i2cdevlib/blob/a98805775f5c14b618387742d5ac6ac9c14e215e/ESP32_ESP-IDF/components/I2Cdev/I2Cdev.cpp#L239) function will `return true`. This behavior is not suitable for a robust production application, considering I2C is a sensitive protocol.

Isn't it better to leverage a more appropriate error handling using `if/else` statements or just by returning the `esp_err_t` from the called function?  What do you think?

jrowberg, 2025-08-11:
This is a good observation, but I feel like I don't have much to offer directly because I've never used the ESP IDF development approach--only the Arduino platform implementation. Someone else contributed the IDF variant of the I2Cdevlib core. Could you fork, modify, and submit a tested pull request that improves things to add the functionality you recommend?

devprofile98, 2025-08-12:
Of course, I will try to give the function correct return types and replace ESP_ERROR_CHECKs with robust error handling. Thank you.